PowerGrids/PowerGrids #8: BusFault model

adriguir, 2020-02-21:
@AndreaBartolini @casella I planned to add a new test case with a bus fault for IEEE14 system as well as potentially developing a line fault model (ie having a fault applied somewhere on a line) and thus I had a look to the bus fault model. 

The most natural implementation of such a model for me would have been something like:
when (t >= tBegin and t < tEnd) then ZI = V elsewhen then I = 0 
which directly reflects the fact that I add a link between the port and the earth during the short-circuit.

Is there a particular reason to have implemented the when loop inside algorithm blocks? Is there an advantage of using the intermediate fault boolean value?

casella, 2020-02-24:
As we understand, there is a fundamental difference between time events and state events. In the case of time events, since time goes on independently of anything else, I guess it would be more or less the same to have
```
algorithm
when time > startTime then
  fault:=true
end when;
when time > stopTime then
  fault:=false
end when
algorithm
when fault then
  Y := 1/Complex(R,X);
end when;
when not fault then
  Y := 0
end when;
```
and
```
equation
if time > startTime and time < stopTime then
  Y = 1/Complex(R,X);
else 
  Y = 0;
end if;  
```
In fact, in the latter case the backend must automatically convert the if equations into equations with conditional terms, such as 
```
  Y = if time > startTime and time < stopTime then 1/Complex(R,X) else 0;
```
which is potentially problematic if there are multiple equations in the if-equation branches.

The situation is different in case of _state_ events, because then using the when clauses with the boolean variables corresponding to the event triggering allows to use pre() and thus to decouple potentially unwanted large mixed algebraic loops, which you can't do if you just write if.

There is a potential catch also in models that have events depending on boolean inputs (e.g. `LineConstantImpedanceWithBreakers`), because then it depends how those inputs are generated: if they are computed as time events, evertything is going to be fine, while if you have something like `breakerStatusA = I > Imax`, then you may run into trouble with unwanted algebraic loops.

All things considered, we would deem it safer to always use the same approach:
- use a when statement to catch events and change the value of a boolean variable
- trigger the right action based on pre(boolanVariable)

This guarantees that no matter how you generate that event, no algebraic loops will ever be generated, and event iteration will take care of the job.

If you are ok with this, I guess we should update some of the library models (e.g. `LineConstantImpedanceWithBreakers`)

casella, 2024-07-25:
@AndreaBartolini I understand this has already been implemented a long time ago. Can we close this ticket?

AndreaBartolini, 2026-09-17:
I read again the entire ticket... now the picture is more clear in my mind...

I'll proceed to harmonize all the components according to  https://github.com/PowerGrids/PowerGrids/issues/8#issuecomment-590360303.