PowerGrids/PowerGrids #15: Model verification with discrete keyword

adriguir, 2020-06-21:
Apparently, the two following models (LineConstantImpedanceWithBreakers and IEEEbusPowerFlow which is using the first mentioned) don't validate with Dymola.
The problem comes from the variable YA_Actual, YB_Actual and Y_Actual that are defined with the prefix discrete and used outside a when loop (indeed into an if condition statement, itself depending on an algorithm output). What I see in the language specification is that a discrete variable can only change its value at events during simulation. I suppose that the condition of the if being set by a when, this statement is verified in the current implementation but I would like to have other advices. 

@casella @AndreaBartolini what is your opinion?

casella, 2020-06-21:
@adriguir, you are absolutely right. In fact, OMC currently does not detect such errors, see [#5836](https://trac.openmodelica.org/OpenModelica/ticket/5836).

One option is to remove the discrete keyword and make those variable normal ones.

The other option is to directly set the values of those variables in the when statement. 

@AndreaBartolini, do you remember why we used this pattern and did not set those values directly in the when statement?

casella, 2020-06-30:
In the meantime, ticket [#5836](https://trac.openmodelica.org/OpenModelica/ticket/5836) was fixed, which means that now several models in PowerGrids correctly fail because of this. I'll make a PR to remove the discrete keyword, which is the easiest way to fix the problem, then we can continue the discussion if this is the most efficient way to deal with this kind of situations.

adriguir, 2020-06-30:
Fine for me.

closed 2020-06-30 by adriguir

AndreaBartolini, 2020-06-30:
> 
> @AndreaBartolini, do you remember why we used this pattern and did not set those values directly in the when statement?

In the first version of the model the values of the discrete variables YA_Actual, YB_Actual and Y_Actual was assigned inside the when statement in the algorithm section.
Recently we moved said variables to the equation section (but forgot to remove the discrete prefix... ), I don't exactly remember the reason why we move them but it was related with some problems we got compiling the IEEE 14 bus network.

casella, 2020-06-30:
I vaguely remember something like that. I guess keeping them inside the when statement created some (real or fictitious) algebraic loop that OMC could not untangle, because from the backend point of view, whatever happens inside an algorithm is a black box where all left-hand side variables depend directly on all right-hand-side variables.

So, I think this PR is indeed the right solution to the problem.

PR #16 (2020-06-30): Removed illegal discrete prefixes. Fixes #15
Fixes #15. Also checked with Dymola
commit 0ac867ca (2020-06-30): Merge pull request #16 from casella/fix-discrete

Removed illegal discrete prefixes. Fixes #15
commit 496cebab (2020-06-30): Removed illegal discrete prefixes. Fixes #15