PowerGrids/PowerGrids #36: Publish version 1.0.2 in OpenModelica?

pierre-haessig, 2021-09-22:
Hello,

I'm not familiar with how libraries are pushed to the OpenModelica library distribution, but I see that installing PowerGrids with the Linux version of OpenModelica (through the `deb http://build.openmodelica.org/apt buster stable` Debian package source which currently serves OM 1.18 beta 2), I can only install version 1.0.0 (package `omlib-powergrids-1.0.0`). What steps are needed to publish the latest release to OM?

Best,
Pierre

casella, 2021-09-22:
See the [documentation of the package manger](https://openmodelica.org/doc/OpenModelicaUsersGuide/latest/packagemanager.html), if you have further questions @sjoelund can help you.

As far as I understand, PowerGrids 1.0.2 is released, so the package manager should get it.

sjoelund, 2021-09-22:
The package manager has it in https://libraries.openmodelica.org/index/v1/index.json

The libraries in the repository are to be removed (soon; we just had the release).

pierre-haessig, 2021-09-23:
Thanks for the feedback. I was not aware of this transition in the way of distributing the libraries (although I've heard a bit about the package manager). So the issue is not relevant anymore.

closed 2021-09-23 by pierre-haessig