PowerGrids/PowerGrids #37: Update the compatibility & solver section of the README

pierre-haessig, 2021-09-22:
Hello, 

Present version of README includes a quite technical paragraph about compatibility and efficiency:

> PowerGrids was developed and tested using OpenModelica 1.14.1, but is expected to run on all Modelica tools that fully support Complex numbers and the homotopy transformation. For numerical efficiency of models with more than a few buses, it is necessary that the tool does not transform the equations into ODE form, but rather directly uses a sparse DAE solver. The reason for this requirement is that while the DAE is sparse, the corresponding ODE is dense. Note that most of the models and test cases run with this version of OpenModelica but some solver issues are still under consideration and should only be available from OpenModelica 1.16 - in case you want to benefit from the latest developments, we recommend you to download the nightly-build from OpenModelica.

Since OM 1.16 was released in late 2020, I proposed to simplify the message, because newcomers may not be interested in the full history and may not be aware of technical details like homotopy initialization and DAE vs ODE solver:

> Paragraph labeled PowerGrids is developed and tested using OpenModelica (≥1.16), but is expected to run on all Modelica tools that fully support Complex numbers and the homotopy transformation [I would add here 2-3 major compilers that enter this category, if any]. 

Of course, the `≥1.16` recommendation should be incremented if some planned bugfix were delayed.

Then, I would split the paragraph about the solver selection and reformulate more towards _what I should do, as a user_. For example, when I open an example in OMEdit 1.18, and enter the simulation setup, the default solver is `dassl`, and I have no idea if it's good or not (searching for it on the web sends me to a [ Petzold, 1982 technical report](https://www.osti.gov/biblio/139120-dassl-differential-algebraic-system-solver) where I understand that it's a DAE solver, but the short tooltip in OMEdit is silent on this fact). So I recommend adding a list of reasonable solvers. If I need to tinker with the "Translation flags" in OM's Simulation Setup dialog, this should be stated here. This is how I imagine the paragraph:

> For numerical efficiency of models with more than a few buses, it is necessary that the tool does not transform the equations into ODE form, but rather directly uses a sparse DAE solver. The reason for this requirement is that while the DAE is sparse, the corresponding ODE is dense. Practically, for OpenModelica users, this implies _doing this and that..._ [and perhaps add similar explanations for other common tools, if applicable]

All these paragraphs could be headed by a title "Compiler and solver compatibility and recommendations"

casella, 2021-09-23:
@pierre-haessig, thanks for bringing this up! There are two problems.

First, we should try to avoid repeating the same information in multiple places, because it is inevitably bound to be obsolete. Currently we say something about compatibility in the readme.md file and in the embedded library documentation. This is not a good idea in general, because you have to update it in two places all the time and keep it consistent. On the other hand, asking people to download the library and open it with a Modelica tool to access the documentation, just to discover that it doesn't work, won't be nice. I think I will shorten the compatibility info in the Readme.md file and refer to the library documentation for more detailed information.

Regarding the information itself, it is definitely obsolete, as it still refers to old versions. I'll update it following your advice and release 1.0.3 ASAP.

pierre-haessig, 2021-09-24:
> I think I will shorten the compatibility info in the Readme.md file and refer to the library documentation for more detailed information

This solution seems to me very appropriate. Perhaps just mention the appropriate tool (like "use OM version X.Y) in the README and keep the discussion on DAE vs ODE sparsity and solver selection for the doc (which I just happen to read now... and  this is a nice proof of usability: one can start playing with PowerGrids Examples without reading the manual first ;-) ).

casella, 2021-09-24:
> This solution seems to me very appropriate. Perhaps just mention the appropriate tool (like "use OM version X.Y) in the README

Please check #38 and comment there, if you wish.

> and keep the discussion on DAE vs ODE sparsity and solver selection for the doc (which I just happen to read now... and this is a nice proof of usability: one can start playing with PowerGrids Examples without reading the manual first ;-) ).

I am aware of the fact that nobody reads the documentation, ever 😄 

If things work out of the box, fine, otherwise you can always read it afterwards. Comments on that are also very welcome, it is difficult for us developers, who worked on this problem for years, to really understand how a newcomer gets it, so we can for sure further improve it.

casella, 2023-05-08:
@pierre-haessig we will actually do this for 2.0.0 as soon as we roll it out, we are skipping 1.0.3.