PowerGrids/PowerGrids #39: Convert PowerGrids to MSL 4.0.0?

casella, 2021-09-23:
@adriguir, MSL 4.0.0 has been out for a while, should we make a 2.0.0 release which is MSL 4.0.0 compatible? Future versions of MSL will be backwards compatible for a few years, so I think we should do this ASAP to avoid new users the trouble of converting later to that.

What do you think?

AndreaBartolini, 2023-03-30:
I looked at what happens by converting the PowerGrids library to MSL 4.0.0 and it preserves the compatibility with developed models that use it.
Obviously, if a model that uses PowerGrids has some incompatibilities with MSL 4.0.0 then for that model the back-compatibility is not preserved.

My proposal is to:
1) merge the PR (https://github.com/PowerGrids/PowerGrids/pull/38) that preserves the back-compatibility
2) solve and merge the issue (https://github.com/PowerGrids/PowerGrids/issues/23), whose solution may not preserve the back-compatibility
3) solve and merge the issue (https://github.com/PowerGrids/PowerGrids/issues/53), whose solution does not preserve the back-compatibility
4) convert PowerGrids to MSL 4.0.0 and TAG the new version as 2.0.0

AndreaBartolini, 2023-03-30:
Probably is better to include also the PR (https://github.com/PowerGrids/PowerGrids/pull/46), rewritten accordig to the relevant comments, before to convert PowerGrids to MSL 4.0.0

AndreaBartolini, 2023-04-03:
The same regarding the issue https://github.com/PowerGrids/PowerGrids/issues/49, for that the solution is similar to that of issue https://github.com/PowerGrids/PowerGrids/issues/23. For both the issues my proposal is to apply the most coherent solutions from the structural point of view (i.e., delete the parameter `c` in https://github.com/PowerGrids/PowerGrids/issues/23 and made `final `the parameters in https://github.com/PowerGrids/PowerGrids/issues/49), that unfortunately does not preserve the back compatibility but, at this point, it is not a problem. 

So the updated to-do list becomes the follow:

1. (ALREADY DONE) merge the PR (https://github.com/PowerGrids/PowerGrids/pull/38) that preserves the back-compatibility
2. solve and merge the issue (https://github.com/PowerGrids/PowerGrids/issues/23),
3. solve and merge the issue (https://github.com/PowerGrids/PowerGrids/issues/49),
4. solve and merge the issue (https://github.com/PowerGrids/PowerGrids/issues/53),
5. rewrite and and merge the PR (https://github.com/PowerGrids/PowerGrids/pull/46),
6. convert PowerGrids to MSL 4.0.0 and TAG the new version as 2.0.0

AndreaBartolini, 2023-04-03:
In order to have an "all-in-one" transition to the "non backward-compatibility" version 2.0.0 my proposal is to prepare the PRs and then merge all of them in the same time (by respecting the right order to avoid conflicts).

AndreaBartolini, 2023-05-13:
solved by PR #70

closed 2023-05-15 by AndreaBartolini