PowerGrids/PowerGrids #40: Numerical problem at initialization when the AVR is saturated

casella, 2021-10-04:
The model in the tutorial that is initialized in steady-state with the AVR in saturation,  `PowerGrids.Examples.Tutorial.GridOperation.Controlled.ControlledGridWithReferenceCalculationSat` fails because of [singular initialization](https://libraries.openmodelica.org/branches/master/PowerGrids/files/PowerGrids_PowerGrids.Examples.Tutorial.GridOperation.Controlled.ControlledGridWithReferenceCalculationSat.sim). This error was not caught in the past by our CI infrastructure because of a bug that made the outcome of the simulation look positive, while in fact there was none.

I have briefly checked the problem. My current understanding is that the culprit is the AVR model. When there is no saturation involved, the steady-state system equations determine `efdPu`, and then the blocks of the AVR are solved backwards in steady-state until the `fixed=false` offset value for the voltage set point is computed in the offset block.

When I lower the VrMax parameter of the AVR, this affects the firstOrderWithNonWindupLimiter block. However, the [initial equation](https://github.com/PowerGrids/PowerGrids/blob/98bbd943f03846c6b83bf855ebfe46e68c6db687/PowerGrids/Electrical/Controls/ExcitationSystems/IEEE_AC4A.mo#L58) in the IEEE_AC4A model considers the input limiter, not the output limitation, so I guess this somehow creates a singularity because there are some variables in-between the two that may be undetermined.

We should carefully analyze this issue and improve the `AC4A` model so that this doesn't happen anymore.

casella, 2024-07-25:
Possibly related to #152?

AndreaBartolini, 2026-09-14:
After a discussion with @casella, we ended-up to the conclusion that it is no possibile to manage the initialization in saturation state by means of the homotopy, so the mechanism has been removed from the AVR.

closed 2026-09-14 by AndreaBartolini