PowerGrids/PowerGrids #49: Questionable nominal and initial power values of buses and loads

max-privato, 2022-05-30:
**Regarding PowerGrids.Electrical.buses.Bus and referenceBus, InfiniteBus, BusFault, InfiniteBusVariableVoltage:**
- In their _General_ page the mask prompts for SNom. but what's the meaning of it? Buses should not absorb neither deliver any power, so their S is always zero and this field should not appear
- In _Initialization_ page the mask prompts for PStart and QStart. But what's the meaning of them? buses should not absorb neither deliver any power, so their Q and Q are always zero and this value should not appear
- if we set PStart=0 in the Initialization page, an error message is issued stating that that value cannot be changed. This confirms that that field must not be there.

regarding 
**Powergrids.Electrical.Loads.LoadPQVoltageDependence and LoadImpedancePQ:**
- In initialization page the mask prompts for PStart and QStart. The user assumes they are active and reactive initially absorbed by the load. But they should not be there, since we know them from the Ustart value (and PRefConst, QRefConst, alpha and beta values).  The default values are also strange sinte they are not consistent with the values set in the _General_ page

pierre-haessig, 2022-06-09:
Just to clarify when you say *"if we set PStart=0 in the Initialization page, an error message is issued stating that that value cannot be changed. "*, what model are you simulating? Directly a `XXXBus` model or an example which instantiates a few buses?

pierre-haessig, 2022-06-09:
I can confirm part of the issue about changing `PStart` for a bus

Test case: 
- using OM 1.19.0~dev.beta1
- using a duplicate of [PowerGrids.Examples.Tutorial.GridOperation.Static.PowerFlow](https://github.com/PowerGrids/PowerGrids/blob/master/PowerGrids/Examples/Tutorial/GridOperation/Static/PowerFlow.mo)
- I can change the Initialization of `PStart` in the `NLTV` bus of type `Bus`. 

Then, trying to simulate yields the error message:

```
[Trying to override final element PStart with modifier '(displayUnit = "W") = 0'.](omeditmessagesbrowser:///PowerGrids.Electrical.Buses.Bus?lineNumber=5)
```

This is coherent with the definition of a [Bus](https://github.com/PowerGrids/PowerGrids/blob/master/PowerGrids/Electrical/Buses/Bus.mo)

```Modelica
model Bus "Bus model"
  extends PowerGrids.Electrical.BaseClasses.OnePortAC(
    final PStart = 0,
    final QStart = 0);
...
```
So this seems to be an OpenModelica issue, because the user should not be able to change final values?

## Extra issue

In addition, the Initialization tab shows wrong default values: it shows `PStart` being equal to `SNom` while the Bus class definition says it's 0. The displayed value `SNom` is in fact the default value set in the parent class [`OnePortAC`](https://github.com/PowerGrids/PowerGrids/blob/master/PowerGrids/Electrical/BaseClasses/OnePortAC.mo#L17).

[image]

AndreaBartolini, 2023-04-03:
The proposed to-do list in https://github.com/PowerGrids/PowerGrids/issues/39 doesn't preserve the back-compatibility, so my proposal is to made `final` the SNom parameter, to be coherent with the others.

AndreaBartolini, 2023-04-03:
The SNom value is used as default value of SBase in PortAC base model, therefore it cannot be set=0 because this produces a lot of "division by 0" errors if the SBase value is left to its default.
Moreover, not in the OnePortAC nor in the Bus models the parameter SBase is accesible to the user, that means taht it is not possible to assign a value different from SNom to it.

One possible solution is to made final = 0 the SNom parameter and made accessible the SBase parameter in OnePortAC model (thus in all the derived models, included the bus), in order to allow the user to set the right base for the powers.

AndreaBartolini, 2023-04-03:
Probably this will break a lot of models, due to the SNom parameter has been used to set the SBase scaling of powers in buses up to now.

ceraolo, 2023-04-04:
> The SNom value is used as default value of SBase in PortAC base model, therefore it cannot be set=0 because this produces a lot of "division by 0" errors if the SBase value is left to its default.

It looks strange for me that the SBase of a bus is used for scaling. From my viewpoint (the viewpoint of a guy that has not worked to programming the library), a base for scaling a quantity has meaning (and should be used) ony for components in which that quantity is different from zero. And the bus by definition should not generate nor absorb powers.

As far as I can understand it should therefore be possible to remove any internal usage of SBase for buses.
Isn't it?

casella, 2023-04-04:
@AndreaBartolini SBase is used to to get p.u. power. For Bus, the power flow is identically zero, so any SBase value would do. If @max-privato agrees, I would recommend to make SNom final with a value of 1 W. That would be enough to get all the p.u. values to be evaluated as zero already at compile time.

For all other Bus models (referenceBus, InfiniteBus, BusFault, InfiniteBusVariableVoltage) the power flow into the connector can be non-zero, so we should keep SNom as a parameter to be set by the modeller.

AndreaBartolini, 2023-04-05:
@casella the proposed solution to make `SNom` final with a value of 1W sounds good to me.

Regarding `SBase`, it is currently used as a _"pure alias"_ in the library, because the `PortAC` is never extended (it is only instantiated) and the classes that instantiate it (`OnePortAC` and `TwoPortAC`) do not propagate `SBase`. 
So the point is to decide if there is a valid reason to maintain the `SBase` as is (pure alias) or if may be useful to propagate it, in order to made it accessible to the user, or if it is better to delete it, in order to avoid the pure aliases.
In case of the propagation of `SBase`, it means that the user will have (in general) both parameters `SNom `and `SBase `(with default `SNom`) available in the parameters pop-up.

ceraolo, 2023-04-05:
> For `Bus`, the power flow is identically zero, so any non-zero `SBase` value would do. If @max-privato agrees, I would recommend to make `SNom` `final` with a value of 1 W

sounds good (but see below regarding the name Snom).

For the other buses:
- infiniteBus. Its behaviour is defined by its impedance, so SNom and Unom are needed only for numerical reasons. **Note that the info mentions also emf E**, which is however missing in the input mask. Is there something wrong?
- referenceBus. It seems to me that its behaviour is defined just by UStart and UPhaseStart, while PStart, QSstart , UNom and SNom are needed only for numerical reasons. Am I wright? 
- busFault. Before fault, it is like a normal bus, so, maybe it should be treated as 'Bus'? (i.e. Snom with 'final' equal to 1 W)

For other components, such as loads and machines, SNom and Unom have direct influence on the voltage behaviour. 
Are we sure that using Snom and Unom equivalently for numerical reasons or, instead, to define che component's behaviour the right choice? Maybe it is better to use different names?

@AndreaBartolini mentions showing to users both Sbase and Snom. 
In case what I said before is right, maibe one could leave the names Snom and Unom in the input masks **only to components where these values are used to define che component's behaviour**, and use, instead, other names (Sbase Pbase, Qbase and Ubase? Sref Pref, Qref and Uref?) as "reference" quantities, whose values are needed to be known just approximately, as needed for convergence reasons (i.e. internal scaling). There seems to me that at present there is not uniformity on these names in PowerGrids

AndreaBartolini, 2023-04-12:
After a global review of the library, and taking into account that `SNom` and `UNom` are defined and used everywhere in the library at very basic low-level, my opinion is to avoid to change the original philosophy of that names, that foresees to use the `"Nom"` qualifier to indicate that the parameter is used as `"nominal"` value of the variable, in the same way of the `"nominal"` attribute of a Modelica `Type` (i.e., for scaling). As previously discussed, these values can be fixed to `1` and made `final` in the components where they play no role (like SNom in the bus).
Regarding the `SBase` and `UBase` pure aliases, my proposal is to delete them.
Regarding the use of `SNom` and `UNom` as default values for local parameters (as currently done in the library), we can remove these defaults if they generate confusion in the user (@ceraolo: what do you think about this last aspect?).

AndreaBartolini, 2023-04-12:
Moreover,
regarding the OpenModelica issue cited above (see https://github.com/OpenModelica/OpenModelica/issues/8425) my opinion is to avoid to add "patches" in order to `hide` the final parameters in the parameters pop-up windows.

ceraolo, 2023-04-12:
The idea of this ticket is to try to avoid what currently happens, i.e. that the word "nominal" means totally different things depending on the position of this word in the library. And to be clean in the use of "nominale", "rated", "base".
From https://github.com/PowerGrids/PowerGrids/issues/49#issuecomment-1505405588 I understand that it is impractical to change the meaning of Snom and Unom whenever they are intended as the modelica "nominal" value of the variable, i.e. a value used internally for scaling, to ease computation robustness.
Am I right? In this case, I think it is highly confusing having names "Snom" and "Unom" used in a totally different meaning, as the rated values for components, e.g. for synchronous machines, as in the screenshot below. 

Maybe in this case the best solution is to use Srat and Urat (or Srated and Urated), and use as description string "Rated line-to-line voltage" and "Rated appartent power"?

If this is done this way, this uniform terminology could be stated in the general part of the documentation. I mean, it should be clearly stated that "nominal" values are to be given just for internal numerical robustness, and they do not need to be precise (just to have the right order of magnitute), where "rated" values are necessary to interpret the numerical values of other parameters, e.g. to convert an internal impedance from p.u. into ohms.

If I understand well, this can be done in conjunction of elimination of Ubase and Sbase, isn't it? This would further clean things.

[image]

AndreaBartolini, 2023-04-12:
this is exactly the proposed idea... by this way the only names to be eventually changed are of local parameters, like SNom and UNom shown above that sould become SRated, URated, and/or their default value where it is equal to Snom or UNom (if we want to supply a default value).
For sure, the final uniform terminology must be added to the documentation.

AndreaBartolini, 2023-04-13:
Here my proposal in detail:

`UNom, SNom` description: "nominal/rated" -> "Nominal for per-unit" or "Nominal for scaling"
`PowerGrids.Electrical.PowerFlow.ZLoad`: `PNom` ->` PRef`, `QNom` -> `QRef`
`PowerGrids.Electrical.Branches.Line...`: remove `UNom` (`UNomA` already available in parameters pop-up)
`PowerGrids.Electrical.Buses.Bus`: `SNom` -> `final SNom=1`
`PowerGrids.Electrical.Machines.SynchronousMachine...`: Add parameters `URated=UNom`, `SRated=SNom` and use `SNom` and `UNom` only for scaling

pierre-haessig, 2023-04-13:
> This would be backwards-compatible. In fact, since 2.0.0 won't be backwards compatible, maybe it would make sense to define defaults the other way round
> 
>   * "nominal" values are used for scaling (-> they set the nominal attribute). We set final nominal = base for models having base parameters, final nominal = rated for models having rated parameters, final = 1 when there is nothing to scale (e.g. in Bus models). I'd use final because there is no reason to use different values for scaling than those that are used for p.u. conversion
> 
>   * "base" values are used to compute per unit values (for monitoring!) in models whose behaviour is defined by SI-dimensioned quantities (e.g., transmission lines). No default value, they actually have to be given explicitly
> 
>   * "rated" values are used to convert per-units into SI units in models of machines where equations are written in per-unit. No default value, they actually have to be given explicitly
> 
> 
> I personally prefer this last option [...]

As a teacher of introductory courses on power grids, I also find the alternate version easier to use. The "nominal" value is more of an implementation detail of the library.

Now, there one more thing I don't get. For me each device can be a rated power (often related to thermal limits). However, the base power is *unique for a given power system* in power flow tools like Matpower (not sure how it is in PowerGrids though). So this creates an ambiguity with respect to per-unit values: is it with respect to one global base value or with respect to a device-specific rated value? Since Matpower doesn't include machine models, I don't know what is the common practice in the domain.

AndreaBartolini, 2023-04-17:
As far as I know, per-unit is device-specific because in a transmission network different voltage-levels are usually involved (high voltage 100KV, very high voltage 400KV, generators voltage 10KV and so on), thus it is difficult to define a common reference for the per-unit, this is the reason because PowerGrids allows to set the per-unit base for each component.

AndreaBartolini, 2023-04-17:
Regarding the definition of the default, I agree with the second proposal of @casella .

ceraolo, 2023-04-17:
Regarding the dedfinition of the defaults, I also agree with the second proposal of @casella.

Regarding the pu, the comment from @pierre-haessig is very sensible. When we do manual computations we usually use a unique base power for the whole system and different base voltages for different sections of it (the ratios of base voltages of two adjacent sections must be equal to the ratios of the nominal voltages of the transfomers in between to have the wanted advantages in terms of system analysis).
From the discussion in this ticket, however, I understand that the pu output values in PowerGrids are _with respect to the machine base_ that confirms the importance for the user to be able to clearly understand  what that base is, which, according to @casella 's proposal carry the "base" word in their name.

(BTW, for teaching reasons, I've sometimes used PowerGrids putting everywhere "1" for base powers and voltages, and inserting all machine and line parameters direcly in the system pu, so that to reproduce the computations are done in university Power System classes with the general pu system mentioned by @pierre-haessig, with excellent results)

AndreaBartolini, 2023-04-18:
I've just finished to made `final` the `SNom` parameter in the `Electrical.Buses.Bus` model (included the update of all test and examples in the library), and I'm starting with the implementation of the `nominal/base/rated` behavior as per @casella option 2.
The idea is to put all the modifications (multiple commits) in a unique PR that will close the issue.

As general solution, I propose to maintain the current policy that foresees to set the `base/rated` parameters for each component, this implies that is responsibility of the user to give congruent `base/rated` values according to, e.g., transformers voltages ratio. Different solutions may be considered (e.g., to put some common `base` values in the `system` object) but I suggest to open a new issue for that discussion.

AndreaBartolini, 2023-04-20:
Sorry, I closed the issue by mistake...

AndreaBartolini, 2023-04-20:
Second point: the calculation of some SI parameters (like `Z` in the `ZLoad` component) are currently made by using `xxNom` parameters, and this doesn't match the new `nominal/base/rated` philosophy.
My proposal is to change to `xxRef` all the parameters used to calculate SI parameters like Resistances, Impedances and so on.

casella, 2023-04-20:
> To avoid said warnings/errors a default value should be supplied to the `base/rated` parameters, but this doesn't match the intended philosophy that foresees to given to them the values explicitly . Considering that said warnings/errors are generated only during the component development (thus not in the normal use of the library by a not modelica-expert user) we can decide to accept this side effect.

I think you could give some start value to them. This is used whenever an actual value is not available (e.g. for checking) , but it will generate a warning when you compile for simulation if the parameter is not properly set with a modifier.

ceraolo, 2023-04-20:
> Second point: the calculation of some SI parameters (like `Z` in the `ZLoad` component) are currently made by using `xxNom` parameters, and this doesn't match the new `nominal/base/rated` philosophy. My proposal is to change to `xxRef` (i.e., `reference`) all the parameters used to calculate SI parameters like Resistances, Impedances and so on.
> 
Why not xxBase?
AFAIK xxNom is used for p.u.  (i.e. for p.u. and for scaling as well. For the user the prevailing meaning is related to p.u.)

casella, 2023-04-20:
@ceraolo, the logic is the folloing
- xxNom is used for scaling.
- xxBase is used for p.u, and in that case normally xxNom = xxBase
- xxRated is used in machines, and in that case xxBase = xxRated and xxNom = xxRated
- xxRef is used for loads (the reference load is the value you get when U = URef and omega = omegaRef). In this case not necessarily xxBase should be xxRef. You could have a variable load for which xxBase (or xxRated) is 100 MW, but for one specific simulation it starts at 40 MW reference power, i.e. 0.4 p.u.

ceraolo, 2023-04-20:
@casella, I agree. That's why it seems to me better for lines using xxBase than xxRef.

AndreaBartolini, 2023-04-20:
to better explain, the idea is that xxRef are now used when a phisical characteristic (like an impedance) is given by means of a reference working point (e.g., parameters `URef`, `PRef `and `QRef `in `Electrical.PowerFlow.ZLoad` now defines the `final parameter Z = URef^2/Complex(PRef, -QRef))` where in the current version we have `final parameter Z = UNom^2/Complex(PNom, -QNom)`)).
In the case of the lines the parameters `R, X, G and B` are given explicitly by the user, thus xxRef quantities are not needed.
The xxBase or xxRated quantities shall always given by the user and are used for the p.u.
Maybe that in some cases is useful to set xxRef=xxBase, but this is a choice of the user.

AndreaBartolini, 2023-04-21:
Regarding the lines again...
Currently the parametrisation of UNom in the line is implemented as following:
```
  extends BaseClasses.PiNetwork(
    UNomA = UNom,
    UNomB = UNomA);
  ...
  parameter Types.Voltage UNom(start = 400e3) "Nominal/rated voltage";
```
This allows the user to give either a unique UNom or two different UNomA/B but as counterpart all the parameters UNom, UnomA and UNomB are shown in the pop-up window and in my opinion this redundancy may induce confusion in the user.

In the new approach `xxNom `become `xxBase`, but the problem of the redundancy is not solved. My proposal is to eliminate said redundancy by modifying the implementation in order to allow only one of the following possibilities:
1. The user can give only a unique `UBase`, that is propagated as `final UBaseA=UBase` and `final UBaseB=UBase`
2. The user can give both the `UBaseA `and the `UBaseB`, by default `UBaseB = UBaseA`

I think the first is the better because I don't expect a diferent UBase across a line, but I'm not an expert in power transmission.

casella, 2023-04-21:
> to better explain, the idea is that xxRef are now used when a physical characteristic (like an impedance) is given by means of a reference working point (e.g., parameters `URef`, `PRef `and `QRef `in `Electrical.PowerFlow.ZLoad` now defines the `final parameter Z = URef^2/Complex(PRef, -QRef))` where in the current version we have `final parameter Z = UNom^2/Complex(PNom, -QNom)`)).

LGTM

> In the case of the lines the parameters `R, X, G and B` are given explicitly by the user, thus xxRef quantities are not needed. 

Yes. However, you need xxBase to compute p.u. quantities in the ports (always defined), as well as xxNom quantities for scaling. I understand that in the case of transmission lines, we ask the user to provide xxBase and then set final xxNom = xxBase.

> The xxBase or xxRated quantities shall always given by the user and are used for the p.u. Maybe that in some cases is useful to set xxRef=xxBase, but this is a choice of the user.

This could be a reasonable default value.

casella, 2023-04-21:
> Regarding the lines again... Currently the parametrisation of UNom in the line is implemented as following:
>
> ```
>   extends BaseClasses.PiNetwork(
>     UNomA = UNom,
>     UNomB = UNomA);
>   ... [...]

I agree the first choice is the best one. Yes, lines have losses, but they are normally small, and besides they depend on the direction of the current flow. A 380 kV line is a 380 kV line, you see the losses because the voltage at the other side is 0.98 p.u., not because the other side has 370 kV base value. That would be really confusing.

ceraolo, 2023-04-21:
> I agree the first choice is the best one. Yes, lines have losses, but they are normally small, and besides they depend on the direction of the current flow. A 380 kV line is a 380 kV line, you see [...]

100% agree

AndreaBartolini, 2023-04-21:
> But... maybe some difference in their icon is advisable? Not sure, just proposing something to give two minute of thinking, since we are making a deep revision of the library. Maybe just a red "i" added to the icon to mean "internal parameters"?

I agree with the idea to highlight the difference between the two kind of generators by adding something to the icon, the little red "i" may be a solution.

casella, 2023-04-21:
> In technical language the most common attribute is nominal, but we cannot use it because we decided to use nominal for scaling (which in turn, descends from the fact that Modelica has a different meaning for the word "nominal" than in technical language).
> 
> As a substitute for nominal, rated is pretty good, so I agree with your list.

Wait, then maybe we can make it even simpler. I mean, we will never use a value different from the "nominal" (in the sense of electrical engineers) one for the purposes of scaling. Right?

In that case, we can keep `xxNom` and `xxBase` only

- `xxNom` is always defined. It means nominal (i.e. rated) when applicable, _and_ "used for scaling" in all cases
- `xxBase` means the base value for p.u. definition and is always defined (there are p.u. values in the ports). Defaults to xxNom. In most cases that's ok, but you can change it if you want.
- no need for final modifiers
- `xxRef` is used e.g. for loads, where "reference" means "value when voltage and frequency are nominal". Defaults to xxNom but could be change

I am a fan of Ockham's razor, so I prefer this proposal to the previous one because it uses less concepts to achieve the same goal.

In fact, one question remains: do we ever use values different from nominal as base values for p.u.? If not, then we could make `final xxBase = xxNom`. I would keep xxBase as an alias, to make very clear what are the base values for p.u.

casella, 2023-04-21:
>     * the `type PowerGrids.Types.Choices.ReferenceFrequency` currently includes the value `nominalFrequency`

This is no longer an issue if (as I recommend) we implement the proposal laid out in #60.

ceraolo, 2023-04-22:
Let me see if I understand well, looking at it from a user's perspective.

**The idea, I think, is that average user won't even know of the need for internal scaling.**  Only those reading all the details of the documentation, will possibly be aware of this.

If this is the idea, I very much welcome this proposal

I try to list details from a user's perspective, considering the most important components. 

1. Machines (Synchronous and transformer). The user  will be presented in the components' mask xxNom, intended as the nominal, or rated values of the machine, the ones from which p.u. impedances are computed from the corresponding ohm values. They will also be hinted for xxBase, even though xxBase will be defauted to xxNom. The machine will not ask for xxRef nor xxRated. For transformer we will have one xxNom per winding.

2. Lines. The user  will be presented in the components' mask just xxBase, intended as the values from which p.u. output values are computed (one xxBase for both ends)

3. Impedance loads described through P and Q. The user  will be asked in the components' mask just for vRef, intended as the value at which, from the inputed P and Q, the internal impedances are computed. They will also be hinted for xxBase, even though vBase will be defaulted to vRef

To summarise:
- xxNom in terms of internal scaling will not be exposed to users
- depending on the component, the user will be asked from the most relevant xx: xNom for machines, xxBase for lines, xxRef for loads.
- except from lines, for which xxBase is directly asked, the user will also be hinted for xxBase, which will normally be defaulted to the usrer inputs (xxNom or xxRef).

Am I right?
It seems to me that this will work well.

ceraolo, 2023-04-28:
In https://github.com/PowerGrids/PowerGrids/issues/49#issuecomment-1518561206 I forgot to mention buses.

In principle buses should be equivalent to direct connection of two or more components' terminals, but in practice they are not. 
I remember having met a case in which the simulation converged with a bus, did not with a direct connection.
I suppose this is due to the fact that in buses we can specify xxNom, and they are used somehow as a facilitator for the evaluation at least of the initial conditions of the simulation.

Now, I wonder. What is the meaning of xxNom (or, with the new approach xxBase) for buses? Since buses connect other components we can use to ease convergence xxBase of the connected components' terminals, isn't it? If this is true, or made true by means of some modification in the library, can we safely state (even in documentation) that buses and direct connections are totally equivalent?

AndreaBartolini, 2023-05-02:
> Let me see if I understand well, looking at it from a user's perspective.
> 
> **The idea, I think, is that average user won't even know of the need for internal scaling.** Only those reading all the details of the documentation, will possibly be aware of this....

I'm not shure but it seems to me that we are changing again the philosophy ... I've already implemented (not yet merged) the one proposed by @casella and here below reported:

**xxNom** is used for scaling.
**xxBase** is used for p.u, and in that case normally xxNom = xxBase
**xxRated** is used in machines, and in that case xxBase = xxRated and xxNom = xxRated
**xxRef** is used for loads (the reference load is the value you get when U = URef and omega = omegaRef). In this case not necessarily xxBase should be xxRef. You could have a variable load for which xxBase (or xxRated) is 100 MW, but for one specific simulation it starts at 40 MW reference power, i.e. 0.4 p.u.

that means that xxNom is never shown to the user.
The user has access to xxBase (or xxRated for generators) and to xxRef (for example for loads), xxNom is final and is assigned as before inside the library.

Now it seems that the user shall assign again the xxNom... 

Please note that this issue are bloking all the others, because it defines a low-level behaviour that impacts everywere in the library, so it is necessary to close it if we want to proceed with the work.

pierre-haessig, 2023-05-02:
> Regarding the pu, the comment from @pierre-haessig is very sensible. When we do manual computations we usually use a unique base power for the whole system and different base voltages for different [...]

Thanks for clarifying my point. Indeed @AndreaBartolini's https://github.com/PowerGrids/PowerGrids/issues/49#issuecomment-1511579501 is correct: per-unit voltage can be different in different areas of a network (e.g. HV/MV/LV areas). See for example the [Matpower's webapp](https://matpower.app/), where base voltage is defined for each bus (not for lines since they just connect two buses) while base power (named "baseMVA" in Matpower which makes it clear what the unit is...) is a global setting of a given system (https://matpower.app/options).

casella, 2023-05-02:
I discussed this issue again with @AndreaBartolini. My [previous proposal](https://github.com/PowerGrids/PowerGrids/issues/49#issuecomment-1518340024) tried to improve the situation by removing some unnecessary parameters, but at the same time making xxBase _explicit_. However, it was not satisfactory because at the end of the day, in practice one would never pick a value of xxBase which is different from xxNom, so what's the point of leaving it in the input mask (not making it final)? In fact, it is not even clear if picking a different value could actually cause trouble. Not a good design.

We just have to make sure that users understand _what is used in order to compute p.u. values on the port_. No need to keep the input field on xxBase to do that, just make it clear in the parameter descriptions.

From these premises, we agreed that we can essentially keep what we have now on master, but with a few key changes
- xxNom is always present in the input mask, possibly with some default value (see below the PQ load case). The comment to those parameters is "Nominal/rated xx value, **also used as p.u. base**"
- the models in the library actually also use it for scaling, but that is explained in the documentation, non need to bother end users trying to explain it on the input parameter mask
- exception to this rule: SNom for the Bus component is declared as final SNom = 1, so it is removed from the input mask, since SPu = 0 all the time (no power flow through the port). As a consequence port.P, port.Q, port.PPu, port.QPu etc. will all be zero, as they should, with no divisions by zero.
- xxRef parameters are used to define the behaviour of a components by specifying a reference operating point. In this case, the default value for xxNom is defined in terms of xxRef to avoid double input if one wants to see 1 p.u. at the reference operating point - this can be changed by explicitly inputting some different value for xxNom.

Some examples:
1. Machines (Synchronous and transformer). The user will be asked for xxNom in the input mask, intended as the nominal, or rated values of the machine, _but also_ as the values from which p.u. impedances are computed from the corresponding ohm values, as made it clear from the comment to the parameter. No need of xxRef. For transformer we will have one UNom per winding. xxNom will also be used for scaling, but we don't mention that in the parameter input mask.
2. Lines: same as machines, will be asked for xxNom in the input mask, intended as the nominal, or rated values of the line, _but also_ as the ones from which p.u. impedances are computed from the corresponding ohm values. xxNom values are asked only once and used for both ports, contrary to transformers that have one SNom but two separate UNomA and UNomB. We can give a default value for SNom = UNom^2/X, which corresponds to the maximum power transfer when neglecting the losses, which is probably good enough in most cases, but can always be changed.
3. Constant impedance loads described through P and Q. The user will be asked in the components' mask for PRefConst, QRefConst and URef (intended as the value at which, from the inputed P and Q, the internal impedances are computed). Regarding UNom and SNom, they will also appear in the input mask, but they will have the defaults UNom = URef and SNom = sqrt(PRefConst^2+ QRefConst^2) already set. One could always change them if you want to see a value of p.u. different from 1 at the reference operating point, which can be different from the nominal one (e.g. you are working at 50% load, so you want to see 0.5 p.u.)

@ceraolo, we hope this is also acceptable for you, please give feedback.

casella, 2023-05-02:
> In principle buses should be equivalent to direct connection of two or more components' terminals, but in practice they are not. I remember having met a case in which the simulation converged with a [...]

If you add a bus, then the bus _voltage_ becomes an unknown of the grid operating point. It will get a start value which is by default UNom, unless you change UStart to be something different. The _same_ voltage is also present as the terminal voltage of the two (or more) connected components. If they are not consistent, which one is actually picked is non-deterministic. 

This is a bit redundant, because every voltage is present in the problem at least twice (if you connect two components) or three times (if you connect them to a bus). Unfortunately we can't see any way to avoid that. The bottom line is that UNom should be set correctly everywhere.

In the case you mention, maybe some parameters (e.g. UNom) were not correctly set in the two components that you connected, but then adding the bus gave that information to the solver.
 
> Now, I wonder. What is the meaning of xxNom (or, with the new approach xxBase) for buses?

- to set base values so you can see the node (bus) voltages of the network in p.u.
- to act as default value for UStart

> Since buses connect other components we can use to ease convergence xxBase of the connected components' terminals, isn't it? If this is true, or made true by means of some modification in the [...]

They are equivalent if you set consistent start values at all components ports starting from some power flow computation (as we state in the documentation). If you don't, and then rely on default start attributes set from nominal values to set the initial guesses of the Newton solver, your mileage may vary.

ceraolo, 2023-05-03:
Regarding https://github.com/PowerGrids/PowerGrids/issues/49#issuecomment-1531862058 I agree.

ceraolo, 2023-05-03:
Regarding https://github.com/PowerGrids/PowerGrids/issues/49#issuecomment-1531877386 I feel like stating again that a user it appears to me a bit wierd to have three different starting voltages, when connecting two lines though a bus, since I know that indeed I have just one voltage.

As a user, I would have expected some internal technique that allowed to set just one voltage, or, anyway, force the three of them to be equal to each other, possibly issuing a warning. Probably any non-modelica simulation program does this (just thinking, never checked).

But I can imagine that this has difficulties with the PowerGrids architecture, and therefore as a user I may accept living with this redundancy.

casella, 2023-05-03:
> Regarding [#49 (comment)](https://github.com/PowerGrids/PowerGrids/issues/49#issuecomment-1531877386) I feel like stating again that a user it appears to me a bit wierd to have three different starting voltages, when connecting two lines though a bus, since I know that indeed I have just one voltage.
> 
> As a user, I would have expected some internal technique that allowed to set just one voltage, or, anyway, force the three of them to be equal to each other, possibly issuing a warning. Probably any non-modelica simulation program does this (just thinking, never checked).

[Section 8.6.2](https://specification.modelica.org/maint/3.6/equations.html#recommended-selection-of-start-values) of the Modelica Specification introduces some priority criteria to select which start attribute to pick when you have several alias variables (i.e., variables that are set to be equal).

In the specific case, one would set an explicit value for UStart when instantiating the bus, while leaving the default binding UStart = UNom for the two transmission lines. I would expect that the priority criteria would select the value that was modified when instantiating the bus over the other two that were left to their default.

OpenModelica still does not implement these rules, but it will in the future. I tried this with Dymola (which should implement them) but still didn't get the expected result. We have to investigate further if this is a problem with how the priority rules are formulated or rather a tool implementation issue, but the idea is that eventually, if you only set the start attributes explicitly on the bus, this value should have precedence over the default values set by the library models.

> 
> But I can imagine that this has difficulties with the PowerGrids architecture, and therefore as a user I may accept living with this redundancy.

casella, 2023-05-03:
We have considered this further. In fact it turns out that with the current priority rules the start attributes for voltage in the bus and in the two lines have the same priority, because they both stem from modifiers applied at the same level; one (directly) to UStart, for the bus, and the other two (indirectly) to UNom, which is then propagated to UStart.

We should probably futher improve those rules, by stating that when the priority is the same, one should count how many biniding must be considered from the original modifier to the actual start attribute under question. In this case, the route when modifying UStart is shorter than when modifying UNom (which has an additional binding UStart = UNom), so the former should have precedence.

We will discuss this in the Modelica Language group, and if we can find agreement we'll implement the rule in OpenModelica.

If we can't find an agreement, we are pretty stuck, we can't see any other possible way to avoid this redundancy. In fact, there could be a "brute force" way: we could mandate that buses are always present, and then add to them output variables for UStart, and add input variables for UStart to all other components. However, this is unacceptable, because it _forces_ you to always use buses everywhere, which is pretty much against proper object-orientation.

AndreaBartolini, 2023-05-04:
Regarding the documentation, where is in your opinion the best place to put the explanation of the new philosophy of the use of  xxNom, xxBase, xxRef and Nominal?

casella, 2023-05-04:
> Not sure. Presently we can make connection, say of a line to a transformer, either directly or through a bus. Direct connection looks a bit strange to me as a power engineer; even stranger would be systems part whose connections are direct, part through buses. So maybe it is better to be forced to have always buses intersposed? I cannot see, at the moment disadvantages to this philosophy.

This would be quite a change compared to the current library, and also compared to all other Modelica libraries I know of, where you normally don't have to use "connection components" such as buses to connect device models.

_If_ people in this community are used to always have buses at connection points, then it could make sense to make this compulsory. This would mean that start values of voltage will no longer need to be given _at all ports_, but only _at all buses_ (including infinite ones and equivalent grids). However, you will still need to provide PStart and QStart at all ports (not for buses, which have Q = P = 0). 

There would be two types of terminalAC connectors, one for buses including two outputs for UStart and UPhaseStart, and one for all other components, including two inputs for the same quantity. If you forget to put the bus in the connection set, the compiler would complain about missing UStart outputs, that would be easy to recognize.

An advantage would be that we avoid depending on arcane start values to ensure that proper initial voltage values are used. 

We should definitely discuss this with RTE.

casella, 2023-05-04:
> Regarding the documentation, where is in your opinion the best place to put the explanation of the new philosophy of the use of xxNom, xxBase, xxRef and Nominal?

User's Guide | Tutorial | Library Architecture | Basic Modelling Framework

ceraolo, 2023-05-05:
_I suggest to move the last part of the discusson to a new issue, in order to close the already implemented part after the merge._

For me it is ok.

AndreaBartolini, 2023-05-05:
Discussion has been moved to #62

AndreaBartolini, 2023-05-05:
solved by the PR #63

casella, 2023-05-05:
@AndreaBartolini I guess we can now close this, right?

AndreaBartolini, 2023-05-05:
waiting for the continuous integration check :)

casella, 2023-05-05:
> waiting for the continuous integration check :)

Sure 😅

casella, 2024-07-25:
After some further considerations, we were not fully happy with the outcome in the case of the load models.

First of all, there was a circular dependency of the default values: `UNom = URef` and `URef = UNom`. This is confusing from a user's perspective, because leaving the default values will lead to a compile-time error and it is not clear which of the two should be set. 

Furthermore, setting the default `SNom` on the basis of `PRefConst` and `QRefConst`, which both default to zero, means that with default values one gets divisions by zero. BTW, it is perfectly OK to have both P and Q equal to zero, so the current default is not good (defaults should always work).

After some discussion with @AndreaBartolini, we decided to remove the default for `SNom`, which should be supplied explicitly. We also decided to keep UNom as the master parameter and to have `URef = UNom` by default, rather than the other way round, which is a bit confusing. We also removed the `SRef` parameter which is not really used for anything; `SNom` should be supplied instead, as with all other `OnePortAC` components.

All tests were updated accordingly, as well as the relevant tutorial section.

See ef1815b

closed 2024-07-25 by casella