PowerGrids/PowerGrids #51: Erroneously null current and voltage phases

max-privato, 2022-06-02:
Consider model mdl inside the following package:
[Test.mo.txt](https://github.com/PowerGrids/PowerGrids/files/8825050/Test.mo.txt)
It shows the transient due to a short circuit at bus42, initiated at t=0.1 and removed at t=0.2.
Angular  speeds and voltage amplitudes all are very reasonable.

However, a lot of phases are strangely exactly zero, such as:
- phases of voltage and current at ports A and B of line L2_1
- phases of voltage and current at ports of loads pq13 and pq42.

They cannot be all zero. In particular, in a line with zero resistance, with some power flowing the phases of the two port voltages cannot be equal to each other, and therefore cannot both be zero.

casella, 2022-06-02:
@max-privato, the computation of phase angles is optional, to save time on large models. I guess it is off by default. The variables are defined anyway, so they are trivially computed to zero in that case. Can you please check?

AndreaBartolini, 2023-03-31:
To activate the phase calculation in a component the parameter `portVariablesPhases `shall be set to `true` (the default value is `false` to save simulation time on large models).

Maybe useful to add a global setting in the `System object`, to allow the activation of the phase calulation in the entire model from a unique place.

casella, 2023-03-31:
To avoid misunderstandings, we could also set the HideResult annotation based on portVariablePhases, so the result is not shown when it is actually not computed.

casella, 2023-03-31:
BTW, one can also avoid getting those outputs by setting the -d=evalOutputOnly compiler flag, and by adding one top-level generator frequency as top-level output variable. This causes the backend to elminate from the simulation problem all the variables and equations that are not strictly necessary to compute the output.

This feature is discussed in #8272. Results posted there show that the feature is very effective in reducing the actual simulation times of non-trivial systems (up to -50%), even though it still scales badly, probably due to naive use of MetaModelica data structures in the implementation.

AndreaBartolini, 2023-05-05:
> To avoid misunderstandings, we could also set the `HideResult` annotation based on `portVariablePhases`, so the result is not shown when it is actually not computed.

This feature was already present in the library but it didn't work due to a bug (it will be fixed in the PR).

Note that the  `HideResult` annotation doesn't work in OpenModelica if the `portVariablePhases` parameter is not evaluated (see [#10588](https://github.com/OpenModelica/OpenModelica/issues/10588)

AndreaBartolini, 2023-05-05:
solved by PR #64

casella, 2023-05-05:
> Note that the `HideResult` annotation doesn't work in OpenModelica if the `portVariablePhases` parameter is not evaluated (see [#10588](https://github.com/OpenModelica/OpenModelica/issues/10588)

@AndreaBartolini, maybe this is solved if you use the new instance-based GUI.

casella, 2023-05-06:
I guess we should use conditional components instead, as in this MWE:
```modelica
package TestConditional
  model M
    parameter Boolean compute_phase;
    Complex z;
    Real phase = Modelica.ComplexMath.arg(z) if compute_phase;
  equation
    z = Complex(1, time);
  end M;
  
  model Test
    M m1(compute_phase = true);
    M m2(compute_phase = false);
  end Test;
end TestConditional;
```

AndreaBartolini, 2023-05-06:
In this case we can follow this way, I will update the code.

casella, 2024-02-13:
Unfortunately, having UPhase as conditional variable doesn't work since we have introduced the visualization of the phase in buses. This is currently done with this statement
```modelica
 extends Icons.Bus(VPuIcon = port.VPu, UIcon = port.U, UPhaseIcon = port.UPhase);
```
UPhaseIcon is a variable defined in the Icons.Bus model, that is used in the DynamicSelect annotations to display the phase of buses on diagrams. As reported in #126, this generates warnings (that should probably be errors) because you cannot refer to conditional variables outside connect statements.

Bottom line, if we want to display phases in diagrams, we can't have port.UPhase as a conditional variable. It must be defined all the time, with value zero if portVariablePhases==false. As we already do with other port variables such as PGen, QGen, etc.

IF we could use HideResult with parameters, we could hide them when they are default computed to zero, but unfortunately we can't, because for some reason the specification wants a boolean literal. So I'm afraid we'll have to live with it

AndreaBartolini, 2024-02-16:
fixed by PR #129