PowerGrids/PowerGrids #52: Invalid root when simulating a short circuit on IEEE14

max-privato, 2022-07-21:
I'm trying to see the effects of a short circuit at node 14 using a model obtained with a minimal modification from the PowerGrids.Examples.IEEE13bus.IEEE14BusStaticNetwork.
I Just added the short at node 4. The resulting file is here (pls remove "txt" from the name):
[IEEE14busShort4.mo.txt](https://github.com/PowerGrids/PowerGrids/files/9158816/IEEE14busShort4.mo.txt)

I've chosen this example because it is already available on OpenIpsl. Although the OpenIpsl IEEE14 model is not perfectly equivalent to PowerGrids', it is near to it, so it would be interesting to compare the two results.

When I run it with OpenModelica 1.19 I get several instances of the error:
`IEEE14busShort4_12jac.c:14213: Invalid root: (0)^(-0.5).
`

The simulation continues, but the result is plainly wrong, since the voltage on node 4 is nearly zero after the fault is removed, while it should rapidly get back to the previous value.

The troubles seem to start when the fault is removed since the following message is also issued:
`While solving non-linear system an assertion failed at time 1.20006.
`

For completeness, I show below the node 14 voltage with OpenIpsl and PowerGrids (first 4 seconds):

OpenIPSL:
[image]

PowerGrids:
[image]

casella, 2023-05-12:
@ceraolo, I think this is a problem caused by the behaviour of the load models at extremely low voltage.

The load models consume active and reactive power depending on the voltage: `P = P0*(V/V0)^alpha`, `Q = Q0*(V/V0)^beta`. These are reasonable system-level models when V/V0 remains within the normal operating range of voltage, e.g. 0.8 to 1.05 p.u. or something like that. In the test model you posted, alpha = 1.5 and beta = 2.5.

When the fault is applied, V/V0 gets very close to zero, hence active and reactive power consumption becomes very small, and so do the voltage and current:

[image]

So far so good. 

The problem arises when the bus fault is cleared at time 1.2. One would then expect that the bus voltage returns to reasonable values, e.g. 0.9 p.u. and the load returns to consuming the "normal" amount of active and reactive power. Unfortunately, the nonlinear system of equations of the network has another solution, where the large current that was previously absorbed by the shorted bus is now absorbed by the load:
[image]
You get a consistent situation in which the load has a voltage very close to zero (e.g. 1e-30 p.u., or some time even just zero p.u.), so its absorbed power is zero (there's virtually no voltage) even with a large current, e.g. 20 p.u.. This large current enacts a large voltage drop in the transmission lines from the nearby buses, which allows the voltage of the formerly shunted bus to be practically zero.

So, as I see it the problem is that the model without the fault has two solutions: one is the correct solution representing normal operation, and the other is a bogus solution where the loads absorb a fantastic amount of current and manage to ge the bus voltage to zero. This solution has no physical meaning (why should loads do something like this?) but is mathematically correct, even though it is numerically challenging because sqrt(v_re^2 + v_im^2) is very close to zero, which is a singularity point for the Jacobian.

Unfortunately, the fault is cleared instantaneously by event handling, so the Newton solver tries to find the solution after the event using the voltage and current values before the event as an initial guess, and since they are closer to the bogus solution than to the right one, the solver converges to the former instead of on the latter. Then, it stays there by continuity.

I suspect that this solution would be unstable if we had an electromagnetic model of the transmission line, accounting for the dynamics of the inductive and capacitive effects, so an E/M model won't latch to it, but immediately start  fast transient ending up in the correct solution, which is instead stable. But as we are using phasors, there is no such thing.

I currently don't know why this doesn't happen with the OpenIPSL load model. Maybe it is formulated in a smarter way when the voltage becomes very low, or maybe it's just sheer luck with the Newton solver. This needs to be investigated.

In general, I wonder whether these simple load models actually make sense in shorted conditions. What does the literature on the topic (if any) say?

casella, 2023-05-12:
In fact, I suspect this is really a fundamental problem of "textbook" PQ load models. When the voltage approaches zero, the model becomes nearly singular, which means that the current is not really well-determined, and can become really large, which is of course completely unrealistic.

Maybe we could add some internal impedance to the PQ model, to prevent unrealistic flows of very large currents into the load when the voltage approaches zero, e.g.:
```
P = P0*(V/V0)^alpha + R_int*i^2;
```
This simple addition would prevent bogus solutions such as the one shown above, because even a fairly small internal resistance would absorb a huge amount of active power when the current is 20 p.u. 

On the other hand, this would be a nuisance during normal operation, because the actual active power consumption would always be (slightly) higher than the required value P0*(V/V0)^alpha when the voltage is close to 1 p.u., because of this parasite power consumption. This could be prevented by doing something like this:
```
P = max(P0*(V/V0)^alpha, R_int*i^2);
```
but then when P0 is set to zero (switched off), there will still be some parasite consumption. This could be prevented by changing the model to:
```
P = P0*max((V/V0)^alpha, eps*IPu^2);
```
The only problem with the last model is that it has a discontinuous derivative when the voltage gets to some specific low voltage value. However, this is probably not worth regularizing, since one will never simulate transients where the voltage is gradually reduced - in the case of a short-circuit simulation the voltage drops instantaneously to very low value, and then abruptly rises again to almost normal. 

I would make this extra feature optional by setting the default value of eps to zero and evaluating it, so we avoid increasing the computational load unecessarily when there are may loads involved and no short-circuit simulation is going to be performed.

casella, 2023-05-12:
I just tried that solution. It is successful at avoiding the bogus solution with near-zero voltage, but unfortunately there are still two solution, one good and one bad with lower voltage (not close to zero, more like 0.8 p.u.) and higher current (not as high as previously but still well above 1 p.u.) and the solver latches onto the second one after the short fault is cleared. I need to think about some other solution.

casella, 2023-05-12:
I had an alternative idea. The problem here is that when you try to fix the power, there are oftentimes two solutions, one good, one bad. This is a basic fact of life in power systems, as the risk of voltage collapse. The problem is, once the solver has latched onto one, there's no way to make it jump on the other. In our specific case, the problem is the when the fault is cleared, the initial guess is closer to the bad solution than to the good one.

The idea is the following: at the moment when the fault is cleared, the bus becomes floating again. What we could do instead is to have two event iterations. In the first one, the bus will be connected to a 1 p.u. voltage by a large conductance; this will force the load voltage to get near 1 p.u.. Then, at the next iteration, this connection is removed and the bus will be floating, but at that point, the solver will latch onto the "good" solution. The previous solution will be discarded, normally Modelica tools only record the values at the beginning and at the end of event iterations, but not the intermediate ones. I'll try this this ASAP.

ceraolo, 2023-05-12:
We've made a fast check with Digsilent, and it resulted that it allows making short circut computations according to a choice of standards, such as:
- IEC 60909, 
- VDE 0102 Part 0 /DIN EN 60909-0. 
- IEC 61363
- ANSI/IEEE 946

I've not made any checks, even in the names of these standards, just blindly quoting.

Probably to make a decision on how to deal with loads during short circuit, it is good to take inspiration from some of these.
I'm not saying that short circuits should be computed in PowerGrids _according to some IEC or IEEE standard_, which would be overkill, but just take inspiration so that to go in the right direction at least.

I'll ask someone to make this evaluation of how standards manage load modelling during short circuit.

In the meanwhile I just put forward a hypothesis: probably a good solution is to model in Powergrids loads with the alpha-beta model only up to a _minimum voltage_, say 90%. Below that threshold, they would be converted into impedance (alpha=beta=2), with P-Q continuity. 

Let's see what comes from the evaluation of standards.

ceraolo, 2023-05-13:
Well, I've just checked IEC 60909-2016, which is the most authoritative source of information for electrical engineers in an international framework.
In section 5.2 "calculation assumptions" clause states:

_Shunt admittances of non-rotating loads shall be neglected in the positive-, the negative- and the zero-sequence system._

So, the general idea is that loads do not absorb significant amounts of current during short circuits.

Now I try to find the rationale beyond this.
Consider that non rotating loads can be of the following types:
- impedances, whose current absorption decreases linearly with voltage, and whose currents are therefore smaller than normal currents, which even at nominal voltage are small in comparison with short circuit currents
- loads fed through converters (e.g. HVAC inverters), which usually are constant power loads, have currents that increase when voltage reduces; those loads, however, get disconnected at very low voltages, and therefore are assumed to be inactive during short circuit.

Disconnecting loads during short circuits, is something can be done by specific network programs to give IEC compiant results, not by PowerGrids for two reasons:
- we don't want to distinghish short circuit transients from any other transient, which would have been needed to use the "disconnect loads" principle
- the idea of removing loads during short circuits has been elaborated many decades ago, when computing power was much lower than today's, and has remained in the IEC standard, because no strong needs of removing it has apparently appared insofar; however, it is apparent that in real life short circuits loads absorb some power. In other words, it is allowed neglecting loads during short circuits, but not really required.

After this analysis I launch again my previous proposal: 

_In Powergrids we implement loads with the alpha-beta model only up to a minimum voltage, say 90%. Below that threshold, they would be converted into impedance (alpha=beta=2), with P-Q continuity. The minimum voltage threshold can be defaulted to 90% and made changeable by the user._

Using this we get several useful results:
- we should avoid the numerical difficulties of this ticket
- we make something physically sound. If the user knows some internal details of its load, they can even choose their own minimum voltage threshold (e.g. they know that an electronic load gets disconnected at V=0.8)
- we go in the direction of IEC standard, while getting even more realistic results since constant impedance loads do absorb some current even during short circuits
- what we do is good also in preparation of future PowerGrids additions of rotating loads.

Any comments? Is some power engineer reading this?

ceraolo, 2023-05-13:
@casella says:
_I currently don't know why this doesn't happen with the OpenIPSL load model. Maybe it is formulated in a smarter way when the voltage becomes very low, or maybe it's just sheer luck with the Newton solver. This needs to be investigated._

In OpenIPSL ieee14 model all loads have alpha=beta=2, i.e. are constant impedances. This should solve all numerical issues. 
Unfortunately, doing the same in PowerGrids does not give good results:
[IEEE14busShort4MOD.mo.txt](https://github.com/PowerGrids/PowerGrids/files/11469238/IEEE14busShort4MOD.mo.txt)

casella, 2023-05-13:
I think I have a solution on [this branch](https://github.com/casella/powergrids/tree/improve-bus-fault). Unfortunately OMC does not compile it because of OpenModelica/OpenModelica#10694, as soon as it get fixed I'll try it out.

casella, 2023-05-15:
Adding @AndreaBartolini to the loop

ceraolo, 2023-05-15:
I don't know how long it will take for https://github.com/OpenModelica/OpenModelica/issues/10694 to be fixed.
In case of need, one could workaround the issue separating assignments of real and imaginary parts as follows:
```
model TestRecordAssignment
  Complex v;
  discrete Complex v0;
equation
  v = Complex(sin(time), cos(time));
  when time > 0.5 then
    v0.re = pre(v.re);
    v0.im = pre(v.im);
  end when;
end TestRecordAssignment;
```

casella, 2023-05-15:
> In Powergrids we implement loads with the alpha-beta model only up to a minimum voltage, say 90%. Below that threshold, they would be converted into impedance (alpha=beta=2), with P-Q continuity. The minimum voltage threshold can be defaulted to 90% and made changeable by the user.

In principle this looks good to me. In practice, however, how switch from a P-Q representation to an impedance-based representation _with continuity_ using an equation-based representation is not immediately clear to me.

casella, 2023-05-15:
> In OpenIPSL ieee14 model all loads have alpha=beta=2, i.e. are constant impedances. This should solve all numerical issues. Unfortunately, doing the same in PowerGrids does not give good results: [...]

Yes, I also tried that. In principle `alpha=beta=2` corresponds to constant impedance, but the problem is, it becomes singular (a 0/0 singularity) at very low voltage. That is precisely what happens, also with alpha = 2 and beta = 2 we get convergence with a solution where VPu = 1e-31, and obviously the nice linear behaviour is lost there because of numerical reasons, and we get a bogus current, and a lot of numerical errors.

casella, 2023-05-15:
One reason why the solution around the 0/0 form may be numerically challenging is a poor choice of tearing variables. I switched off tearing with compiler flag --tearingMethod=minimalTearing, and I got the correct results both with `alpha = 1.5, beta = 2.5` and with `alpha = 2, beta = 2` :smile.

Apparently, the way the equations of the OpenIPSL are torn is more numerically robust. This is of course completely out of our control.

Bottom line: tearing in power system models hurts, see also #6108, #8275.

I don't have a completely satisfactory solution for this issue now. The recommendation for the time being is to run models that show any numerical issue with --tearingMethod=minimalTearing.

casella, 2023-05-15:
> In principle this looks good to me. In practice, however, how switch from a P-Q representation to an impedance-based representation _with continuity_ using an equation-based representation is not [...]

If I knew how to handle this, we would solve the 0/0 numerical issue. The problem is, we need to regularize this behaviour:

```modelica
if VPu > 0.9 then
  v*conj(i) = Complex(P,Q);
else
  v = Zi;
```
which corresponds to 
```modelica
if VPu > 0.9 then
  v.re*i.re + v.im*i.im = P;
  v.re*i.im - v.im*i.re = Q;
else
  v.re = Z.re*i.re - Z.im*i.im;
  v.im = Z.re*i.im + Z.im*i.re;
```
where Z could be obtained from UNom, PNom and QNom. The problem is, how can you join together smoothly those two behaviours?

Maybe one option is not to join them smoothly at all. We simply switch to an impedance-based solution when VPu goes below a certain threshold. But I would choose a small one, e.g. 0.1 p.u. This means that the impedance-based formulation is only used when a short fault takes place; since that always happens abruptly, there's no point trying to have a continuous behaviour.

On the other hand, doing this for all loads would significantly increase the model complexity, because of all those conditional equations. Not a good idea in general, when you don't need to support scenarios with shorted buses.

After all, I'm still inclined to prefer the solution with the three-stage shorted bus model, because that requires additional complication only for those buses who can get shorted.
> In principle this looks good to me. In practice, however, how switch from a P-Q representation to an impedance-based representation _with continuity_ using an equation-based representation is not [...]

casella, 2023-05-15:
I just tried the the three-state shorted bus model (on [this branch](https://github.com/casella/powergrids/tree/improve-bus-fault)) but it still gets me zero voltage when the fault is cleared 😞

casella, 2023-05-15:
I now tried the strategy suggested by @ceraolo 
> In Powergrids we implement loads with the alpha-beta model only up to a minimum voltage, say 90%. Below that threshold, they would be converted into impedance (alpha=beta=2), with P-Q continuity. The minimum voltage threshold can be defaulted to 90% and made changeable by the user.

in this way:
```modelica
parameter Types.PerUnit VPuThr = 0.1 "Threshold of p.u. voltage for low-voltage fixed-impedance approximation";
...
equation
...
  if noEvent(port.VPu > VPuThr) then
    port.P = PRef*U_URef^alpha;
    port.Q = QRef*U_URef^ beta;
  else
    port.v = port.i/CM.conj(Complex(PRef*VPuThr^alpha,QRef*VPuThr^ beta)/URef^2);
  end if;

```
i.e., when the p.u. voltage is below a certain threshold, instead of determining `P` and `Q` in the regular way, we implement an impedance, whose value matches the current and voltages when `vPu = VPuThr`. You can get a modified 1.0.2 version with this fix on [this branch](https://github.com/casella/powergrids/tree/regularize-load). 

I tried @ceraolo's original [IEEE14busShort4.mo](https://github.com/PowerGrids/PowerGrids/files/9158816/IEEE14busShort4.mo.txt) test model and it worked fine also without turning off tearing, which I guess means that this model is more robust at low voltage.

As I wrote previously, the only problem with this load model is that it introduces additional conditional statements in each and every load model, even though this won't be needed in 99% of the cases. We could make this behaviour optional by introducting a boolean parameter lowVoltageAsImpedance
```modelica
parameter Boolean lowVoltageAsImpedance = false annotation(Evaluate = true);
...
equation
...
  if noEvent(port.VPu > VPuThr) or not lowVoltageAsImpedance then
    port.P = PRef*U_URef^alpha;
    port.Q = QRef*U_URef^ beta;
  else
    port.v = port.i/CM.conj(Complex(PRef*VPuThr^alpha,QRef*VPuThr^ beta)/URef^2);
  end if;
```
This would ensure that the extra low-voltage behaviour is only implemented in those loads where `lowVoltageAsImpedance` is set to true.

The other thing that is not making it 100% sure is that I first tried this variant
```modelica
  if noEvent(port.VPu > VPuThr) then
    port.P = PRef*U_URef^alpha;
    port.Q = QRef*U_URef^ beta;
  else
    port.i = port.v*CM.conj(Complex(PRef*VPuThr^alpha,QRef*VPuThr^ beta)/URef^2);
  end if;
```
which in principle should have been equivalent, but in practice made the solver fail because of singular Jacobian as soon as the fault was activated. The problem is, I have no idea why this variant doesn't work, while the other does. I hope this is general, and not depending on the specific test case.

@ceraolo what do you think?

ceraolo, 2023-05-16:
Good job, Francesco!

What I like of the solution with commutation from the alpha-beta to the constant impedance model is that it is not due to numerical reasons, but, instead, beause it simulates more closely the actual behaviour of real-life load aggregates. The fact that this also solves numerics is just an additional advantage.

In fact, as I already mentioned:
- power electronics loads cannot absorb their nominal power at low voltage levels, since it would require to absorb huge currens. Typically they have overload capability up to 100%, which in general terms means they cannot absorb the rated power, even for limited time, below V=0.5pu. Below that value they can disconnect or absorb some current, which however does not correspond to the P=VI=const pattern (at most I can be controlled to stay contant at its maximum allowed value for low voltages).
- motors are often protected by undervoltage relays, which trip at their voltage setpoints, typically at V=0.4-0.6pu
- the remaining loads behave as impedances.

As a consequence of this I would recommend to set (differently from my previous proposal) the default voltage at which the switch towards contant impedance takes place at V=0.5 p.u. 
Thanks again, @casella

casella, 2023-05-16:
> What I like of the solution with commutation from the alpha-beta to the constant impedance model is that it is not due to numerical reasons, but, instead, because it simulates more closely the actual behaviour of real-life load aggregates. The fact that this also solves numerics is just an additional advantage.

Indeed. The best models are the closest to the physical behaviour, that's one of my core beliefs as a research practitioner in the system modelling field 😃 

> In fact, as I already mentioned:
>
>     * power electronics loads cannot absorb their nominal power at low voltage levels, since it would require to absorb huge currens. Typically they have overload [...]

The third case is obviously covered quite accurately by my proposed change. However, if there is a significant amount of loads of the first two kinds in the aggregate load model, the current proposal will over-estimate the current. I am wondering whether we can further improve the proposal by considering, e.g. a fraction of the load which is completely disonnected below some threshold.

On the other hand, I understand that it is unlikely that these models will operate with a sustained voltage in the range 0.1-0.7 p.u., so I guess what really matters is the behaviour at very low voltage, which will end up being very low current anyway also with the current proposal. So maybe it's overkill to further detail it. Do you agree?

> As a consequence of this I would recommend to set (differently from my previous proposal) the default voltage at which the switch towards contant impedance takes place at V=0.5 p.u.

OK.

ceraolo, 2023-05-16:
_On the other hand, I understand that it is unlikely that these models will operate with a sustained voltage in the range 0.1-0.7 p.u., so I guess what really matters is the behaviour at very low voltage, which will end up being very low current anyway also with the current proposal. So maybe it's overkill to further detail it. Do you agree?_

I agree. It was my proposal, after all!
I explained the motivations for this in my previous messages.
If a user knows actual details of their (algebraic) loads to be simulated, e.g. some undervoltage trippings, they will implement this behaviour directly.

casella, 2023-05-16:
> If a user knows actual details of their (algebraic) loads to be simulated, e.g. some undervoltage trippings, they will implement this behaviour directly.

Yes. We should mention this explicitly in the revised component documentation.

@AndreaBartolini can you please update the two LoadPQVoltageDependence models in version 2.0.0 according to my final proposal? I can help writing the updated documentation.

ceraolo, 2023-05-16:
Maybe it is a good idea to add to the library an example with short circuit, possibly the one (or derived from the one) in this ticket.

AndreaBartolini, 2023-05-17:
I've implemented the modifications and the example, this is what I've obtained:
[image]

Maybe I've made some mistakes or I've misunderstood something in the discussion, please check the code on the branch `Issue#52_InvalidRootWhenSimulatingShortCircuitIEEE14`,
the example to be run is `PowerGrids.Examples.IEEE14bus.IEEE14busShort4`.

max-privato, 2023-05-17:
Your plot seems unacceptable for a few reasons. The most important is that
the final voltage is different from the initial one.

I would like to contribute with some analysis, but I don't know how to
access that branch. Can you, please, create a zip with it and send it to me
privately or post here?

AndreaBartolini, 2023-05-17:
I think the fast way is to send you the zip...

casella, 2023-05-17:
That's not what I got in my simulations, the voltage was returning to about 1 p.u. as soon as the fault was cleared.

casella, 2023-05-17:
@ceraolo, you can access that branch from [this link](https://github.com/PowerGrids/PowerGrids/tree/Issue%2352_InvalidRootWhenSimulatingShortCircuitIEEE14). Then, you can get a zip file from there.

AndreaBartolini, 2023-05-17:
As additional note, I've put in low-tension fixed impedance only the load4 connected to the bus that goes in fault mode, maybe that also the other loads need to be put in the same condition (from the discussion I understood that the problem was on the load directly connected to the fault bus).

casella, 2023-05-17:
@AndreaBartolini I checked out the PR branch and ran the simulation. This is what I got:
[image]
which is exactly what I got in my previous simulations with the modified 1.0.2 library. Not sure why you get a different result.
I'm using the latest nightly build, maybe you are using an older version of omc?

AndreaBartolini, 2023-05-17:
> That's not what I got in my simulations, the voltage was returning to about 1 p.u. as soon as the fault was cleared.

I cannot find the branch with your test, does it pushed in the repo?

casella, 2023-05-17:
The branch with my previous test is under my fork: https://github.com/casella/PowerGrids/tree/regularize-load

AndreaBartolini, 2023-05-17:
> @AndreaBartolini I checked out the PR branch and ran the simulation. This is what I got

I'll try with the last nightly buid...

AndreaBartolini, 2023-05-17:
I got the same result of before also with the latest nightly buid: 
OpenModelica v1.22.0-dev-111-g16dbe81bb0 (64-bit)

To obtain the right result I must set `--tearingMethod=minimalTearing`

[image]

I understood that the solution doesn't require it...

casella, 2023-05-17:
We worked a bit more on this, looks complicated.

The difference in behaviour was due to @AndreaBartolini having turned on "Evaluate all parameters". Of course that is a perfectly legitimate choice and it shouldn't kill the solution. What happened instead was that the current of _another_ load (load9) went to 4 pu, while it voltage converged to near zero. 

We then set the lowVoltageAsImpedance parameter in load9, and then the same problem took place in load14. Then we set the parameter in all loads (in fact, we used a default in the system model to do so) and at that point the problem became singular when the short is activate, no matter what we do (use the default solver vs use kinsol, evaluate vs. not evaluate parameters, use vs. not use tearing).

This is actually a very bad result. I was already suspicious when I noticed that we got a singular solution if we wrote the impedance explictly as v =  instead of i = ...

Per ora buco nell'acqua :(

casella, 2023-05-17:
Maybe the solution with all the impedances and the shorted bus has some other issues because of the nonlinear behaviour of the machines?

casella, 2023-05-17:
I thought that it would have been wise to use noEvent(port.VPu > VPuThr), but it turned out it was a bad idea. As we removed noEvent, we got a simulation running with all the loads having lowVoltageAsImpedance = true.

If you look at load14.VPu, it becomes clear what the problem was:
[image]

[image]

As you can see, when bus4 is shorted, due to the various line impedances, the voltage of load14 gets very close to VThr = 0.5, which means that the solution of the nonlinear system is very close to the sharp corner caused by the switch from PQ to Z behaviour (which is continous, but not smooth). Event handling can deal with it.

It would be better if we could avoid the corner point, but we have no idea how to do that.

@ceraolo maybe we finally found a solution. We'll sleep over it and come back to you soon.

ceraolo, 2023-05-18:
Huge work, guys last night!
and good advances, too.
The last plots are intriguing: is it possible that we got by chance exactly a case which has that slight "chattering" over V=0.5? Consider how we chose the threshold 0.5, and that the short impedance was chosen totally independently!

We'll see whether it is an incredible coincidence or not. Unfortunately, I have no time to put my hand on on this today, but... there is no need, I see that there are tons of good activity on this from you, @casella  and @AndreaBartolini. Thank you for your time and effort.

ceraolo, 2023-05-18:
The plots in this ticket's description were for R_short = 0.0 with OpenIpsl and 0.05  with PowerGrids.
The reason for this is that otherwise PowerGrids simulation would not proceed up to the end time.
The plot with 0.05 with OpenIpsl is as follows:

[image]

casella, 2023-05-18:
>The plot with 0.05 with OpenIpsl is as follows:

@ceraolo, In OpenIPSL means 0.05 _p.u._, whatever that means. In PowerGrids, it means 0.05 `Ohm`. I guess that's why you get a different behaviour in OpenIPSL, 0.05 p.u. is probably a lot more than 0.05 Ohm.

>The reason for this is that otherwise PowerGrids simulation would not proceed up to the end time.

I guess R = 0.0 should work now in PowerGrids. You can try that, or maybe 1e-4 or 1e-6 in case there is some singularity at zero.

ceraolo, 2023-05-18:
@casella, can you link the library where I can reproduce the plots in https://github.com/PowerGrids/PowerGrids/issues/52#issuecomment-1552221065?

ceraolo, 2023-05-19:
>Sure, we are pushing everything to this branch: 

I downloaded this a few minutes ago, but I did not get the plots as in https://github.com/PowerGrids/PowerGrids/issues/52#issuecomment-1552221065: I got just no convergence, probably because the SW has changed in the meanwhile.

So, I cannot replicate that behaviour.

AndreaBartolini, 2023-05-19:
The SW has not been changed, maybe we are running a different version of omc or the same whith different settings.

The version used to produces the plot in #52 is  the following nightly buid (you can check the version in the help->about menu):
OpenModelica v1.22.0-dev-111-g16dbe81bb0 (64-bit)

Can you please report the version have you installed?

ceraolo, 2023-05-19:
> Can you please report the version have you installed?

I have many.

Here's what I get with OM 1.22.0-dev111:
[image]

AndreaBartolini, 2023-05-19:
It seems that you are runnig an older version of the load model, please check that the operator `noEvent()` shown below has been removed from the `LoadPQVoltageDependence` model:

  ```
if noEvent(port.VPu > VPuThr) or not lowVoltageAsImpedance then
    port.P = PRef*U_URef^alpha;
    port.Q = QRef*U_URef^beta;
  else
    port.v = port.i/CM.conj(Complex(PRef*VPuThr^alpha, QRef*VPuThr^beta)/URef^2);
  end if;
```

ceraolo, 2023-05-19:
As I said I, used PowerGrids downloaded last night from the zip available in the link in https://github.com/PowerGrids/PowerGrids/issues/52#issuecomment-1552956283

I prefer not to hunt for changes. 
I'll try again when we have a link to SW able to reprudce the plots in https://github.com/PowerGrids/PowerGrids/issues/52#issuecomment-1552221065, otherwise I'll wait for further advances from you or @casella .

AndreaBartolini, 2023-05-19:
Not to change the code, only do a check to verify that the link of the commit that we have posted is the right one.

casella, 2023-05-19:
Weird. The same code that ran successfully two days ago and that I committed now hangs when load14.port.VPU = 0.5.

Maybe I had set some special flag, which is now lost when the session was closed.

casella, 2023-05-19:
Indeed, it turns out I had set `--tearingMethod=minimalTearing` to check if avoiding tearing would get us convergence in the case when some loads used the low-voltage impedance approximation and some didn't. Indeed, that did not work, but then I probably forgot to remove the flag, so it stayed active for the rest of the session.

If I set `--tearingMethod=minimalTearing`, the simulation works successfully when all loads are set to use an impedance approximation for VPu < 0.5. In fact, with this setting it also works when that feature is turned off and all loads use the PQ representaton throughout. This is the good news.

The bad news is that if I also set `--daeMode` (which is the only way to get efficient simulations when you have more than a handful of generators, because of the well known sparsity issue), it doesn't work 😞 

More specifically, if I use plain vanilla PQ loads, the simulation already fails at time = 1, when he short is activated, regardless of how I set `tearingMethod`. If I set the model to use low-voltage loads as impedance, the short is handled successfully, but then the simulation hangs when load14.VPu reaches the 0.5 threshold with the following error:
```
Simulation terminated due to too many, i.e. 20, event iterations. This could either indicate an inconsistent system
or an undersized limit of event iterations. The limit of event iterations can be specified using the runtime flag
 '–mei=<value>'.
```
i.e., chattering. I tried to increase the max number of events to 100, but that had no effect, so it's most likely chattering that is taking place.

We need to investigate further. @ceraolo, in the meantime you can get the working example with the annotation that sets the tearing method flag from the usual [link](https://github.com/PowerGrids/PowerGrids/commits/Issue%2352_InvalidRootWhenSimulatingShortCircuitIEEE14).

casella, 2023-05-19:
> The last plots are intriguing: is it possible that we got by chance exactly a case which has that slight "chattering" over V=0.5? Consider how we chose the threshold 0.5, and that the short impedance was chosen totally independently!

That was actually good luck, since this showed a potential problem of this solution 😅 

> Honestly, it seems to me that continuity of P and Q when we switch to constant impedance should not have caused those (even though small) steps on voltage. I would expect a change in _slope_ on the voltage behaviour, not in _value_ when we switch to constant impedance.

You are absolutely right, and I suspect this is the reason why we get chattering behaviour with --daeMode. Apparently, the switching law is not continuous as it should have been. I'll investigate more to understand what's going wrong there.

casella, 2023-05-19:
I turned off equidistant time grid to see the actual computed time steps of the solver
[image]
What is weird is that when VPu first reaches 0.5, it jumps to 0.51, but then the next time it reaches 0.5, it continues smoothly. Then, it jumps again when it reaches 0.5, and the next time it goes through it smoothly.

casella, 2023-05-19:
One more clue, when the voltage has its first jump at time 1.012, the active power drops

[image]

while when it crosses again VPu = 0.5 at time 1.31, it doesn't. Same thing in reverse later on when the voltage goes up again.

ceraolo, 2023-05-19:
So, these power steps are the possible culprit of voltage steps. Obviously they  should not be there.

I was curious to understand whether there is some OM issue involved in their presence. Therefore, I tried Dymola, but unfortunately found that no IEEE14bus PowerGrids model, except IEEE14BusPowerFLow, is able to initialise (and then run) in Dymola.

This is not good news, not only because it jeopardises a possible help for this ticket from Dymola, but also hints for some more general potential weakeness of the library.

casella, 2023-05-19:
BINGO 😄 

The expression for the equivalent impedance was wrong, it was a bit tricky. 

Now that we have corrected it, the discontinuities are gone, and so is the chattering. Disabling the tearing is recommended for faster and more robust performance in all cases (e.g. using daeMode, evaluating parameters), but other than that everything works fine.

You can check it from the usual [link](https://github.com/PowerGrids/PowerGrids/commits/Issue%2352_InvalidRootWhenSimulatingShortCircuitIEEE14).

PowerGrids is as strong as ever 😎

casella, 2023-05-20:
> I was curious to understand whether there is some OM issue involved in their presence. Therefore, I tried Dymola, but unfortunately found that no IEEE14bus PowerGrids model, except IEEE14BusPowerFLow, is able to initialise (and then run) in Dymola.

I tried that as well, and indeed what I found points out to some other possible issue with the library.

Dymola hangs because it cannot solve the initial equation for `GEN8.GEN.rQ2Pu`. That shouldn't be a big deal: it's simple nonlinear system with just one unknown.

The problem is, OMC manages to solve it, but it finds `GEN8.GEN.rQ2Pu = -0.035` which is obviously not a sensible solution.

I'll open another ticket about this. For sure we need to add some asserts and/or min attributes to all the parameters of synchronous machines that are found by solving nonlinear equations, to guard against bogus solution. We also need to investigate why do we get such a bogus solution: do the initial equations have multiple solutions and we converge to the wrong one, or is there something wrong with the initial equations themselves and/or with the machine parameters?
> 
> This is not good news, not only because it jeopardises a possible help for this ticket from Dymola, but also hints for some more general potential weakeness of the library.

AndreaBartolini, 2023-05-20:
Initialization fail in Dymola has been also analyzed in #56, and is related to the wrong alias propagation of some initial values. My hoping is that the solution proposed in #62 can solve the problem.

ceraolo, 2023-05-20:
> Initialization fail in Dymola has been also analyzed in https://github.com/PowerGrids/PowerGrids/issues/56,

Well, my memory tends to make strange things in recent years.
I myself had alread opened #56! So the inability of Dymola running these models was already well known...

ceraolo, 2023-05-20:
What happended to line terminals?

One of the two terminals now has disappeared, probaly as a consequence of some mistake made in PR #46.

Here's what is see on IEEE 14 bus, based on the current situation as available in this ticket's  link: https://github.com/PowerGrids/PowerGrids/commits/Issue%2352_InvalidRootWhenSimulatingShortCircuitIEEE14:
[image]
Hers's what I expected, based on my  PRs 68 and 47:
[image]

casella, 2023-05-20:
> Initialization fail in Dymola has been also analyzed in #56, and is related to the wrong alias propagation of some initial values. My hoping is that the solution proposed in #62 can solve the problem.

Problem (almost) solved, see #72. I will try to change the parameters of GEN8 in a sub branch of https://github.com/PowerGrids/PowerGrids/tree/Issue%2352_InvalidRootWhenSimulatingShortCircuitIEEE14 to see what Dymola does.

casella, 2023-05-20:
> What happended to line terminals?

Let's discuss this in a separate ticket #73, othewise this ticket will pretty soon become a complete mess 😅 
> One of the two terminals now has disappeared, probaly as a consequence of some mistake made in PR #46.
>
> Here's what is see on IEEE 14 bus, based on the current situation as available in this [...]

casella, 2023-05-20:
> > Initialization fail in Dymola has been also analyzed in #56, and is related to the wrong alias propagation of some initial values. My hoping is that the solution proposed in #62 can solve the problem.
> 
> Problem (almost) solved, see #72. I will try to change the parameters of GEN8 in a sub branch of https://github.com/PowerGrids/PowerGrids/tree/Issue%2352_InvalidRootWhenSimulatingShortCircuitIEEE14 to see what Dymola does. However, this change should be managed in a separate branch, otherwise we risk messing up everything.

casella, 2023-05-20:
OK, you can check https://github.com/PowerGrids/PowerGrids/tree/Issue%2352_updated_GEN8, which contains all the short circuit fixes we applied so far, plus updated parameters on GEN8. It works fine in OMC. Dymola 2023 handles it until the short is cleared at time = 1.2, then it fails.

We should try this with the latest version of Dymola, and if it doesn't work, submit a ticket to Dassault. I don't think this is a problem with the library, it's really a tool issue.

BTW Dymola has other problems, for example I can't see `load14.port.VPu`. Maybe once @AndreaBartolini is through with the main redesign of the connectors we should check what doesn't work in Dymola and submit a bunch of tickets to Dassault. But I would wait until the redesign is done, I don't think doing it now is wise.

[8 more replies]