PowerGrids/PowerGrids #53: terminal hides terminal()

max-privato, 2022-08-09:
I'm trying to create a model that at the end of simulation writes some data on disk.
To do this I would use, inside a PVBus model
```
when terminal()
...
end when;
```
Unfortunately, this is impossible, since terminal() is hid by terminal.
In Dymola I get:
Error: Attempting to use the variable terminal as if it were a function.
Could be a spelling error, the local variable hiding a function in an enclosing scope, or using () instead of [] for subscripts. 
Variable terminal declared at File: C:/Program Files/Dymola 2021/Modelica/Library/PowerGrids 1.0.2/Electrical/BaseClasses/OnePortAC.mo, line 21

I would use a different name for terminals in PowerGrids, e.g. PowerTerminal, or PGTerminal

casella, 2022-08-25:
You are right, we overlooked this aspect when designing the library. Unfortunately, changing this name will break existing models. Since we need to make a non-backwards compatible version to support MSL 4.0.0, we could take this opportunity to do that.

The original idea was to differentiate the electrical `Port` model (that contains all the interface variables and equations relating them) from the connector. We could simply call the connector class `PortConnector`, and the instance `portConnector`.

@max-privato what do you think?

AndreaBartolini, 2023-03-30:
we are going to solve this issue (see my proposed schedule in issue (https://github.com/PowerGrids/PowerGrids/issues/39)), so it is time to finalise the new name for the "Terminal".

In my opinion the proposed name "PowerTerminal" is the best choise

casella, 2023-05-08:
I wanted to propose to name the actual connectors `terminalAC` instead of `terminal,` but @AndreaBartolini prevented me with his PR 😅.

That would have the advantage of being shorter. Also, powerTerminal seems to convey the fact that the port carries power, whereas in fact it carries AC current and voltage.

I'm sorry for getting late with this proposal, what do you think?

In hindsight, maybe just having the instances named terminalAC with a small letter would have been better: shorter and without

AndreaBartolini, 2023-05-08:
At the end the PR is just a global find&replace with some arrangements regarding comments and descriptions... so it's not a big deal to change the name again...
The proposed name `terminalAC` has the advantage to specify that the terminal is of AC type, so if in the future the CC terminal will be introduced it can be accordingly named `terminalCC`.
Personally I don't like the underscore in the names, so my proposal is to use `terminalACA` and `terminalACB` for the two ports component, but this is a minor issue.

casella, 2023-05-08:
> The proposed name `terminalAC` has the advantage to specify that the terminal is of AC type, so if in the future the CC terminal will be introduced it can be accordingly named `terminalCC`. 

Good point, though I guess it will just be an alias of Modelica.Electrical.Analog.Interfaces.Pin.

> Personally I don't like the underscore in the names

Me neither, but ACA and ACB look really confusing to me, and even worse when we get CCA and CCB. The use of underscores for _a and _b ports is commonplace in the Modelica standard library, as it is used for electrical, mechanical, and thermal connectors. So, I guess that would be OK here.

ceraolo, 2023-05-08:
I also in general don't like underscore to separate words in names. For instance large_value is for me an old-style name for largeValue.

But for me compliance with MSL has a stronger impact. 
We already live (and probably will live forever) with flange_a and flange_b for mechanics. 

Therefore, I'm in favour of terminalAC_a (note the a lowercase for similarity with MSL) and terminalAC_b.

AndreaBartolini, 2023-05-09:
solved by PR #67

closed 2023-05-10 by AndreaBartolini

casella, 2023-05-10:
So far so good, nothing got broken, see [report](https://libraries.openmodelica.org/branches/master/PowerGrids_dev/PowerGrids_dev.html) 😃