PowerGrids/PowerGrids #54: Embedded Power Flow feature

casella, 2022-08-25:
In the original design of the PowerGrids library, the initialization of the system is based on power flow data provided at the port of components as parameters: amplitude and phase of the voltage, active and reactive power flows. The idea was that in general solving a power flow problem may require specialized software, possibly including features as optimization, and this was out of scope for Modelica models, which are mostly aimed at running transient simulations. This is true in particular for large systems, whose models may be automatically built from net-lists inside software that uses Modelica as a service to run the transient simulations but already has robust power-flow computation capabilities, that can be used to compute such initialization parameters. Hence, the library was designed to provide reliable initialization, _assuming that proper power-flow parameters are supplied_.

It turned out, for simple systems one can also use the PowerGrids library directly from a Modelica editor such as OMEdit or Dymola, e.g. for teaching or research purposes. For such simple systems, it is possible to build power flow models using PowerGrids, relying on the Modelica tool solver to obtain such power flows, and then use the results of their simulations, either manually, or via scripts, to provide the appropriate parameter values for the transient simulations. However, this is not particularly convenient; moreover, lack of properly computed initialization parameters can lead to initialization problem, or possibly to convergence to bogus initial conditions.

The alternative is to _optionally_ embed power flow models in the library, so they can be used automatically to compute the power flow initialization parameters. The idea is to use conditional components for this purpose: sub-connectors to carry the PF current and voltage can be added to the Terminal connector, and suitable conditional embedded power flow (EPF) models attached to these connectors can be conditionally instantiated inside each component, with provisions to compute the PF initialization parameters for the initialization of the full dynamic models during initialization.

A sketch of the implementation is reported here:

1. A Boolean parameter `computeEPF` (compute embedded power flow) is added to the system object.

2. The `TerminalAC` connector can be defined as:

```
connector TerminalAC "Terminal for phasor-based AC connections"
  Types.ComplexVoltage v "Phase-to-ground voltage phasor";
  flow Types.ComplexCurrent i "Line current phasor";
  parameter Boolean computeEPF "= true includes embedded power flow sub-connector";
  TerminalAC terminalEPF(final computeEPF = false) if computeEPF "Conditional sub-connector for emedded power flow computation";
end TerminalAC;
```
3. The `OnePortAC` model can then be expanded as follows:
```
partial model OnePortAC "Base class for AC components with one port"
  final parameter Boolean computeEPF = systemPowerGrids.computeEPF;
  PowerGrids.Interfaces.TerminalAC terminal
    (v(re(start = port.vStart.re), im(start = port.vStart.im)),
     i(re(start = port.iStart.re), im(start = port.iStart.im)),
     computeEPF = computeEPF,
     terminalEPF(v = vEPF, i = iEPF);
  replaceable OnePortAC componentEPF if compute EPF "Conditional embedded power flow component";

  parameter Types.Voltage UStart = if computeEPF then UstartEPF else UNom "Start value of phase-to-phase voltage phasor, absolute value";
  parameter Types.Angle UPhaseStart = if computeEPF then UPhaseStartEPF else 0 "Start value of phase-to-phase voltage phasor, phase angle";
  parameter Types.ActivePower PStart =  if computeEPF then PStartEPF else SNom "Start value of active power flowing into the port";
  parameter Types.ReactivePower QStart =  if computeEPF then QStartEPF else 0 "Start value of reactive power flowing into the port";

  parameter Types.Voltage UStartEPF(fixed = false) "Start value of phase-to-phase voltage phasor, absolute value, computed by the EPF";
  parameter Types.Angle UPhaseStartEPF(fixed = false) "Start value of phase-to-phase voltage phasor, phase angle, computed by the EPF";
  parameter Types.ActivePower PStartEPF(fixed = false) "Start value of active power flowing into the port, computed by the EPF";
  parameter Types.ReactivePower QStartEPF(fixed = false) "Start value of reactive power flowing into the port, computed by the EPF";

  Types.ComplexVoltage vEPF "Phase-to-ground voltage phasor of embedded power flow model";
  Types.ComplexCurrent iEPF "Line current phasor of embedded power flow model";
  ...
initial equation
  if computeEPF then
    // set values of initialization parameters based on EPF solution
    UStartEPF = CM.'abs'(vEPF)*sqrt(3);
    UPhaseStartEPF = CM.arg(vEPF);
    PStartEPF = 3*CM.real(vEPF*CM.conj(iEPF));
    QStartEPF = 3*CM.imag(vEPF*CM.conj(iEPF);
  else
    // set dummy values (not used)
    UStartEPF = 0;
    UPhaseStartEPF = 0;
    PStartEPF = 0;
    QStartEPF = 0;
  end if;
equation
  ...
  // set dummy values if the conditional sub-connectors with their binding equations are not activated
  if not computeEPF then
     vEPF = Complex(0) "Dummy value";
     iEPF = Complex(0) "Dummy value";
  connect(componentEPF.terminal, terminal.terminalEPF);
  ...
```

4. The `TwoPortAC` model can be handled similarly
5. Every time a new component is defined that extends the base `OnePortAC` or `TwoPortAC` models, the conditional `componentEPF` sub-component must be redeclared to the desired model for the EPF computation, and parameterized accordingly. For example, the model of a transmission line with breakers can use the model of a constant impedance line as an EPF model, of course setting `computeEPF = false` for it to avoid infinite recursion:
```
model LineConstantImpedanceWithBreakers "Transmission line with constant impedance and breakers"
  extends BaseClasses.PiNetwork(
    redeclare LineConstantImpedance componentEPF(
      computeEPF = false,
      R = R, X = X, G = G, B = B,
      UNomA = UNom, UNomB= UNomA, SNom = SNom),
    UNomA = UNom, UNomB = UNomA,);
  parameter Types.Voltage UNom(start = 400e3) "Nominal/rated voltage";
  parameter Types.Resistance R "Series resistance";
  parameter Types.Resistance X "Series reactance";
  parameter Types.Conductance G = 0 "Shunt conductance";
  parameter Types.Conductance B = 0 "Shunt susceptance";
  ...
```

If `computeEPF = false`, there is actually no runtime overhead compared to the current architecture: the conditional EPF components and sub-connectors are not instantiated, and the dummy variables are set to zero once and for all during initialization. Start values for the PF can be set from some upstream source, or computed by default based on the nominal parameters, exactly as in the current architecture.

If `computeEPF = true`, a "mirror" system made by the `componentEPF` sub-components of the instantiated components, connected in the same way the main components are, is automatically built by the Modelica compiler when flattening the model. This sub-system does not depend on the main component equations, so it can be solved independently. In fact, since the power flow models have all constant inputs (e.g. PQ or PV values), the corresponding sub-system can be solved once and for all at initialization. 

The dependency analysis of the Modelica compiler then automatically finds out that the parameters providing the start values for the main component equations depend on the solution of the EPF sub-system. In this way, the generated code automatically provides the right sequence of computations: first the EPF problem is solved, then the steady-state initialization problem for the transient model is solved, using start values obtained from the EPF model, and finally the dynamic simulation is carried out from the found initial steady-state.

Also note that most of the modifications to the existing library are found in the `OnePortAC` and `TwoPortAC` base classes. The impact on the actual model definitions is just the need to define the model for the EPF computation via a redeclare statement, using existing components. There is no need to add extra equations anywhere, when developing new component models with this enhancement.

casella, 2022-08-25:
@max-privato I'd like to hear your comments

max-privato, 2022-08-26:
@casella having this EPF feature would be great!

I've used PowerGrids for teaching last semester and found the present way of combining PF and transient very clumsy. What is time-consuming and error-prone is:
1. the need to create two different models for PF and transient
2.  the need to manually copy selected PF outputs from the PF run to the transient, with a good degree of precision.
3. the need to transfer manually the numbers from PF to Transient for every PF run, ever and ever again.

This month I've spent some time generating a solution that solves issues 2 and 3. This saves users the majority of time: with the enhancement I added, once they have created the two models, running PF and transients in a row is straightforward. 

It works as I'll explain below. In the examples shown, for transients, I've converted PQ nodes into constant impedance ones, as is very often done. This has the additional advantage to avoid severe convergence issues that arise when we simulate short circuits in a network containing constant P and Q nodes.

This conversion is not mandatory. I've prepared examples of other types, e.g. with constant impedance loads in both the PF and the transient, but I've not included them here.

My examples are in the following package:
[PFTdemo.mo.txt](https://github.com/PowerGrids/PowerGrids/files/9432724/PFTdemo.mo.txt)

To run you should  use Dymola and not OpenModelica (OM is expected to be able to run them when tickets  #9301, #9302, #9303 on https://github.com/OpenModelica/OpenModelica/issues are fixed)

I've made my tests with Dymola 2011, since it uses as default MSL 3.2.3, and PowerGrids is based on MSL 3.2.3 (I tried also to use a version of it converted into 4.0.0, but it had some issues, so, for the time being, I stuck with 3.2.3).

As for Powergrids, the version available from the following link can be used, which has the nice feature of showing numerical values of PF output on diagrams, as shown in the picture included in this ticket: 
[PowerGrids 1.0.2.zip](https://github.com/PowerGrids/PowerGrids/files/9432725/PowerGrids.1.0.2.zip)

PFTdemo contains the following simulation files:
1. PowerFlow3, which contains a version of PowerGrids.Examples.Tutorial.GridOperation.Static.PowerFlow enhanced in order to send the PF output to disk, so that it can be retrieved from the following transient simulation
2. StaticGrid3, which contains a version of PowerGrids.Examples.Tutorial.GridOperation.Static.StaticGrid enhanced in order to retrieve the PF output from disk, and take it as initial conditions for transient
3. netPF_PQconst, which contains a larger network for power flow, in which loads are considered constant-P, constant-Q
4. netUmach1, which contains the same network used for the previous bullet, which takes the PF output from disk, uses them as initial conditions, then runs a simulation. 

Note that the chaining PF->T (i.e., PowerFlow->Transient) is made so that loads in the example in the last two bullets are ConstP-constQ in PF, and const impedance in transient.

For reference here's the output of the last point 3, directly shown on the network diagram:
[image]

and here's the voltage on bus 1, showing the perfect correspondence of PF voltage with initial transient voltage, and the effect on this voltage of the following short circuit on bus 1.

[image]

**********************************************
The technique I used to create automatic transfer of PF output into transient initial conditions is just to show the feasibility, it is not (yet) for inclusion in any library.
For instance, the correspondence between saved (from PF) and retrieved (by T) information is based on the instance name of the saving object, which must be equal to the one of the retrieving one. This can cause undesired effects in rare cases. These effects, however, are avoided if the user runs a transient following the corresponding PF (and not following PF related to other systems).

Needless to say, having and Embedded PowerFlow would be much cleaner and more useful. 
Once it works, maybe it is to be evaluated how to change the load node types: as I said, usually they are constP-constQ for PF, constZ (or alpha-beta) for transient. 

One possibility is to use alpha-beta dependent loads in both PF and transient, with the possibility to have different values for PF and transient.

AndreaBartolini, 2023-05-27:
> ```modelica
> connector TerminalAC "Terminal for phasor-based AC connections"
>   Types.ComplexVoltage v "Phase-to-ground voltage phasor";
>   flow Types.ComplexCurrent i "Line current phasor";
>   parameter Boolean computeEPF "= true includes embedded power flow sub-connector";
>   TerminalAC terminalEPF(final computeEPF = false) if computeEPF "Conditional sub-connector for emedded power flow computation";
> end TerminalAC;
> ```

It seems to me that there is a recursive declaration of `TerminalAC` in this connector.... Am I wrong?

casella, 2023-05-27:
Of course it is recursive, but setting `computeEF = false` stops the recursion after one iteration. We need to check if OMC supports that, but I don't think this is forbidden by the specification. Only infinite recursion is, obviously.

casella, 2023-05-27:
In fact, the MLS [Section 5.6.1.3](https://specification.modelica.org/maint/3.5/scoping-name-lookup-and-flattening.html#the-instantiation-procedure) specifically says that instantiation is a _recursive_ procedure, so it should be OK.

ceraolo, 2023-06-14:
The implementation proposed by @casella, in comparison with the one in https://github.com/PowerGrids/PowerGrids/issues/54#issuecomment-1228387962,  has the advantage of not requiring that the user creates different models for PowerFlow and transient.
However it has the advantage that all transients require some CPU time is spent to do PowerFlow, even when they start from the same initial state. The solution in https://github.com/PowerGrids/PowerGrids/issues/54#issuecomment-1228387962 does not have this disadvantage.

A solution which would save all the advantages is to implement @casella's solution, but adding two flags: "save PF result" and "reuse PF result" which could put and retrieve data from disk respectively, e.g. using what implemented in https://github.com/PowerGrids/PowerGrids/issues/54#issuecomment-1228387962.
For simplicity I would not add the possibility to save  different states for the same model. Just the latest one.

casella, 2023-06-15:
@ceraolo, I don't think this is really necessary. 

Please check the results of the [OSMC CI report on PowerGrids](https://libraries.openmodelica.org/branches/master/PowerGrids/PowerGrids.html), in particular those of the largest model contained there, the IEEE14Bus model. Computing the power flow takes 20 ms, including all the dead times, see [sim](https://libraries.openmodelica.org/branches/master/PowerGrids/files/PowerGrids_PowerGrids.Examples.IEEE14bus.IEEE14busPowerFlow.sim) report. The entire simulation of a load transient requires 795 steps of the IDA solver, and takes 1.57 s, see [report](https://libraries.openmodelica.org/branches/master/PowerGrids/files/PowerGrids_PowerGrids.Examples.IEEE14bus.IEEE14busLoad2Variation.sim). It seems to me that the time spent for the computation of the power flow, about 1% of the total simulation time, is totally negligible.

If you think about it, the solution of the power flow requires to solve the grid equations (lines, transformers, loads) with trivial models of the machines (PV sources). The computation of one simulation step by the IDA solver needs to solve the same network equations, plus the differential-algebraic equaitons of the machines and of their controllers. In a typical simulation you initialize once, but then you typically need hundreds (or thousands) of time steps to compute a transient. It is true that getting the initial power flow to converge will require a lot more iterations than those who are required at each time step; the solution of the new step is initialized with some smart predictor-corrector strategy, so it typically only requires 2-3 iterations to converge, while the power flow could need a lot more, maybe even 50 or a 100. But yet, you do initialize only once.

If you want to see this on a larger model, you can check the largest currently running example of [ScalableTestGrids](https://libraries.openmodelica.org/branches/heavy_tests/ScalableTestGrids_AnalyticJacobian/ScalableTestGrids_AnalyticJacobian.html), a library of scalable grid models built with PowerGrids, namely Type1_N_6_M_4. This model has 72 generators, 144 transformers, 684 lines and 288 loads. The system is initialized with a trivial power flow (all voltages with 1 p.u.  and zero phase), so the initialization resembles the solution of a power flow, that starts from the same initial guess. As you can see in the [report](https://libraries.openmodelica.org/branches/heavy_tests/ScalableTestGrids_AnalyticJacobian/files/ScalableTestGrids_AnalyticJacobian_ScalableTestGrids.Models.Type1.Type1_N_6_M_4.sim), initialization takes 163 ms, while simulation overall takes 7.7 s. In this case, the time spent for computing the exact power flow (with the full machine models, not with PV sources) is 2% of the total.

In fact, it could be that at the end of the day, loading the power flow from a file could actually take more. Besides, it's always annoying to have to locate the file.

casella, 2023-06-15:
In my 20+ yrs of experience with Modelica tools, I always found that the benefit of having models which can reliably initialize in steady state out of the box is substantial. In particular, every time you change some parameter of the model, the saved steady-state/power-flow becomes obsolete, and you need to worry about that. If it's recomputed all the time, you just go ahead.

IMO only if the embedded initialization time took more than 10-20% of the total simulation time, then it would be worth considering such a solution.

BTW, in OpenModelica it is always possible to load the result of a previous simulation and use it to set the initial start values of a simulation, without the need of programming this feature explicitly in the library, see the [user's guide](https://openmodelica.org/doc/OpenModelicaUsersGuide/latest/solving.html#importing-initial-values-from-previous-simulations). Dymola has a similar feature (Import Initial), and so I guess do all other Modelica tools.

But again, even with fairly large models, the time taken to locate the file on the file system will probably be much larger than the time spent recomputing the power flow every time.

ceraolo, 2023-06-16:
> But again, even with fairly large models, the time taken to locate the file on the file system will probably be much larger than the time spent recomputing the power flow every time.

My implementation does not require any time to locate any file on the file system.

casella, 2023-06-16:
You are right, I misunderstood that. But I still think that it might take more time to do the name-based processing than to solve the PF problem, which is solved real fast.

ceraolo, 2023-06-16:
> which is solved real fast.

In ths case, I obviously agree that saving and retrieving numbers would be useless (or worse).
They are, instead, very , very useful with the present version of PowerGrids, which requires manually transferring them from PF to transient studies.

I did not make any scientific tests. I just wrote https://github.com/PowerGrids/PowerGrids/issues/54#issuecomment-1592081458 having expecenced lengthy PF solutions. But maybe those long times were due to suboptimal initial conditions, possibly due to misalignments of the initial conditions of buses and components connected to them.

AndreaBartolini, 2023-07-10:
I'm starting to implement this feature, in the following my proposal for the `componentEPF` models:

- generators --> `PowerFlow.PVbus`,
- loads --> `PowerFlow.PQBus`  (or `PowerFlow.ZLoad` ?, what is the purpose of ZLoad model in the power-flow scenario? There are no examples in PowerGrids (other than the test of the component),
- grid or reference bus --> `PowerFlow.SlackBus`,
-  lines --> `Branches.LineConstantImpedance`,
- transformers --> `TransformerFixedRatio`,

**any other componento or bus is left unchanged**

Please give a feedback if something should be added/changed...

casella, 2023-07-10:
>     * loads --> `PowerFlow.PQBus`  (or `PowerFlow.ZLoad` ?, what is the purpose of ZLoad model in the power-flow scenario? There are no examples in PowerGrids (other than the test of the component),

Good question. If you were actually running the power flow because you are iterested in it, PQ would be mandatory. However, for the sake of providing good enough start values (which need not be very precise to ensure convergence), maybe ZLoad with Z computed based on the nominal voltage could be a good idea, because we then get rid of a lot of nonlinearity.

@ceraolo what do you think?

ceraolo, 2023-07-10:
Maybe we could use
- LoadImpedancePQ ->  Z 
- LoadPQVoltageDependence ->PQ (in the future maibe also Pn(V/Vn)^alpha and Qn(V/Vn)^beta?)

I'll check with my colleagues what's their experience on commercial PF programs.

ceraolo, 2023-07-11:
I've received a couple of answers, that I summarise here:

Colleague 1: Digsilent allows alpa - beta formulations. Constant impedance modelling in AT is rare, because actual load agregates behave nearer to constant power

Colleague 2: Commercial power flow programs allow alpha-beta formulations. A simple power flow app developped for Matlab by Power System Engineering Research Center (PSERC) allows for each node a constant P-Q and a constant Z (mixing the two one could simulate some alpha-beta pair, up to 2).

So, I would say that:
- commercial programs allow generic alpha-beta pairs
- scripts and applications for general-purpose SW able to perform PF studies may well accept just constant P-Q and constant Z: mixing them the user can reach an equivalent alpha-beta behaviour.

So, I would confirm my comment in https://github.com/PowerGrids/PowerGrids/issues/54#issuecomment-1629351185.

Implementation of alpha-beta is not important for now, maybe it would come in the future, in case PowerGrids is further developped.

AndreaBartolini, 2023-07-11:
do we want to allow te user to change the PF component to be used? If so, the implementation structure is a bit different, and it is better to decide the strategy before to implement the feature.

casella, 2023-07-13:
@ceraolo @AndreaBartolini maybe there was a misunderstanding here. The option of using an impedance for PQ loads is only meant to find an initial guess for the Newton solver, close enough to the solution to ensure the convergence of the steady-state initialization problem. Which _will contain PQ models, possibly with alpha and beta_. 

As long as the voltages on all the buses is close to 1 p.u., which is nomally the case at the initialization point, I believe using PQ or Z for the guess values will be the same. I guess we'll have to do some testing.

casella, 2023-07-13:
> do we want to allow te user to change the PF component to be used? If so, the implementation structure is a bit different, and it is better to decide the strategy before to implement the feature.

If we could make them replaceable, I think it would be great. Unless that complicates the design substantially.

AndreaBartolini, 2023-07-14:
@casella  
We have to decide how manage the components made by using a sub-network, like the `LineConstantImpedanceFault`

[image]

in this case the PF components are already instantated by the single components of the sub-network, so the  `TwoPortAC` interface of the `LineConstantImpedanceFault` should participate in the PF only to made the proper connections. 

One solution could be to force the flag` computeEPF = false` by a modifier where the interface is extended but this is not applicable because the `computeEPF` flag is declared as `final`, an this is correct because the embedded PF mechanism must be not accessible from the user.

An alternative solution could be declare the `computeEPF` flag as `(non final) constant`, so that it is not shown in the parameters window and only expert users can access it. This solution is more flexible but less robust.

I cannot imagine other alternatives, any suggestion?

casella, 2023-07-18:
After some discussion between @casella and @AndreaBartolini, we updated the proposal at the beginning of this ticket accordingly, by adding the new hasSubEPF parameter to OnePortAC. Of course, it should also be similarly added to TwoPortAC.

AndreaBartolini, 2023-07-21:
It seems that the recursion cannot be properly managed by both OpenModelica and Dymola.

According to the proposal, this should be the new base connector for PowerGrids terminals (the "base" connector has been introduced with the propagation of the start values):

```modelica
connector TerminalACBase
  Types.ComplexVoltage v "Phase-to-ground voltage phasor";
  flow Types.ComplexCurrent i "Line current phasor";
  parameter Boolean computeEPF "=true, includes embedded power flow sub-connetor";
  TerminalACBase terminalEPF(final computeEPF = false) if computeEPF "conditional sub-connector for EPF";
  Types.ReferenceAngularVelocityPu omegaRefPu "Reference angular speed in PerUnit";
end TerminalACBase;

```
When you try to save the new connector OpenModelica crashes.

Dymola allows to write and save the new connector but if you check it the following error is shown:
```
Check of [PowerGrids.Interfaces.TerminalACBase](modelica://PowerGrids.Interfaces.TerminalACBase):
Recursion limit reached in instantiation of class Real.
Probably caused by mutually dependent classes.
Context: Real
```

Trying to open the modified connector in OMEDit causes the OpenModelica crash.

At the end, it seems that the proposed strategy is not applicable ...

AndreaBartolini, 2023-07-21:
If I right remember, I had a similar issue in the past and the cause was that the conditional classes are first instantiated and then removed... 
If this is the instantiation mechanism, it is no possible to stop the recursion...

casella, 2023-09-15:
Regarding the recursion, I was mistaken. The MLS says that conditional components are first instantiated and then removed, so this makes the conditional recursion a pipe dream.

The solution is to have separate classes for the "naked" components that will be used to compute the embedded power flow and another one that extends it for the "wrapped" components that contain the power flow sub-component. This holds for the ports (terminalAC) and for the base classes of components (OnePortAC, TwoPortAC, etc.).

casella, 2023-09-15:
Here is a sketch of the new design
```modelica
package TestRecursion

  partial connector TerminalACVI "Bare base class for AC connectors"
    Types.ComplexVoltage v "Phase-to-ground voltage phasor";
    flow Types.ComplexCurrent i "Line current phasor";
  end TerminalACVI;
  
  connector TerminalACPF "To be used for embedded power flow computations"
    extends TerminalACVI;
  end TerminalACPF;

  connector TerminalACBase "Base class for connectors of components including optional embedded power flow"
    extends TerminalACVI;
    Types.ReferenceAngularVelocityPu omegaRefPu "Reference angular speed in PerUnit";
    TerminalACPF terminalPF "Connector for embedded power flow";
  end TerminalACBase;

  connector TerminalACBus "Base class for bus connectors"
    extends TerminalACBase;
    output Types.Voltage UStart "Start value of phase-to-phase voltage";
    output Types.Angle UPhaseStart "Start value of phase-to-phase voltage angle";  
  end TerminalACBus;

  connector TerminalAC "Base class for buses"
    extends TerminalACBase;
    input Types.Voltage UStart "Start value of phase-to-phase voltage";
    input Types.Angle UPhaseStart "Start value of phase-to-phase voltage angle";  
  end TerminalAC;

  partial model OnePortACBase
    replaceable TerminalACVI terminalAC;
    parameter Boolean computePF = true "= true to activate the embedded power flow";
    // all the monkey stuff here only once
  end OnePortACBase;

  partial model OnePortACPF
    extends OnePortACBase(
      redeclare TerminalACPF terminalAC);
  end OnePortACPF;

  partial model OnePortAC
    extends OnePortACBase(
      redeclare TerminalAC terminalAC);
    replaceable OnePortACPF componentPF if computePF;
  end OnePortAC;

  partial model OnePortACBus
    extends OnePortACBase(
      redeclare TerminalAC terminalAC);
    replaceable OnePortACPF componentPF if computePF;
  end OnePortACBus;

  model InfiniteBus
    extends OnePortACBus(redeclare InfiniteBusPF componentPF);
  end InfiniteBus;

  model InfiniteBusPF
    extends OnePortACPF;
  end InfiniteBusPF;

  model Test
    InfiniteBus bus annotation(
      Placement(visible = true, transformation(origin = {12, -10}, extent = {{-10, -10}, {10, 10}}, rotation = 0)));
  end Test;

package Types "Domain-specific type definitions"
  extends Modelica.Icons.TypesPackage;
  import Modelica.Units.SI;
  type Voltage = SI.Voltage(nominal = 1e4, displayUnit = "kV");
  type Current = SI.Current(nominal = 1e4, displayUnit = "kA");
  operator record ComplexVoltage = SI.ComplexVoltage(re(nominal = 1e4, displayUnit="kV"), im(nominal = 1e4, displayUnit="kV"));
  operator record ComplexCurrent = SI.ComplexCurrent(re(nominal = 1e4, displayUnit="kA"), im(nominal = 1e4, displayUnit="kA"));
  type Angle = SI.Angle;
  type PerUnit = SI.PerUnit;
  type ReferenceAngularVelocityPu
    extends PerUnit;
    function equalityConstraint
      input ReferenceAngularVelocityPu omegaPu1;
      input ReferenceAngularVelocityPu omegaPu2;
      output PerUnit residue[0];
    algorithm
      assert(abs(omegaPu1-omegaPu2)<1e-10, "inconsistent omega reference across the branch");
    end equalityConstraint;
  end ReferenceAngularVelocityPu;
end Types;

end TestRecursion;
```

casella, 2023-09-15:
One potential issue is how to deal with the modifiers for the start values, which are different for the PF models and for the regular models. One solution could be to define computePF in the OnePortACVI base class, and then set it to false in OnePortACPF and to true in OnePortAC and OnePortACBus. Unfortunately, this doesn't work with the multiple inheritance such as in
```modelica
  model InfiniteBusPF
    extends OnePortACPF;
    extends InfiniteBusVI(redeclare TerminalACPF terminalAC);
  end InfiniteBusPF;

  model InfiniteBus
    extends OnePortACBus(redeclare InfiniteBusPF componentPF);
    extends InfiniteBusVI(redeclare TerminalAC terminalAC);
  end InfiniteBus;
```
because the first extends would inherit either computePF = true in the first model and computePF= false in the second, while the second extends would inherit computePF without any default, causing a conflict because the two declarations are not identical. I guess the only reasonable way to handle that is to set computePF = true by default in the OnePortACVI base class, and then reset it to final computePF = false in each and every PF component. This is slightly annoying, because you have to do several times instead of doing it once in the base class, but it's a small price to pay to avoid the need of duplicating the PF components.

casella, 2023-09-15:
This is the updated sketch:
```modelica
package PowerGridsSketchPF

  partial connector TerminalACVI "Bare base class for AC connectors"
    Types.ComplexVoltage v "Phase-to-ground voltage phasor";
    flow Types.ComplexCurrent i "Line current phasor";
  end TerminalACVI;
  
  connector TerminalACPF "To be used for embedded power flow computations"
    extends TerminalACVI;
  end TerminalACPF;

  connector TerminalACBase "Base class for connectors of components including optional embedded power flow"
    extends TerminalACVI;
    Types.ReferenceAngularVelocityPu omegaRefPu "Reference angular speed in PerUnit";
    TerminalACPF terminalACPF "Connector for embedded power flow";
  end TerminalACBase;

  connector TerminalACBus "Base class for bus connectors"
    extends TerminalACBase;
    output Types.Voltage UStart "Start value of phase-to-phase voltage";
    output Types.Angle UPhaseStart "Start value of phase-to-phase voltage angle";  
  end TerminalACBus;

  connector TerminalAC "Base class for buses"
    extends TerminalACBase;
    input Types.Voltage UStart "Start value of phase-to-phase voltage";
    input Types.Angle UPhaseStart "Start value of phase-to-phase voltage angle";  
  end TerminalAC;

  partial model OnePortACVI
    replaceable TerminalACVI terminalAC;
    parameter Boolean computePF = true "= true computes the start value with the embedded power flow";
    // all the monkey stuff here only once
  end OnePortACVI;

  partial model OnePortACPF
    extends OnePortACVI(
      redeclare TerminalACPF terminalAC,
      final computePF = false);
  end OnePortACPF;

  partial model OnePortAC
    extends OnePortACVI(
      redeclare TerminalAC terminalAC);
    replaceable OnePortACPF componentPF if computePF;
  equation
    connect(terminalAC.terminalACPF, componentPF.terminalAC);
  end OnePortAC;

  partial model OnePortACBus
    extends OnePortACVI(
      redeclare TerminalACBus terminalAC);
    replaceable OnePortACPF componentPF if computePF;
  equation
    connect(terminalAC.terminalACPF, componentPF.terminalAC);
  end OnePortACBus;
  
  model InfiniteBusVI
    extends OnePortACVI;
  equation
    terminalAC.v = Complex(1);
  end InfiniteBusVI;

  model InfiniteBusPF
    extends OnePortACPF;
    extends InfiniteBusVI(redeclare TerminalACPF terminalAC,
                          final computePF = false); // this modifier is needed in each ***PF model, can't be put in the base class
  end InfiniteBusPF;

  model InfiniteBus
    extends OnePortACBus(redeclare InfiniteBusPF componentPF);
    extends InfiniteBusVI(redeclare TerminalAC terminalAC);
  end InfiniteBus;

  model Test
    InfiniteBus bus annotation(
      Placement(visible = true, transformation(origin = {12, -10}, extent = {{-10, -10}, {10, 10}}, rotation = 0)));
  end Test;

package Types "Domain-specific type definitions"
  extends Modelica.Icons.TypesPackage;
  import Modelica.Units.SI;
  type Voltage = SI.Voltage(nominal = 1e4, displayUnit = "kV");
  type Current = SI.Current(nominal = 1e4, displayUnit = "kA");
  operator record ComplexVoltage = SI.ComplexVoltage(re(nominal = 1e4, displayUnit="kV"), im(nominal = 1e4, displayUnit="kV"));
  operator record ComplexCurrent = SI.ComplexCurrent(re(nominal = 1e4, displayUnit="kA"), im(nominal = 1e4, displayUnit="kA"));
  type Angle = SI.Angle;
  type PerUnit = SI.PerUnit;
  type ReferenceAngularVelocityPu
    extends PerUnit;
    function equalityConstraint
      input ReferenceAngularVelocityPu omegaPu1;
      input ReferenceAngularVelocityPu omegaPu2;
      output PerUnit residue[0];
    algorithm
      assert(abs(omegaPu1-omegaPu2)<1e-10, "inconsistent omega reference across the branch");
    end equalityConstraint;
  end ReferenceAngularVelocityPu;
end Types;

end PowerGridsSketchPF;```

casella, 2023-09-25:
After some discussion with @AndreaBartolini I recognized that there was a fundamental flaw in the design sketched earlier in this ticket. The question is: what are, ultimately, the inputs of the (embedded) power flow? And the answer is: P and Q of the loads, P and U of the generators.

For the case of loads, P and Q are either determined by PRefConst and QRefConst, or by some external input or binding equation expression (the variation with the voltage is neglected in this simplified power flow). We can define two parameters PPF and QPF defaulting to PRefConst and QRefConst, that will be passed to P and Q of the embedded power flow load component. The default works out of the box if the constants are used, otherwise one needs to type in explicitly the initial value provided by the binding expression or input. This is done for safety, because not necessarily these inputs are known functions of time; if one wants to model demand-side management strategies, they could actually depend on whatever signal, that may cause ugly algebraic loops. 

For the generators, we can set U = UNom, and then we need a parameter PPF (default SNom) that will need to be set by the user. Also in this case, it is not possible to use the initial value of the input Pm of the machine model, because that may depend on complicated control logic.

For all other power flow components (lines, transformers), we set UStart = UNom, but PStart = 0 and QStart = 0 as defaults. PStart = SNom is highly questionable. BTW, lines and transformers are linear, so their start values are ultimately irrelevant for convergence of the Newton-Raphson method.

AndreaBartolini, 2023-09-27:
During the implementation of the proposed solution I removed the terminalACPF if the embedded PF is not computed.
This because if the embedded PF is not computed the variables vPF and iPF are forced to Complex(0) in each xxPortAC interface with the result that each xxPortAC interface connected to the same PF connection set (generated by the terminalACPF) generates  four eaquations (vPF=Complex(0) and iPF=Complex(0)) that involve the same pair of complex variables in the connection set, thus generating more equations than necessary. The result is that the overall model becomes unbalanced.
The problem can be easly solved by removing the PF connection sets (that is by removing the terminalACPF) if the embedded PF is not computed.

casella, 2024-02-13:
After some iterations, the feature seems to be working just fine. Closing the issue.

closed 2024-02-13 by casella