PowerGrids/PowerGrids #56: IEEE14 models do not run on Dymola 2022 (Power Grids 1.0.2)

ceraolo, 2023-02-06:
**When I try to run IEEE14busStaticNetwork I get the following error message:**

```
dymosim started
... "PowerGrids.Examples.IEEE14bus.IEEE14busStaticNetwork" simulating
... "dsin.txt" loading (dymosim input file)
... initial variable "_dummy" in file "dsin.txt" ignored (not in model).
... initial variable "_derdummy" in file "dsin.txt" ignored (not in model).
... initial variable "bus1.port.S.re" in file "dsin.txt" ignored (not in model).
... initial variable "bus1.port.S.im" in file "dsin.txt" ignored (not in model).
... initial variable "bus1.U" in file "dsin.txt" ignored (not in model).
... initial variable "bus1.theta" in file "dsin.txt" ignored (not in model).
... initial variable "Load2.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load2.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load3.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load3.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load4.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load4.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load5.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load5.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load6.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load6.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load9.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load9.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load10.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load10.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load11.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load11.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load12.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load12.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load13.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load13.Q" in file "dsin.txt" ignored (not in model).
... initial variable "Load14.P" in file "dsin.txt" ignored (not in model).
... initial variable "Load14.Q" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.UNom" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.SNom" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.portVariablesPhases" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.portVariablesPu" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.generatorConvention" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.localInit" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.UStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.UPhaseStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.PStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.QStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.SNom" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.UNom" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.SBase" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.UBase" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.PStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.QStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.UStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.UPhaseStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.portVariablesPhases" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.portVariablesPu" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.generatorConvention" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.VBase" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.IBase" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.VStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.IStart" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.vStart.re" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.vStart.im" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.iStart.re" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.iStart.im" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.u.re" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.u.im" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.S.im" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.I" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.PPu" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.QPu" in file "dsin.txt" ignored (not in model).
... initial variable "GEN1.port.vPu.re" in file "dsin.txt" ignored (not in model).
[... 241 lines cut ...]
ERROR: The simulation of PowerGrids.Examples.IEEE14bus.IEEE14busGen2Disconnection FAILED
```

**When I try to run IEEE14busLine1to5Opening I get the following error message:**
```
Warning: Failed to solve nonlinear system using Newton solver.
  During initialization at time: 0
  Tag: initialization.nonlinear[5]
  For debugging information enable
  Simulation/Setup/Debug/Nonlinear solver diagnostics/Detailed logging of failed nonlinear solutions.
Trying to solve non-linear system using global homotopy-method.

Warning: Failed to solve nonlinear system using Newton solver.
  During initialization at time: 0
  Tag: initialization.nonlinear[6]
Trying to solve non-linear system using global homotopy-method.

Warning: Failed to solve nonlinear system using Newton solver.
  During initialization at time: 0
  Tag: initialization.nonlinear[6]
Error: Integrator failed to start model.

ERROR: The simulation of PowerGrids.Examples.IEEE14bus.IEEE14busLine1to5Opening FAILED

```

casella, 2023-02-10:
@ceraolo, @AndreaBartolini will have a look at it. It may be due to the fact that Dymola doesn't use homotopy by default, so it may be getting another (invalid) solution to the initialization problem, and then getting stuck when trying to start. We'll investigate and report ASAP.

AndreaBartolini, 2023-02-13:
I tried in Dymola 2023 with the flag 

`Advanced.OnlyUseHomotopyMethod = true`

but the simulation fails again with the following message:

```
Log-file of program ./dymosim
(generated: Mon Feb 13 09:55:36 2023)

dymosim started
... "PowerGrids.Examples.IEEE14bus.IEEE14busStaticNetwork" simulating
... "dsin.txt" loading (dymosim input file)
Solving using global homotopy-method.

Warning: Failed to solve nonlinear system using Newton solver.
  During initialization at time: 0
  Tag: initialization.nonlinear[9]
  For debugging information enable
  Simulation/Setup/Debug/Nonlinear solver diagnostics/Detailed logging of failed nonlinear solutions.
Error: could not solve simplified initialization for homotopy method.
Error: Integrator failed to start model.
... "IEEE14busStaticNetwork.mat" creating (simulation result file)

... Error message from dymosim

ERROR: The simulation of PowerGrids.Examples.IEEE14bus.IEEE14busStaticNetwork FAILED
```

Please note that the following warnings where rised during the translation phase:

```
Some variables are iteration variables of the initialization problem:
but they are not given any explicit start values. Zero will be used.

Iteration variables:
GEN1.AVR.limiterWithLag.u
GEN1.GOV.errPu.y
GEN2.AVR.limiterWithLag.u
GEN2.GOV.errPu.y
GEN3.AVR.limiterWithLag.u
GEN6.AVR.limiterWithLag.u
GEN8.AVR.limiterWithLag.u
```
```
Conflicting start values

The iteration variable Tgen2.terminalA.v.im has been selected to have the guess value 0.0.
However, the start value has been selected from a set of alias variables having conflicting start values of the same precedence.
0.0, the start value of Tgen2.portA.v.im given as Tgen2.portA.vStart.im. At level 3. Original start-value at level 3.
-1215.0242427144797, the start value of GEN2.GEN.terminal.v.im given as GEN2.GEN.port.vStart.im. At level 3. Original start-value at level 3.

The iteration variable Tgen2.terminalA.v.re has been selected to have the guess value 13856.40646055102.
However, the start value has been selected from a set of alias variables having conflicting start values of the same precedence.
13856.40646055102, the start value of Tgen2.portA.v.re given as Tgen2.portA.vStart.re. At level 3. Original start-value at level 3.
14475.692325514157, the start value of GEN2.GEN.terminal.v.re given as GEN2.GEN.port.vStart.re. At level 3. Original start-value at level 3.

The iteration variable bus14.terminal.v.im has been selected to have the guess value 0.0.
However, the start value has been selected from a set of alias variables having conflicting start values of the same precedence.
0.0, the start value of bus14.terminal.v.im given as bus14.port.vStart.im. At level 2. Original start-value at level 2.
-2284.9924732185527, the start value of Load14.terminal.v.im given as Load14.port.vStart.im. At level 2. Original start-value at level 2.
0.0, the start value of bus14.port.v.im given as bus14.port.vStart.im. At level 3. Original start-value at level 3.
0.0, the start value of L9to14.portB.v.im given as L9to14.portB.vStart.im. At level 3. Original start-value at level 3.
0.0, the start value of L13to14.portB.v.im given as L13to14.portB.vStart.im. At level 3. Original start-value at level 3.

The iteration variable bus13.terminal.v.im has been selected to have the guess value 0.0.
However, the start value has been selected from a set of alias variables having conflicting start values of the same precedence.
0.0, the start value of bus13.terminal.v.im given as bus13.port.vStart.im. At level 2. Original start-value at level 2.
-2194.9432677173977, the start value of Load13.terminal.v.im given as Load13.port.vStart.im. At level 2. Original start-value at level 2.
0.0, the start value of bus13.port.v.im given as bus13.port.vStart.im. At level 3. Original start-value at level 3.
[... 159 lines cut ...]
GEN2.GEN.lambdaDPu
GEN2.GEN.lambdafPu
GEN2.GEN.lambdaQ1Pu
GEN2.GEN.lambdaQ2Pu
GEN2.GEN.omegaPu
GEN2.GEN.theta
GEN3.GEN.lambdaDPu
GEN3.GEN.lambdafPu
GEN3.GEN.lambdaQ1Pu
GEN3.GEN.lambdaQ2Pu
GEN3.GEN.omegaPu
GEN3.GEN.theta
GEN6.GEN.lambdaDPu
GEN6.GEN.lambdafPu
GEN6.GEN.lambdaQ1Pu
GEN6.GEN.lambdaQ2Pu
GEN6.GEN.omegaPu
GEN6.GEN.theta
GEN8.GEN.lambdaDPu
GEN8.GEN.lambdafPu
GEN8.GEN.lambdaQ1Pu
GEN8.GEN.lambdaQ2Pu
GEN8.GEN.omegaPu
GEN8.GEN.theta
```

casella, 2023-02-13:
@ceraolo the library is designed so that all potential iteration variables get "good enough" initial guess values, computed from the voltage magnitude and phase, active and reactive power start values set at all ports, based on the results of the power flow computation. This seemed to work fine in OpenModelica. We can double-check if it also works fine in Dymola. Did you try smaller examples and did they work?

ceraolo, 2023-02-13:
Yes, I have working small examples.

I just want to evidence in this ticket that one of the examples included in the library does not work with Dymola. I think  that having everything working with Dymola, which is acknowledged as something linke a "reference" tool, is an importaant target for any library.
There might be users models not having good enough intiaial guesses that do not converge, but at least the provided examples of this specific library  I imagine should work with both OM and Dymola.

If it is difficult for the library programmers to make a provided example run well with Dymola, it can be easily imagined what can occur to ordinary users, or even students, with their own models.

casella, 2023-02-14:
Agreed, we should figure out why it doesn't work and if it is due to some flaws in the library or in the tool. Then, the course of action will of course be different based on that assessment :)

AndreaBartolini, 2023-03-24:
The problem seems to be related to the selection of the start value in case of alias elimination.
For example, the start value of `GEN2.GEN.terminal.v.im` conficts with the one of the connected terminal `Tgen2.terminalA.v.im` and dymola says:

```
The iteration variable Tgen2.terminalA.v.im has been selected to have the guess value 0.0.
However, the start value has been selected from a set of alias variables having conflicting start values of the 
same precedence.

0.0, the start value of Tgen2.portA.v.im given as Tgen2.portA.vStart.im. At level 3. 
Original start-value at level 3.

-1215.0242427144797, the start value of GEN2.GEN.terminal.v.im given as GEN2.GEN.port.vStart.im. At level 3. 
Original start-value at level 3.
```

The start value of the terminal `GEN2.GEN.terminal.v.im` is set  in the instance of `PortAC` at level 3,  by means of the modifiers `UStart = 25.1608e3` and `UPhaseStart = -0.0837392` that are set at top level, where the GEN2 is instantiated (level 1):

Level 1:
```
  PowerGrids.Examples.IEEE14bus.ControlledGeneratorIEEE GEN2(
    GEN(  
      ...
      UStart = 25.1608e3,
      UPhaseStart = -0.0837392
    ));
```

Level 3:
```
model PortAC "AC port computing auxiliary quantities"
 ...
 parameter Types.Voltage UStart = UNom "Start value of phase-to-phase voltage modulus";
 parameter Types.Angle UPhaseStart = 0 "Start value of voltage phase";
 ...
 final parameter Types.ComplexVoltage vStart = CM.fromPolar(UStart/sqrt(3), UPhaseStart)
 ...
 InputComplexVoltage v(
        re(nominal = VBase, start = vStart.re),
        im(nominal = VBase, start = vStart.im)) "Port voltage (line-to-neutral)";

 ...
```
For some reason (maybe it is correct but it is not completely clear for me in the Modelica specification) dymola considers the confidence number = 3 for the start value calculated by GEN2, also if it depends on the parameters `Ustart `and `UPhaseStart` set at Level 1.

By this way, the start value `-1215.0242427144797` calculated by the `GEN2 `and the default value `0.0` in the `Tgen2` have the same confidence number and dymola in this case chooses the wrong one (Murphy law...).

@casella : can you help me to undestand if the calculation of the confidence number is according to the Modelica specification? If this is the case, the problem is structural, because the hierarchy of the classes from top to terminal level is the same in a lot of components in PowerGrids.

casella, 2023-05-20:
See #72. Some parameters of GEN8 are invalid (they lead to negative `GEN8.GEN.rQ2Pu`), if they are fixed, Dymola 2023 is indeed able to run the IEEE14bus model.

ceraolo, 2023-05-30:
Should this now be closed?

AndreaBartolini, 2023-06-01:
Regarding Dymola, discussion has been modved to #72

closed 2023-06-01 by AndreaBartolini