PowerGrids/PowerGrids #58: Confusing info in Infinitebus

ceraolo, 2023-04-12:
In PowerGrids.Electrical.Buses.InfiniteBus i read the following info:
_Infinite bus model with constant voltage e and internal impedance Z. The port voltage is v = e + Zi, where i is the current entering the bus. The default value of the series impedance Z = R + jX is zero._

From this it is unclear how to specify "e". the use of lowercase letters and the change of internal e.m.f. from, "E" to Uref in the info do not help. After some study one understands that this it is to be specified by means of parameters URef and theta. I propose to modify the info things clearer, 

A possible substite for the info could be (if this is accepted parameter URef should be changed into ERef and theta into thetaE):

_Infinite bus model with constant voltage phasor E and internal impedance Z. The port voltage is U = E + ZI, where I is the current entering the bus and E has as amplutude parameter ERef, and phase thetaE. The default value of the internal impedance Z = R + jX is zero._

casella, 2023-04-27:
This issue reminds me a lecture of my electrical engineering professor, who said that three-phase AC with phasors was just as one-phase, with some sqrt(3) terms here and there 😅. Luckily, he did not mention per-unit at the time, we were still using good ole SI units 😄.

We have to be very careful not to cause confusion. _Phase-to-phase_ voltage (e.g. 400 V for low-voltage 3-phase AC) is normally used to define the nominal or rated value (including this case!), but line impedances are referred to _phase_to_ground_ voltage instead. If you are considering a 400 kV EHV grid, it would be odd if you had to put in ERef = 230 kV instead of URef = 400 kV. I would expect that many (most?) users won't really read the instructions, and just put in the wrong value. 

In order to avoid confusion, we clearly defined both voltages in PowerGrids (see [Naming Conventions](https://build.openmodelica.org/Documentation/PowerGrids.UsersGuide.NamingConventions.html)) for the _model equations_, so you can use whichever is most convenient for your specific model - conversion factors are built-in the base port model. On the other hand, we decided to always ask for phase-to-phase voltage values for _parameters_, because those are the values that people normally refer to.

For the record:
- `u` is the RMS phase-to-phase voltage phasor (complex)
- `v` is the RMS phase-to-ground voltage phasor (complex), `u = sqrt(3)*v`
- `U` is the absolute value of the RMS phase_to_phase voltage phasor (real)
- `V` is the absolute value of the RMS phase_to_ground voltage phasor (real) `U = sqrt(3)*V`
- `i` is the current phasor, indicating the RMS value of the current flowing in each of the three conductors (complex)
- `I` is the absolute value of the RMS conductor current (real)
- `Z = R + j*X` is the impedance of a _single_ phase conductor, hence `v = Z*i`
- The complex power absorbed by each phase is thus `S =  v*conj(i)`
- The total power absorbed by the three-phase device is three times that (one for each phase),
- Hence, `S = P + j*Q = 3*v*conj(i) = 3*Z*i*conj(i) = 3*Z*I = 3*(R*I + j*X*I)`
- Since `v = u/sqrt(3)`, it is also possible to compute `S = P + j*Q = sqrt(3)*u*conj(i)`
- It is also possible to compute `S = P + j*Q = 3*v*conj(i) = 3*Z*(v/Z)*conj(v/Z) = 3*Z*V^2/abs(Z)^2 = U^2*Z/abs(Z)^2`. In this case, using the phase-to-phase voltage gets you rid of the 3 or sqrt(3) factors. But, in any case, `Z` always refers to the impedance of a single phase.

That being said, I fully agree that the current explanation is incomplete and potentially misleading, because it does not explain what is the relationship between `e` and the model parameters/inputs. That becomes very clear if you look at the code, which is self-documenting by design:
```modelica
model InfiniteBus "Infinite bus with internal impedance"
  extends BaseClasses.BusBase(
      e = CM.fromPolar(ERef, theta),
      Z = Complex(R, X));
  extends Icons.Bus;
  parameter Types.Resistance R = 0 "Series resistance";
  parameter Types.Reactance X = 0 "Series reactance";
  parameter Types.Angle theta = 0 "Voltage phase angle of ideal voltage generator";
  parameter Types.Voltage URef = UNom "Phase-to-phase voltage of ideal voltage generator";
  final parameter Types.Voltage ERef = URef/sqrt(3) "Phase-to-ground voltage of ideal voltage generator";
end InfiniteBus;
```
but of course people should not be forced to look at the source code to figure out how a component works. Or should they? That's another story 😄

So, I would propose the following text:

> Infinite bus model with constant voltage and internal impedance. The phase-to-ground port voltage is v = e + Zi, where i is the current entering each phase of the bus, Z is the impedance of each phase, and e is a constant reference voltage. The value of the reference voltage is determined by the phase-to-phase voltage magnitude URef and phase theta. The default value of the series impedance Z = R + jX is zero, corresponding to an ideal, fixed-voltage strong node.

For consistency, we should probably change `theta` into `URefPhase` in the code and in the documentation.

We may also add
> The complex power entering the bus is `S = P + j*Q = 3*Z*I = URef^2*Z/abs(Z)^2`

@ceraolo, what do you think?

ceraolo, 2023-04-27:
It is my guilt. Had I read the _naming conventions_ I probably would not have opened this ticket! 
So, your new formulation is better, but also the old one was good enough for someone having read the naming conventions part of the docu.

Regarding changing theta into URefPhase I'm not sure. From the code we see that theta is the angle of the phase-to-neutral (or phase-to-ground, as you call it) voltage "e", which in turn is the phasor representing the first of the three phases. Instead URefPhase, having "U" in its name maybe makes people think to the phase of the line-to-line voltage (which is different from the phase-to-neutral phase)? 

Regarding your proposal for S, I frankly don't understand what you mean.  Your mentioning Z, which is the internal impedance of the infinite bus, confuses me.
In general for every port I would say just that s = P + j*Q = 3*v*conj(i)  (here I used lowercase for complex values as for your naming conventions), but this:
1. is (should be, see point 2) valid is for section of the system, in particular every port, and does not say anything special for this case 
2. is not valid in actual examples (I don't know why)

A final note on infinite bus, even though hardly related to this ticket.
The InfiniteBus icon is identical to the Bus's. For me this is not advisable: different models should have different icons. Bus, ReferenceBus, InfiniteBusVariableVoltage, BusFault have different icons, as they should. But Bus and InfiniteBus do not, which is not advisable.

casella, 2023-04-27:
> It is my guilt. Had I read the _naming conventions_ I probably would not have opened this ticket! So, your new formulation is better, but also the old one was good enough for someone having read the naming conventions part of the docu.

😄 

That is precisely my point. Nobody reads the manuals. Not even the professors. So, the interface should be as fool-proof as possible 😅. This is the true meaning of Murphy's law (who was an aeronautic engineer): if something can be misused, it most certainly will (hence, try to avoid putting it there in the first place, when designing the thing).
 
> Regarding changing theta into URefPhase I'm not sure. From the code we see that theta is the angle of the phase-to-neutral (or phase-to-ground, as you call it) voltage "e", which in turn is the [...]

Good point. If I remember correctly, the phase angle between phase-to-phase and phase-to-ground is 60°. We could actually put it there, but eventually it will be useless: as long as all parameters refer to phase-to-phase, what matters are the relative angles, not the absolute ones. That's why, if I remember well what I read in the textbooks, this phase angle is often neglected.  What really matters is the sqrt(3) factor in the magnitude.

But anyway, if we keep theta like it is today, there is no ambiguity. If there is only one infinite bus, it will always be left to its default of zero (and any default would produce the same results in terms of P's and Q's). If there are two or more, what matters is their phase difference, not their absolute value.

> Regarding your proposal for S, I frankly don't understand what you mean. Your mentioning Z, which is the internal impedance of the infinite bus, confuses me. In general for every port I would say [...]

You are right. In fact, what I wrote holds for the case of impedance loads (i.e. e = 0). Forget about it.

> A final note on infinite bus, even though hardly related to this ticket. The InfiniteBus icon is identical to the Bus's. For me this is not advisable: different models should have different icons. [...]

Absolutely. Based on your experience with other software for power system simulation, how would you suggest to differentiate them?

ceraolo, 2023-04-27:
> Absolutely. Based on your experience with other software for power system simulation, how would you suggest to differentiate them?

Indeed the perception of buses for power engineers is something similar to nodes in circuits. AFAIK, never something that has internal generation.
_I find it queer that you've created a bus with internal power generation._
It seems to me that InfiniteBus and InfiniteBusVariableVoltage are more grid equivalents than buses.
Were in  your feet I would consider for all three models (InfiniteBus, InfiniteBusVariableVoltage, EquivalentGrid), icons recalling grids, with the possibility to reduce them to two!
Indeed, both EquivalentGrid and InfiniteBus model an internal generator in series with an impedance, the only difference being in the way parameters are shown in masks. Maybe They could be named EquilvalentGrid1 and EquivalentGrid2? Or maybe InfiniteBus can be removed from the library?
Also, I would expect for InfiniteBusVariableVoltage an icon and name nearer to a grid than a bus. 
So, speaking freely, it seems to me that the three components could in principle reduced to 2:
- a first grid model with constant internal EMF (derived from present InfiniteBus and/or EquivalentGrid)
- a second grid model (derived from present InfiniteBusVariableVoltage) allowing variable EMF.

casella, 2023-05-03:
So it seems that the fundamental issue is to name "Bus" something that can actually generate power.

However, it seems that the concept of "Infinite Bus" meaning an ideal voltage/frequency source is [pretty commonplace](https://www.google.com/search?client=firefox-b-d&q=infinite+bus+electrical). And also its graphical representation as a horizontal thick line is similar to the "regular" bus. On the other hand, it seems that it normally has a zero internal impedance, maybe that option should be reserved to an "Equivalent Grid" model?

ceraolo, 2023-05-04:
You're right. The most authoritative source of info for teminology is electropedia, from IEC, and it confirms the term:
https://www.electropedia.org/iev/iev.nsf/display?openform&ievref=603-02-30

However, it is a totally different concept from an ordinary bus, which just connects stuff.
I would recommend a totally different icon, possibly much nearer to an equivalent grid icon than a bus's.

But again: an Infinite bus (according to electropedia) is an "Equivalent grid" with zero internal impedance. So the best solution from the users' point of view, for me, is to have just a component named "equivalent grid", allowed to have infinite short circuit power (so implementing also an infinite bus).

If unifying the two components creates problems to the programmers, as a second choice I would recommend:
1. radically change the infinite bus icon making it nearer to an equivalent grid than a bus
2. avoiding allowing an infinite bus to have an internal impedance (which would conflict with the name and be a duplicate of equivalent grid) 

Still, I don't see any valid reason to use an "infinite bus" component when we have an "Equivalent grid", in case which does the same when its internal imp

casella, 2023-05-04:
> However, it is a totally different concept from an ordinary bus, which just connects stuff. I would recommend a totally different icon, possibly much nearer to an equivalent grid icon than a bus's.

This sounds reasonable.

> But again: an Infinite bus (according to electropedia) is an "Equivalent grid" with zero internal impedance. So the best solution from the users' point of view, for me, is to have just a component [...]

I guess it's mostly a matter of names and definitions.

As I understand, the "infinite bus" is an abstract and very simple model: fixed voltage and frequency. It doesn't need to be an equivalent grid: if you were using PowerGrids to model a local distribution network, it could actually be a diesel generator. I also agree that adding internal impedance to it was actually overkill, is probably confusing and should be removed. If you really want that, just connect a transmission line to it. Ockham's razor.

The equivalent grid model tries to be a bit more sophisticated in the description of the connection to an actual grid, by adding some non-ideal behaviour, which is specified by the SSC, c, and R_X parameters instead of directly specifying R and X. 

> If unifying the two components creates problems to the programmers, as a second choice I would recommend:
> 
>     1. change the infinite bus icon, e.g. making it nearer to an equivalent grid than a bus
> 
>     2. avoiding allowing an infinite bus to have an internal impedance (which would conflict with the name and be a duplicate of equivalent grid)

I would propose to keep the InfiniteBus model, removing the internal impedance, and adding an infinity symbol to the thick bar so that it is clearly distiguished from the regular bus

Maybe we should hear RTE's comments on that.

casella, 2024-10-10:
@marcochiaramello, @rosiereflo, @joyelfeghali last chance to comment on this, I'm going to make a PR soon