PowerGrids/PowerGrids #72: Issue with machine parameters in IEEE14

casella, 2023-05-20:
Simulate `PowerGrids.Examples.IEEE14bus.IEEE14busStaticNetwork`. 

With OpenModelica, one gets `GEN8.GEN.rQ2Pu  =  -0.035367765`. Dymola actually fails during initialization while trying to compute GEN8.GEN.rQ2Pu.

Action 1: we should add asserts and/or min/max attributes to all the machine parameters that are found by means of nonlinear initial equations, to guard against bogus solutions.

Action 2: we should figure out if there is something wrong with the initial equations / parameter values / start values that hampers the convergence to a sensible value of that parameter.

casella, 2023-05-20:
Keepint @ceraolo in the loop.

casella, 2023-05-20:
In fact, all those parameters [do have a min attribute](https://github.com/PowerGrids/PowerGrids/blob/325f2ae69d9fcc7793bc6ae261d298c4954718aa/PowerGrids/Electrical/Machines/SynchronousMachine4Windings.mo#L17), we had already thought about that back then. The problem is, OpenModelica ignores it. See OpenModelica/OpenModelica#10446.

casella, 2023-05-20:
I checked the equations for rQ2Pu and LQ2Pu further, and they seem to me to be linear, see #10712. Not sure why both OMC and Dymola see them as nonlinear. In fact, Dymola uses tearing on them, making them significantly nonlinear, and then probably tries to stay above the min limit. OMC instead doesn't use tearing (because we disabled it), and therefore immediately converges to the `rQ2Pu = -0.03` solution.

I don't remember where we took the parameters for the generators when we built this example (maybe @adriguir can remind us); maybe they were copied from iPSL?

Anyway, it seems to me that the set of external parameters for GEN8 has some problems, which leads to invalid internal parameters. We should try to figure out where these parameters come from and fix them, so that a proper solution is found.

casella, 2023-05-20:
In the meantime I copied the machine parameters (not the nominal power) of GEN6 into GEN8 to avoid the negative resistance and I managed to get the IEEE14bus model to also run in Dymola, which will help with #52.

ceraolo, 2023-05-25:
I 've always thought that the swicth towards constant impedance we chose, was the right solution for this ticket.

Now I've found this text from te DigSilent which confirms our approach:
https://www.digsilent.de/en/faq-reader-powerfactory/how-do-i-model-correctly-the-load-behaviour-for-dynamic-simulations.html

In particular it says:
 _If the voltage drops below the limit, the load will have again a constant impedance characteristic. This is necessary, because constant power leads to very large currents in case of very low voltages._

casella, 2024-02-13:
Fixed by acda6e1

closed 2024-02-13 by casella