PowerGrids/PowerGrids #73: One of the line terminals has disappeared

casella, 2023-05-20:
@ceraolo reports:

What happended to line terminals?

One of the two terminals now has disappeared, probaly as a consequence of some mistake made in PR #46.

Here's what is see on IEEE 14 bus, based on the current situation as available in this ticket's  link: https://github.com/PowerGrids/PowerGrids/commits/Issue%2352_InvalidRootWhenSimulatingShortCircuitIEEE14:
[image]
Hers's what I expected, based on my  PRs 68 and 47:
[image]

casella, 2023-05-20:
> One of the two terminals now has disappeared, probaly as a consequence of some mistake made in PR #46.

It's actually PR #68, PR #46 was abandoned.

AndreaBartolini, 2023-05-21:
Strange behavior... it seems to be an OM problem. Herebelow what I see on my PC:
Without new instance API:
[image]

With new instance API enabled:
[image]

This my actual position on GIT repo:
[image]

@casella to implement the feature I've used the following annotations:
```modelica
 extends TerminalAC annotation(IconMap(primitivesVisible=false), DiagramMap(primitivesVisible=false));
```
that as far as I know still have some problems... https://github.com/OpenModelica/OpenModelica/issues/5610

AndreaBartolini, 2023-05-21:
With lower zoom level...
[image]

[image]

AndreaBartolini, 2023-05-21:
Here my OM version:
OpenModelica v1.22.0-dev-111-g16dbe81bb0 (64-bit)

ceraolo, 2023-05-21:
Maybe it is an OM 1.20.0 issue.
OM 1.20.0 => terminals a missing
OM 1.22.0-dev-111=> terminals a visible
Dymola => terminals a  visible 
SImulationX => terminals a visible

closed 2023-05-22 by AndreaBartolini

casella, 2023-05-22:
Regarding OMC, I would suggest to only test the library with the new instance-based interface. This is what will be used in the future, the old interface is going to be retired in 1.22.0. If there is anything wrong with the rendering with the new instance-based interface, please open tickets so we can get it fixed ASAP.