PowerGrids/PowerGrids #75: Use different symbols for different classes

ceraolo, 2023-05-24:
LoadImpedancePQ and LoadPQVoltageDependence inside Electrical.Loads share the same symbols, which is inconvenient.
I propose making a change so that they are different from each other.  Maybe LoadImpedancePQ can have a "Z" in its symbol, while LoadPQVoltageDependence a greek alpha?

casella, 2023-05-24:
I'm fine with the letter Z for impedance loads, maybe we can leave the "regular" PQ load without any additional symbol, I understand PQ loads are the default choice in power system studies.

ceraolo, 2023-05-24:
> I understand PQ loads are the default choice in power system studies.
> Not sure. Consider the load in ENTSOE model

this is what I wrote an year ago on #54:

`for transients, I've converted PQ nodes into constant impedance ones, as is very often done. This has the additional advantage to avoid severe convergence issues that arise when we simulate short circuits in a network containing constant P and Q nodes.`

So, I had the impression that during transients studies constant impedance is more the norm than the exception. 
Curious enough, this was also something that laid ground for the solution of  #52 :-)

casella, 2023-05-25:
I wonder if that won't make the diagrams too cluttered. I mean, you don't see the parameter names and much other information in the icon. Maybe seeing a transformer is enough?

ceraolo, 2023-05-25:
I also share the same doubt.
It would be an exception to the general rule that different models should have different icons. Have you seen any such exceptions anywhere?

Maybe one could make some attenpts. for instance TapChanger is usually indicated as an arrow; the three tapChanger models could have different arrows.

But I'm ok also with keeping the same symbol for all transformers.

casella, 2023-05-25:
OK. If there are standards, we should use them. You can also check what tools like Digsilent do.

ceraolo, 2023-05-25:
It may also exist some redundancy. 
For instance it seems to me that defining both TapChangerTarget and TapChargerInterval is overkill: there is just a tiny difference in the input parameters. Does PowerGrids really need both of them?

casella, 2023-05-25:
I think this was a requirement from RTE.

ceraolo, 2023-05-25:
I've checked with Italian CEI 3-18 that should be equivalent to IEC 60617
There I found a generic symbol for "transformer with variable coupling" which means tap-charger transformer.
I could not find (there, nor everywhere else) any symbol for a phase-shifting transformer.

Furthermore I don't see any reason to differentiate between TransformerWithTapChangerInterval and TransformerWithTapChangerTarget, as well as  TransformerWithPhaseShifterInterval and TransformerWithPhaseShifterTarget. 
If these models are not unified in pairs, for them I don't see any need to distinguish icons (for me they are the same model).

So, the icons to be defined should be five:
1. TransformerFixedRatioWithBreaker
2. TransformerWithTapChangerInterval and TransformerWIthTapChargerTarget
3. TransformerWIthTapChangerMax
4. TransformerWithPhaseShifterInterval and TransformerWithPhaseShifterTarget
5. TransformerWithPhaseShifterMax.

We should investigate for a suitable PhaseShift graphical indication.
Provisionally, the full set of transformers can appear as in the following screenshot.
[image]

ceraolo, 2024-12-05:
In the same domain of this ticket I found (following a hint of a PhD student of mine) another issue.
In PowerGrids 1.02 we used to have different icons for controlled generators and synchronous condensers in IEEE14bus network (models for transients). Now, instead, we have the same symbol, which does not ease understandability.
I would say this is a regression.

AndreaBartolini, 2026-10-29:
The PR #215 implements what is described above, particularly:
- a label has been added to the Load icon in order to distinguish the load type ("Z" for the Impedance loads, "PQ" for the PQ loads), this label is static, so it doesn't show the alpha and beta values due to the poor space available around the icon,
- in the version 2.0.0 the PF and Dynamic connectors are no longer type compatible, an error is shown by OMEdit if the user tries to connect them together,
- Transformers icons are implemented according to the figure above in this ticket,
- Generator icons are now different for the generator, the controlled generator and the synchronous condenser.

Regarding the breakers, it is very difficult to represent them in the icons due to the reduced icons dimensions, in any case they become visible through the breaker input when the breaker is used.

closed 2026-10-29 by AndreaBartolini

ceraolo, 2026-10-29:
> * a label has been added to the Load icon in order to distinguish the load type ("Z" for the Impedance loads, "PQ" for the PQ loads), this label is static, so it doesn't show the alpha and beta values due to the poor space available around the icon,

I do not understand how you distinguish between contant P-Q and voltage dependence trhough alpha. That's why I proposed to use an alpha in the symbol in case of voltage dependence

AndreaBartolini, 2026-10-29:
I agree with you on this point, I've substituted the "PQ" label with "alpa" and "beta" values in the variable voltage load.

Hoping that:
- the area occupied by the icon is acceptable,
- the unicode characters for the Greek symbols are correctly visualized on both Sysop (Linux and win). I tried with OpenModelica on Linux and Dymola on WIn and it seems that the rendering is correct in both of them.