PowerGrids/PowerGrids #76: TestFreeOffset not working

ceraolo, 2023-05-25:
I tried Controls.Test.TestFreeOffset with Dymola 2020x (because it is distributed with MSL 2.3.2) and OM (1.21.0-dev-148 ans 1.22.0-dev-39).
Screenshot from Dymola:
[image]

Screehsot from OM
[image]

casella, 2023-05-25:
@ceraolo I tested it with OMC, and it also complains about initialization issues:
```
The initialization problem is inconsistent due to the following equation: 0 != 1 = 1.0 - freeOffset1.y
```
I'm not sure why we didn't add an experiment annotation, so these models did not end in the [OMC CI testing report](https://libraries.openmodelica.org/branches/master/PowerGrids/PowerGrids.html), but that was obviously a bad mistake 😅 

@AndreaBartolini we should review all those tests and add the experiment annotation to them so they get tested by the CI.

AndreaBartolini, 2026-08-20:
There was two mistakes here:
- the `system`object was not instantiated,
- consequently, the `initOpt` was the default one and the _offset_ was kept fixed, so the additional initial equation was redundant and the initial problems was inconsistent.

AndreaBartolini, 2026-08-20:
Fixed by PR #195

closed 2026-08-20 by AndreaBartolini