PowerGrids/PowerGrids #77: Boxed components should not be transparent.

ceraolo, 2023-05-25:
Consider the following screenshot.
As you see, all MSL boxed components are not transparent, while many Powergrids' are.
Having transparent boxed components makes a bit more difficult to drag them with mosue movement.

I would recomment to fill all of them with white (I can issue a PR in case I'm requested to do so).
Also generator (and, possibly, transformer) circles, for me, should be filled with white.
Doing this for transformer is just slightly trickier, since it involves for one winding using two overlapped circles, the lower filled, the uper transparent.

[image]

casella, 2023-05-25:
Good point, agreed on my side.

AndreaBartolini, 2023-05-25:
I agree with the proposed change, and I suggest to do this at the end of the current work, together with the reshape of the diagrams in order to cover the entire blank field.

AndreaBartolini, 2024-02-22:
Implemented by PR #134

closed 2024-02-22 by AndreaBartolini