PowerGrids/PowerGrids #78: Why in PortAC we hide pu variables when portVariablesPu=true?

ceraolo, 2023-05-27:
Consider  #47.
It used to work with Dymola 2020, while does not with 2023.
After some analysis I found that the reason is that in PortAC Vpu, Ppu, Qpu, have HideResult=portVariablesPu. So, when I choose to add per-unit variables to model, I get HideResult=true.
Is this intended? Maybe it should be the opposite?

To make #47 work also with Dymola 2023 I had to write (in my personal versin of PowerGrids)  HideResult=not portVariablesPu, which appears reasonable to me.

AndreaBartolini, 2023-06-03:
It can be implemented with the same approach used in #51, that works also in OpenModelica.

casella, 2023-06-03:
Beware that voltage magnitude and phase are used by initial equations, so they can't really be made conditional.

AndreaBartolini, 2023-06-03:
> Beware that voltage magnitude and phase are used by initial equations, so they can't really be made conditional.

So I can re-apply the previous implementation just for this two, leaving the others conditional.

AndreaBartolini, 2023-06-03:
or is it preferable to roll back to the previous solution?

casella, 2023-06-03:
Nah, sorry, I think I was mistaken, that issue pops up in other grid models I wrote for testing large-scale models. 

In PowerGrids, the phases are computed only for start attributes, but then in OnePortAC the voltage phase is implicitly considered in initial equations such as:
```
    port.v.re*port.vStart.im = port.v.im*port.vStart.re;
    port.i.re*port.iStart.im = port.i.im*port.iStart.re;
```
I checked the library source code, and as I see, `UPhase` is never really used anywhere, only computed for monitoring in the (now conditional) equations.

Please go ahead with #81.

AndreaBartolini, 2023-06-03:
it seems that the problem is for both vPu and VPu, that are used is several models...

AndreaBartolini, 2023-06-03:
the same for PPu, QPu, iPU and IPu ... it seems me strange that their calculation depend from a flag, but probably they cannot be conditional... so probably the right way is to roll back to the previous implementation (PR #79)

AndreaBartolini, 2023-06-03:
I checked the test `PowerGrids.Electrical.Test.TwoBusesResistiveTransmissionLineFault` that uses the model `LineConstantImpedanceFault` in which the `portVariablesPu` is set to false and, as foreseen, it does not run .... I roll back to the previous solution.

casella, 2023-06-03:
Wait, don't those model have `final portVariablesPU = true`?

AndreaBartolini, 2023-06-03:
I've to check one by one.... I'll do it... in the meantime we'll see also the CI results...

AndreaBartolini, 2023-06-03:
searching for `final portVariablesPU` on the entire library I got no matching .... but to put `final portVariablesPU = true` can be a strategy to be applied in the models that uses xPu variables.

casella, 2023-06-03:
If you check [PortAC](https://github.com/PowerGrids/PowerGrids/blob/f2915e3e0f0f88f4c845ee073ce62900367a1865/PowerGrids/Electrical/BaseClasses/PortAC.mo#L3) there are two flags:
- [parameter Boolean portVariablesPhases = false;](https://github.com/PowerGrids/PowerGrids/blob/f2915e3e0f0f88f4c845ee073ce62900367a1865/PowerGrids/Electrical/BaseClasses/PortAC.mo#L14)
- [constant Boolean portVariablesPu = true "Add per-unit variables to model";](https://github.com/PowerGrids/PowerGrids/blob/f2915e3e0f0f88f4c845ee073ce62900367a1865/PowerGrids/Electrical/BaseClasses/PortAC.mo#L15)

So, _phases can be computed optionally_ since they are not used in the models (see above) and take quite some time to compute because of atan. By default they are _not_ computed, unless the parameter is turned on at initialization

Instead, _per unit variables_ are computed based on a constant, which is true by default. This can't be set by the user with the GUI, only when extending the class, or with a modifier set manually.

I checked the library, the only model that sets portVariablesPu = false is LineConstantImpedanceFault. No big deal.

Maybe we could just get rid of portVariablesPu and always compute them. One can then get more efficient models using backend-based filters, see OpenModelica/OpenModelica#8271.

AndreaBartolini, 2023-06-03:
> Maybe we could just get rid of portVariablesPu and always compute them. One can then get more efficient models using backend-based filters, see [OpenModelica/OpenModelica#8271](https://github.com/OpenModelica/OpenModelica/issues/8271).

I think that this is the best solution, i will implement it asap.

casella, 2023-06-03:
Let's keep portVariablesPhases, though.

AndreaBartolini, 2023-06-03:
last comments implemented in PR #82

closed 2023-06-05 by AndreaBartolini