PowerGrids/PowerGrids #83: LineConstantImpedanceFault has a wrong modifier and wrong description

AndreaBartolini, 2023-06-03:
In the LineConstantImpedanceFault model the modifier of the R parameter of the lineA is the following: `R = faultLocationPu`. It gives the right value = 0 (but wrong Unit) in case of fault at the portA but it gives the wrong value = 1 (instead of R) if the fault is located at the portB.
The right modifier should be `R = faultLocationPu*R`.

Moreover, the current description of the model is the same of the LineConstantImpedance one, it should be updated according to the behavior of the model.

AndreaBartolini, 2023-06-03:
Moreover, reading the description of the parameter that defines the fault location:
```modelica
parameter SI.PerUnit faultLocationPu "Location of the fault (0: port A, 1: port B)" 
```
it is not clear (at least in my mind) the meaning of a value different from 0 or 1, like the value 0.7 used in the example `PowerGrids.Electrical.Test.TwoBusesResistiveTransmissionLineFault`.

@casella do you remember the exact meaning of this parameter?

casella, 2023-06-03:
The test model PowerGrids.Electrical.Test.TwoBusesResistiveTransmissionLineFault had R = 1, so the bug was not visible. Maybe we should use a different value for that, e.g. 1.4 or something.

AndreaBartolini, 2023-07-14:
Fixed by PRs #87 and #90

closed 2023-07-14 by AndreaBartolini