PowerGrids/PowerGrids #85: PowerGrids.Examples.Tutorial.IslandOperation.TwoGeneratorsLocalInitialization fails at initialization

casella, 2023-06-15:
Model PowerGrids.Examples.Tutorial.IslandOperation.TwoGeneratorsLocalInitialization fails at initialization, and was temporarily removed from the set that is tested by the OMC CI.

Some analysis and comparisons was carried out on a reduced version of the model, only including the left-hand-side of the network. As the line is unloaded at initialization, this doesn't change the solution. This reduced model is available on branch https://github.com/PowerGrids/PowerGrids/tree/TestInit

 Comparison with PowerGrids.Examples.Tutorial.IslandOperation.TwoGeneratorsOneReferenceGenerator model, which should give the same results, reveal that the solution of the split sub-system corresponding to the network equations starts from the same start values that immediately lead to convergence of the global initialization problem, but stray away from it dramatically already at the first Newton iteration. For example:
```
TGEN1.iA.re  =        40895.151   step =        27253.764   old =        13641.387
```
this result was obtained by using the default solver; also Kinsol fails in the same place, though it is harder to understand why by looking at the logs.

It seems that the network sub-system equations are wrong for some reason, since they do converge to a wrong solution. This should be investigated.

It is also worth noting that, after converging to the wrong solution at lambda = 0, the solver restarts at the (correct) value when tackling the first non-zero lambda step. This shouldn't happen, the solution found at lambda = 0 should be kept as initial guess for subsequent steps.

casella, 2023-06-15:
Hints for checking:
- the place where the decoupling takes place is OnePortAC
- maybe the localInit parameter is not set correctly and leads to wrong boundary conditions towards the grid

AndreaBartolini, 2026-09-18:
Made some further investigations with the complete network. 

[image]

The simulation fails during the calculation of the lambda-zero step of the initialization.

Looking inside the BLT, I noticed that the `reference node` placed in the `NTHV1` bus allows to solve the network in the following order (it prescribes the voltage and the phase in the bus):

- `GRIDL1` (4103 - linear (torn), unknowns: 2, iteration variables: 2)
- Sub-Grid in the red-box below (4796 - non-linear (torn), unknowns: 32, iteration variables: 32)

[image]
 
- `TGEN2` (5249 - non-linear (torn), unknowns: 10, iteration variables: 10)

The system that doesn't converge is the 4796 (subnetwork in the red-box).

I checked the residuals and they seem to be as expected: 
```
GRIDL2.port.U
  (residual) sqrt(GRIDL2.port.u.re ^ 2.0 + GRIDL2.port.u.im ^ 2.0) - GRIDL2.port.U = 0
  (residual) 1.7320508075688772 * LINE.vB.re - GRIDL2.port.u.re = 0
  (residual) 1.7320508075688772 * LINE.vB.im - GRIDL2.port.u.im = 0

GRIDL2 power
  (residual) 3.0 *   (LINE.vB.re * GRIDL2.port.i.re +       LINE.vB.im * GRIDL2.port.i.im) - GRIDL2.PIcon = 0
  (residual) (-3.0) * LINE.vB.re * GRIDL2.port.i.im + 3.0 * LINE.vB.im * GRIDL2.port.i.re - GRIDL2.QIcon = 0

constitutive eq. of GRIDL2
  (residual) GRIDL2.port.U - GRIDL2.U_URef * GRIDL2.URef = 0
  (residual) GRIDL2.QRefConst * GRIDL2.U_URef ^ GRIDL2.beta - GRIDL2.QIcon = 0
  (residual) GRIDL2.PRefConst * GRIDL2.U_URef ^ GRIDL2.alpha - GRIDL2.PIcon = 0

Kirchhoff laws (voltage) LINE
  (residual) LINE.vB.re + LINE.vz.re - LINE.vAt.re = 0
  (residual) LINE.vB.im + LINE.vz.im - LINE.vAt.im = 0

constitutive eq. LINE.YB
  (residual) LINE.YB.re * LINE.vB.re + (-LINE.YB.im) * LINE.vB.im - LINE.iBs.re = 0
  (residual) LINE.YB.im * LINE.vB.re +   LINE.YB.re *  LINE.vB.im - LINE.iBs.im = 0

constitutive eq. LINE.Y
  (residual) LINE.Y.re * LINE.vz.re + (-LINE.Y.im) * LINE.vz.im - LINE.iz.re = 0
  (residual) LINE.Y.im * LINE.vz.re +   LINE.Y.re *  LINE.vz.im - LINE.iz.im = 0

Kirchhoff laws (current) B LINE
  (residual) LINE.iB.re + LINE.iz.re - LINE.iBs.re = 0
  (residual) LINE.iB.im + LINE.iz.im - LINE.iBs.im = 0

Kirchhoff laws (voltage) TGEN2
  (residual) LINE.vB.re + TGEN2.vz.re - TGEN2.vAt.re = 0
  (residual) LINE.vB.im + TGEN2.vz.im - TGEN2.vAt.im = 0

Kirchhoff laws (current) connection set NTHV2
  (residual) TGEN2.iB.re + LINE.iB.re + GRIDL2.port.i.re = 0
  (residual) TGEN2.iB.im + LINE.iB.im + GRIDL2.port.i.im = 0

Transformer voltage ratio TGEN2
  (residual) TGEN2.k.re * TGEN2.vA.re + (-TGEN2.k.im) * TGEN2.vA.im - TGEN2.vAt.re = 0
  (residual) TGEN2.k.im * TGEN2.vA.re +   TGEN2.k.re *  TGEN2.vA.im - TGEN2.vAt.im = 0
  
Transformer current ratio TGEN2
  (residual) TGEN2.k.re * TGEN2.iAt.im + (-TGEN2.k.im) * TGEN2.iAt.re - TGEN2.iA.im = 0
  (residual) TGEN2.k.im * TGEN2.iAt.im +   TGEN2.k.re *  TGEN2.iAt.re - TGEN2.iA.re = 0

Kirchhoff laws (current) TGEN2 (note that iAs = 0 in the transformer model)
  (residual) TGEN2.iB.re + TGEN2.iAt.re - TGEN2.iBs.re = 0
  (residual) TGEN2.iB.im + TGEN2.iAt.im - TGEN2.iBs.im = 0

constitutive eq. TGEN.YB
  (residual) TGEN2.G * LINE.vB.re + (-TGEN2.B) * LINE.vB.im - TGEN2.iBs.re = 0
  (residual) TGEN2.B * LINE.vB.re +   TGEN2.G *  LINE.vB.im - TGEN2.iBs.im = 0

constitutive eq. TGEN.Y
  (residual) TGEN2.Y.re * TGEN2.vz.re + (-TGEN2.Y.im) * TGEN2.vz.im - TGEN2.iAt.re = 0
  (residual) TGEN2.Y.im * TGEN2.vz.re + TGEN2.Y.re * TGEN2.vz.im - TGEN2.iAt.im = 0

boundary conditions GEN2
  (residual) sqrt(TGEN2.vA.re ^ 2.0 + TGEN2.vA.im ^ 2.0) - GEN2.GEN.port.VStart = 0
  (residual) (-GEN2.GEN.port.PStart) - TGEN2.vA.im * TGEN2.iA.im - TGEN2.vA.re * TGEN2.iA.re = 0
```
unfortunately the kinsol log doesn't report the evolution of the solution, so it is not easy to understand what is the origin of the problem. 
I will try to reduce the network that fails (the one in the red box).

casella, 2026-09-20:
> unfortunately the kinsol log doesn't report the evolution of the solution, so it is not easy to understand what is the origin of the problem.

Isn't LOG_NLS_V providing enough info?

You can try LOG_NLS_JAC to see the jacobian and LOG_NLS_SVD/LOG_NLS_SVD_V to get an SVD decomposition of it, so you can see how far from singularity it is.

AndreaBartolini, 2026-09-30:
I made some further investigation by turning on the following flags:
- LOG_NLS_V
- LOG_NLS_JAC
- LOG_NLS_SVD_V

here below all the information from the log about the system that fails:
```
LOG_NLS		| info	 | <p>############ Solve nonlinear system 4410 at time 0 ############</p>
|		| |	 | | <p>initial variable values:</p>
|		| |	 | | | <p>[ 1]                    TGEN2.vA.re  =        11976.821   nom =                1</p>
|		| |	 | | | <p>[ 2]                   TGEN2.vAt.re  =                0   nom =                1</p>
|		| |	 | | | <p>[ 3]               GRIDL2.port.u.im  =       -480.43889   nom =           380000</p>
|		| |	 | | | <p>[ 4]                     LINE.vz.im  =                0   nom =                1</p>
|		| |	 | | | <p>[ 5]                   GRIDL2.QIcon  =            2e+08   nom =            1e+08</p>
|		| |	 | | | <p>[ 6]               GRIDL2.port.i.im  =        -302.7845   nom =        759.67141</p>
|		| |	 | | | <p>[ 7]                   GRIDL2.PIcon  =          4.5e+08   nom =            1e+08</p>
|		| |	 | | | <p>[ 8]                  GRIDL2.U_URef  =        1.0064162   nom =                1</p>
|		| |	 | | | <p>[ 9]                  GRIDL2.port.U  =        382438.16   nom =           380000</p>
|		| |	 | | | <p>[10]               GRIDL2.port.u.re  =        382437.86   nom =           380000</p>
|		| |	 | | | <p>[11]                    LINE.iBs.re  =                0   nom =                1</p>
|		| |	 | | | <p>[12]                    LINE.iBs.im  =                0   nom =                1</p>
|		| |	 | | | <p>[13]                     LINE.iB.re  =        -1.352791   nom =                1</p>
|		| |	 | | | <p>[14]                     LINE.iz.re  =                0   nom =                1</p>
|		| |	 | | | <p>[15]                     LINE.iB.im  =       -14.210242   nom =                1</p>
|		| |	 | | | <p>[16]                     LINE.iz.im  =                0   nom =                1</p>
|		| |	 | | | <p>[17]                     LINE.vz.re  =                0   nom =                1</p>
|		| |	 | | | <p>[18]                    TGEN2.iA.re  =        13519.989   nom =                1</p>
|		| |	 | | | <p>[19]                   TGEN2.iAt.re  =                0   nom =                1</p>
|		| |	 | | | <p>[20]                    TGEN2.vz.im  =                0   nom =                1</p>
|		| |	 | | | <p>[21]                    TGEN2.vz.re  =                0   nom =                1</p>
|		| |	 | | | <p>[22]                     LINE.vB.re  =         220800.6   nom =                1</p>
|		| |	 | | | <p>[23]               GRIDL2.port.i.re  =        678.96561   nom =        759.67141</p>
|		| |	 | | | <p>[24]                    TGEN2.iB.re  =       -677.61282   nom =                1</p>
|		| |	 | | | <p>[25]                   TGEN2.iBs.re  =                0   nom =                1</p>
|		| |	 | | | <p>[26]                   TGEN2.iBs.im  =                0   nom =                1</p>
|		| |	 | | | <p>[27]                     LINE.vB.im  =       -277.38152   nom =                1</p>
|		| |	 | | | <p>[28]                   TGEN2.vAt.im  =                0   nom =                1</p>
|		| |	 | | | <p>[29]                    TGEN2.vA.im  =         1885.671   nom =                1</p>
|		| |	 | | | <p>[30]                    TGEN2.iA.im  =       -6324.7999   nom =                1</p>
|		| |	 | | | <p>[31]                   TGEN2.iAt.im  =                0   nom =                1</p>
|		| |	 | | | <p>[32]                    TGEN2.iB.im  =        316.99474   nom =                1</p>
|		| |	 | | <p>KINSOL: Using linear solver method klu</p>
|		| |	 | | <p>Solution status: FAILED</p>
|		| |	 | | | <p> number of iterations           : 83</p>
|		| |	 | | | <p> number of function evaluations : 2297</p>
|		| |	 | | | <p> number of jacobian evaluations : 84</p>
|		| |	 | | | <p>solution values:</p>
|		| |	 | | | <p>[ 1]                    TGEN2.vA.re  =        11976.821</p>
|		| |	 | | | <p>[ 2]                   TGEN2.vAt.re  =                0</p>
|		| |	 | | | <p>[ 3]               GRIDL2.port.u.im  =       -480.43889</p>
|		| |	 | | | <p>[ 4]                     LINE.vz.im  =                0</p>
|		| |	 | | | <p>[ 5]                   GRIDL2.QIcon  =            2e+08</p>
|		| |	 | | | <p>[ 6]               GRIDL2.port.i.im  =        -302.7845</p>
|		| |	 | | | <p>[ 7]                   GRIDL2.PIcon  =          4.5e+08</p>
|		| |	 | | | <p>[ 8]                  GRIDL2.U_URef  =        1.0064162</p>
|		| |	 | | | <p>[ 9]                  GRIDL2.port.U  =        382438.16</p>
|		| |	 | | | <p>[10]               GRIDL2.port.u.re  =        382437.86</p>
|		| |	 | | | <p>[11]                    LINE.iBs.re  =                0</p>
|		| |	 | | | <p>[12]                    LINE.iBs.im  =                0</p>
|		| |	 | | | <p>[13]                     LINE.iB.re  =        -1.352791</p>
|		| |	 | | | <p>[14]                     LINE.iz.re  =                0</p>
|		| |	 | | | <p>[15]                     LINE.iB.im  =       -14.210242</p>
|		| |	 | | | <p>[16]                     LINE.iz.im  =                0</p>
|		| |	 | | | <p>[17]                     LINE.vz.re  =                0</p>
|		| |	 | | | <p>[18]                    TGEN2.iA.re  =        13519.989</p>
|		| |	 | | | <p>[19]                   TGEN2.iAt.re  =                0</p>
|		| |	 | | | <p>[20]                    TGEN2.vz.im  =                0</p>
[... cut ...]
|		| |	 | | | <p>[32]                     LINE.iB.im  =       -14.210242</p>
LOG_NLS		| warning	 | <p>nonlinear system 4691 fails: at t=0</p>

```

But KINSOL fails again and in the log there are no detailed information about the reason of the failure.

The only remaining doubt is about the nominal values, set to `1` for many of the iteration variables that can assume the values ​​of a much higher order of magnitude. Maybe this is what is causing the system to be ill-conditioned. I will try adjusting those as well.

Any idea and/or suggestions are welcome  ;)

AndreaBartolini, 2026-10-01:
I made a check in the code and the propagation of the `nominal` attributes seems to be right implemented.
Moreover, I noticed that by removing the annotation `--daeMode` the `nominal` attributes are well propagated:

```
LOG_NLS		| info	 | <p>############ Solve nonlinear system 4674 at time 0 ############</p>
|		| |	 | | <p>initial variable values:</p>
|		| |	 | | | <p>[ 1]               GRIDL2.port.u.im  =       -480.43889   nom =           380000</p>
|		| |	 | | | <p>[ 2]                    TGEN2.vA.re  =        11976.821   nom =        12124.356</p>
|		| |	 | | | <p>[ 3]                    TGEN2.iA.im  =       -6324.7999   nom =        13746.435</p>
|		| |	 | | | <p>[ 4]                    TGEN2.iA.re  =        13519.989   nom =        13746.435</p>
|		| |	 | | | <p>[ 5]                   TGEN2.iAt.im  =       -316.99474   nom =        688.96214</p>
|		| |	 | | | <p>[ 6]                   GRIDL2.QIcon  =            2e+08   nom =            1e+08</p>
|		| |	 | | | <p>[ 7]               GRIDL2.port.i.im  =        -302.7845   nom =        759.67141</p>
|		| |	 | | | <p>[ 8]               GRIDL2.port.i.re  =        678.96561   nom =        759.67141</p>
|		| |	 | | | <p>[ 9]                   GRIDL2.PIcon  =          4.5e+08   nom =            1e+08</p>
|		| |	 | | | <p>[10]                  GRIDL2.port.U  =        382438.16   nom =           380000</p>
|		| |	 | | | <p>[11]               GRIDL2.port.u.re  =        382437.86   nom =           380000</p>
|		| |	 | | | <p>[12]                    LINE.iBs.im  =                0   nom =        759.67141</p>
|		| |	 | | | <p>[13]                     LINE.iz.re  =         1.352791   nom =        759.67141</p>
|		| |	 | | | <p>[14]                     LINE.vz.re  =       -1407.4963   nom =         219393.1</p>
|		| |	 | | | <p>[15]                   TGEN2.iBs.re  =                0   nom =        688.96214</p>
|		| |	 | | | <p>[16]                    TGEN2.iB.im  =        316.99474   nom =        688.96214</p>
|		| |	 | | | <p>[17]                   TGEN2.iBs.im  =                0   nom =        688.96214</p>
|		| |	 | | | <p>[18]                     LINE.vB.re  =         220800.6   nom =         219393.1</p>
|		| |	 | | | <p>[19]                    LINE.iBs.re  =                0   nom =        759.67141</p>
|		| |	 | | | <p>[20]                     LINE.iB.re  =        -1.352791   nom =        759.67141</p>
|		| |	 | | | <p>[21]                    TGEN2.iB.re  =       -677.61282   nom =        688.96214</p>
|		| |	 | | | <p>[22]                   TGEN2.iAt.re  =        677.61282   nom =        688.96214</p>
|		| |	 | | | <p>[23]                    TGEN2.vz.im  =        37901.007   nom =        241909.76</p>
|		| |	 | | | <p>[24]                    TGEN2.vz.re  =          18165.5   nom =        241909.76</p>
|		| |	 | | | <p>[25]                   TGEN2.vAt.re  =         238966.1   nom =        241909.76</p>
|		| |	 | | | <p>[26]                    TGEN2.vA.im  =         1885.671   nom =        12124.356</p>
|		| |	 | | | <p>[27]                   TGEN2.vAt.im  =        37623.626   nom =        241909.76</p>
|		| |	 | | | <p>[28]                     LINE.vB.im  =       -277.38152   nom =         219393.1</p>
|		| |	 | | | <p>[29]                     LINE.vz.im  =        277.38152   nom =         219393.1</p>
|		| |	 | | | <p>[30]                     LINE.iz.im  =        14.210242   nom =        759.67141</p>
|		| |	 | | | <p>[31]                     LINE.iB.im  =       -14.210242   nom =        759.67141</p>
|		| |	 | | <p>KINSOL: Using linear solver method klu</p>
|		| |	 | | <p>Solution status: FAILED</p>
|		| |	 | | | <p> number of iterations           : 124</p>
|		| |	 | | | <p> number of function evaluations : 11255</p>
|		| |	 | | | <p> number of jacobian evaluations : 210</p>
|		| |	 | | | <p>solution values:</p>
|		| |	 | | | <p>[ 1]               GRIDL2.port.u.im  =       -480.43889</p>
|		| |	 | | | <p>[ 2]                    TGEN2.vA.re  =        11976.821</p>
|		| |	 | | | <p>[ 3]                    TGEN2.iA.im  =       -6324.7999</p>
|		| |	 | | | <p>[ 4]                    TGEN2.iA.re  =        13519.989</p>
|		| |	 | | | <p>[ 5]                   TGEN2.iAt.im  =       -316.99474</p>
|		| |	 | | | <p>[ 6]                   GRIDL2.QIcon  =            2e+08</p>
|		| |	 | | | <p>[ 7]               GRIDL2.port.i.im  =        -302.7845</p>
|		| |	 | | | <p>[ 8]               GRIDL2.port.i.re  =        678.96561</p>
|		| |	 | | | <p>[ 9]                   GRIDL2.PIcon  =          4.5e+08</p>
|		| |	 | | | <p>[10]                  GRIDL2.port.U  =        382438.16</p>
|		| |	 | | | <p>[11]               GRIDL2.port.u.re  =        382437.86</p>
|		| |	 | | | <p>[12]                    LINE.iBs.im  =                0</p>
|		| |	 | | | <p>[13]                     LINE.iz.re  =         1.352791</p>
|		| |	 | | | <p>[14]                     LINE.vz.re  =       -1407.4963</p>
|		| |	 | | | <p>[15]                   TGEN2.iBs.re  =                0</p>
|		| |	 | | | <p>[16]                    TGEN2.iB.im  =        316.99474</p>
|		| |	 | | | <p>[17]                   TGEN2.iBs.im  =                0</p>
|		| |	 | | | <p>[18]                     LINE.vB.re  =         220800.6</p>
|		| |	 | | | <p>[19]                    LINE.iBs.re  =                0</p>
|		| |	 | | | <p>[20]                     LINE.iB.re  =        -1.352791</p>
|		| |	 | | | <p>[21]                    TGEN2.iB.re  =       -677.61282</p>
[... cut ...]
|		| |	 | | | <p>[28]                     LINE.vB.im  =       -277.38152</p>
|		| |	 | | | <p>[29]                     LINE.vz.im  =        277.38152</p>
|		| |	 | | | <p>[30]                     LINE.iz.im  =        14.210242</p>
|		| |	 | | | <p>[31]                     LINE.iB.im  =       -14.210242</p>
LOG_NLS		| warning	 | <p>nonlinear system 4674 fails: at t=0</p>

```

But KINSOL fails again... starting from the solution and with the right `nominal` values...

@casella, I think we have to take a look together inside this issue...