PowerGrids/PowerGrids #88: Issue in PowerGrids.Electrical.Test.TwoBusesResistiveTransmissionLineFault

casella, 2023-07-13:
The frontend [reports](https://libraries.openmodelica.org/branches/master/PowerGrids_dev/files/PowerGrids_dev_PowerGrids.Electrical.Test.TwoBusesResistiveTransmissionLineFault.err):
```
[/home/hudson/saved_omc/libraries/.openmodelica/libraries/PowerGrids 2.0.0-develop/Electrical/BaseClasses/TwoPortAC.mo:11:3-12:47:writable] 
Error: Trying to override final element UStartA with modifier '= UStartA'.
Notification: Performance of FrontEnd: time 0.1697/0.267, allocations: 5.571 MB / 427 MB, free: 36.78 MB / 350.1 MB
```
Probably something went wrong with the refactoring, or has not yet been properly updated.

casella, 2023-07-13:
Here is the [regression report](https://libraries.openmodelica.org/branches/history/master/2023-07-08%2005:28:41..2023-07-10%2011:26:03.html).

AndreaBartolini, 2023-07-13:
here the regression report that I check....
https://libraries.openmodelica.org/branches/master/PowerGrids_dev/PowerGrids_dev.html

`Examples.Tutorial.GridOperation.Controlled.ControlledGridWithReferenceCalculationSat.err`

is breoken for a long time, the other model to be correct is 

`PowerGrids_dev_PowerGrids.Electrical.Test.TwoBusesResistiveTransmissionLineFault.err`

maybe you are referrin to an old report. ?

AndreaBartolini, 2023-07-13:
Here the report of teh Master branch (I don't work on it) checked today:

https://libraries.openmodelica.org/branches/master/PowerGrids/PowerGrids.html

AndreaBartolini, 2023-07-13:
Regarding the model `PowerGrids.Electrical.Branches.LineConstantImpedanceFault`, it seems to me that the start phase used in the originalmodel for the Fail point (the PhaseB of the first line):

```Modelica
UPhaseStartFaultPoint = (1 - faultLocationPu) * UPhaseStartA + faultLocationPu * UPhaseStartB
```

is just an approximation, because the voltage phase generally doesnt't change linearly with the Fault point position, is it correct?

AndreaBartolini, 2023-07-13:
closed by PR #89

closed 2023-07-13 by AndreaBartolini

casella, 2023-07-14:
> Regarding the model `PowerGrids.Electrical.Branches.LineConstantImpedanceFault`, it seems to me that the start phase used in the originalmodel for the Fail point (the PhaseB of the first line):
> 
> ```modelica
> UPhaseStartFaultPoint = (1 - faultLocationPu) * UPhaseStartA + faultLocationPu * UPhaseStartB
> ```
> 
> is just an approximation, because the voltage phase generally doesn't change linearly with the Fault point position, is it correct?

I guess so, but as usual approximated values are expected to be good enough to prime the Newton iterations.