PowerGrids/PowerGrids #106: Duplicated but nonidentical definitions

maltelenz, 2023-12-07:
When validating `PowerGrids.Examples.Tutorial.HowToBuild.Examples.TestLoadPQ` from the `develop` branch in Wolfram System Modeler, I get:

```
Error: PowerGrids.Electrical.Buses.InfiniteBus: Invalid class definition. Element BUSINF.terminalAC is duplicated, but its definitions are not identical.
```

We can see this is the case when we compare:

https://github.com/PowerGrids/PowerGrids/blob/41977fb5aa8e9cac3a044d96abd8265f44630257/PowerGrids/Electrical/Buses/InfiniteBus.mo#L13

with:

https://github.com/PowerGrids/PowerGrids/blob/41977fb5aa8e9cac3a044d96abd8265f44630257/PowerGrids/Electrical/BaseClasses/OnePortACBus.mo#L6-L11

This is not valid Modelica.

casella, 2023-12-10:
@maltelenz I'm not 100% sure about that. I've checked the model also with Dymola, which accepts it.

[Section 7.1](https://specification.modelica.org/maint/3.6/inheritance-modification-and-redeclaration.html) of the MLS says:
> The declaration elements of the flattened base class shall either:
> - Not already exist in the partially flattened enclosing class (i.e., have different names).
> - **The new element is a long form of redeclare or uses the class extends A syntax**, see [section 7.3](https://specification.modelica.org/maint/3.6/inheritance-modification-and-redeclaration.html#redeclaration).
> - Be exactly identical to any element of the flattened enclosing class with the same name and the same level of protection (public or protected) and same contents. In this case, the first element in order (can be either inherited or local) is kept. It is recommended to give a warning for this case; unless it can be guaranteed that the identical contents will behave in the same way.

Although the English formulation is a bit shaky (given the premise before the bullet points, I guess it should start with a verb), I understand the second clause to be applicable in this case.

From a library developer perspective, I could of course take the modifiers to `terminalAC` showing up in the base class [OnePortAC](https://github.com/PowerGrids/PowerGrids/blob/41977fb5aa8e9cac3a044d96abd8265f44630257/PowerGrids/Electrical/BaseClasses/OnePortAC.mo#L6) and write them again in redeclare clause of `terminalAC` inside [InfiniteBus](https://github.com/PowerGrids/PowerGrids/blob/41977fb5aa8e9cac3a044d96abd8265f44630257/PowerGrids/Electrical/Buses/InfiniteBus.mo#L13). However, that really looks like useless code duplication - my intention is to get those modifications when extending the other base class, and I expect them to be merged properly.

@perost, @HansOlsson, do I miss something?

perost, 2023-12-10:
I'm not sure what the second point is supposed to mean, but I guess "long form of redeclare" = redeclare as an element. In that case it wouldn't apply in this case since the redeclares are modifiers and not elements.

OpenModelica does not complain about the model because we don't check if the elements are identical well enough, and it will keep the first element in order and discard the rest as per the third point. In this case that happens to be the one with the modifiers on it, so it's kind of a "two wrongs make a right" situation where both the compiler and the model is wrong but the result happens to be correct.

casella, 2023-12-10:
So, there is basically no way to apply the modifiers in OnePortAC to the extends clause of InfiniteBusVI except by re-applying them in the second extends statement?

casella, 2023-12-22:
@maltelenz we extensively reviewed the inheritance architecture of PowerGrids 2.0.0 and found several flaws, some of them quite serious.

We have now completely refactored the base classes and the way they are inherited in the components to avoid this kind of issues - basically, we put all the binding expressions in the base class at the root of each inheritance tree, in order to avoid the need of re-applying them when doing multiple inheritance.

The code currently found in the HEAD node of the develop branch was also tested successfully for correctness with OpenModelica and with Dymola using pedantic mode, so it should now be OK.

Please check with System Modeller at your earliest convenience and reopen this issue if you still see problems.

maltelenz, 2024-01-08:
I still see issues.

When validating `PowerGrids.Examples.Tutorial.HowToBuild.Examples.TestLoadPQ` System Modeler complains about an issue that can be narrowed down to `PowerGrids.Electrical.PowerFlow.InfiniteBusPF`:

```
Error: PowerGrids.Electrical.PowerFlow.InfiniteBusPF Invalid class definition. Element isLinear is duplicated, but its definitions are not identical.
  parameter public final Boolean isLinear = true
  parameter public Boolean isLinear = true
```

The first definition is here:
https://github.com/PowerGrids/PowerGrids/blob/251b8478c55fa44f58b1fc6e3f1293f62d9e0ae2/PowerGrids/Electrical/PowerFlow/InfiniteBusPF.mo#L7

and the second comes in through the inheritance chain of `InfiniteBusVI` -> `BusBaseVI` where we find:

https://github.com/PowerGrids/PowerGrids/blob/251b8478c55fa44f58b1fc6e3f1293f62d9e0ae2/PowerGrids/Electrical/BaseComponents/BusBaseVI.mo#L5

Note the difference in `final`.

I don't have permission to reopen the issue...

casella, 2024-01-08:
@maltelenz thank you for the analysis. Apparently, System Modeller is much better at detecting these issues than Dymola and OpenModelica 😅.

I'll fix this ASAP.

You wrote "I still see issue**s**". Are there any other ones pending?

maltelenz, 2024-01-08:
I said "issues" because I have around 60 models currently failing that I'm blaming on this kind of problem. A quick browsing through them _seem_ to show the same root cause, so let's hope it is all the same single issue.

I'll let you know when you have fixed the specific case I reported if all the other ones also work :)

casella, 2024-01-09:
@maltelenz I just fixed BusBaseVI to have `final isLinear = true`, which is just sensible since all bus models are linear. I hope this solves all the 60+ failures, since buses are found pretty much anywhere in the runnable models of the library.

If there are still issues of the same kind, please report them here, otherwise please close the ticket.

Thanks for your feedback, it was really helpful to spark the final redesign effort.

maltelenz, 2024-01-09:
Unfortunately I still have the same number of failures. The next one I see:

Validating `PowerGrids.Electrical.Buses.EquivalentGrid`:

> Error: PowerGrids.Electrical.Buses.EquivalentGrid [1:1-1:1] Invalid class definition. Element terminalAC is duplicated, but its definitions are not identical.

Here is one:
https://github.com/PowerGrids/PowerGrids/blob/ccbd690d41a40cefc46883339596419ff33e2814/PowerGrids/Electrical/Buses/EquivalentGrid.mo#L9-L11

And in `OnePortAC` we have another:

https://github.com/PowerGrids/PowerGrids/blob/ccbd690d41a40cefc46883339596419ff33e2814/PowerGrids/Electrical/BaseClasses/OnePortAC.mo#L8-L10

The `final` for `computePF` is different.

The same is true for other cases of `final computePF` in the library, such as:

https://github.com/PowerGrids/PowerGrids/blob/ccbd690d41a40cefc46883339596419ff33e2814/PowerGrids/Electrical/BaseClasses/TwoPortACBase.mo#L9-L13

vs

https://github.com/PowerGrids/PowerGrids/blob/ccbd690d41a40cefc46883339596419ff33e2814/PowerGrids/Electrical/Branches/TransformerFixedRatio.mo#L8-L13

casella, 2024-01-09:
@maltelenz, please try now.

maltelenz, 2024-01-09:
I no longer see this issue for any of the models in the library!

Thanks for the quick fixes.

Closing.

closed 2024-01-09 by maltelenz