PowerGrids/PowerGrids #109: Refactor OnePortAC and OnePortACBus

casella, 2023-12-12:
Putting the common parts in OnePortACBase

casella, 2023-12-22:
We eventually decided to get rid of OnePortACBus

AndreaBartolini, 2024-01-11:
the solution based on the class refactoring (second in the previous comment) needs to duplicate the 99% of the code of the class `OnePortAC` into the new class `OnePortACbus`. This is not accettable because the `OnePortAC` class contains a lot of code and its duplication may generates issues in the code maintenance activities.

A third solution is to left unchanged the electrical `BaseClasses` structure and to work at Icons level. The dynamic behaviour of the icon text can be implemented directly in the icons, that receive the dynamic values to be shown as inputs.

AndreaBartolini, 2024-01-11:
Implemented in PR #115

AndreaBartolini, 2024-01-12:
Regression test successfully done.

closed 2024-01-12 by AndreaBartolini