PowerGrids/PowerGrids #110: Improve voltage control in IEEE14Bus test cases

casella, 2023-12-27:
Voltage control in the IEEE14Bus test cases has some stability issues. In all the test cases where some perturbations of the voltage are applied, the AVRs eventually bring the voltages to completely irrealistic values, possibly over 2 p.u., with consequent increase of the power adsorption by the PQ loads and eventual frequency collapse.

The AVRs should be re-tuned so as to ensure stable operation in all the transient scenarios.

casella, 2023-12-27:
This issue has always been present also in version 1.0.x. I think we should fix it both in a final version 1.0.3 and in the new version 2.0.0 which is currently available in the [develop](https://github.com/PowerGrids/PowerGrids/commits/develop) branch.

casella, 2024-10-01:
I looked into this issue and I now understand what the problem is.

The IEEE14Bus in version 1.0.x was set up reproduce the results of a previous test carried out by RTE with a code they trusted.. The problem here is that the original version of the test in release 1.0.3 was set up to reproduce the results of a test carried out with the PSM tool, which RTE trusted. So, it was a verification test of the whole library.

This was obtained by manually setting the power flow results of the PSM test in the UStart, UPhaseStart, PStart and QStart parameters of the system model, and then initializing it so that the initial power flow is fixed as initial states, adapting the AVR and turbine power inputs accordingly.

In the 2.0.0 version, we wanted to demonstrate the embedded power flow concept, then we kept the fixed intial power flow flag. Unfortunately, this means that the initial steady state is significantly different from the 1.0.3 version, and I strongly suspect that this is the root cause of the bad behaviour of the voltage control system.

This is very much related to the discussion in #135 and #151. After some reflections, I am now pretty convinced that we need to have two clearly separated sets of IEEE14bus simulations. One is meant for verification and should produce exactly the same results as 1.0.3, which were verified with PSM. The other one is meant for tutorial purpose; in this case nobody cares about the PSM tests, and we should instead clearly demonstrate a) how to handle such a non-trival case and b) what is the non-trivial relationship between the initial power flow and the initial steady-state solution, as discussed in #151.

casella, 2024-10-01:
See #160 for a proposed solution to this issue.

ceraolo, 2024-10-02:
IEEE14busShort14 stays stable for 20s, then it starts to drift (plots below).
Frankly speaking, it seems to me that it is very unlikely that this is due to poor initialization. 

[image]

A behaviour like this can be expected from systems having tiny stability regions around their equilibriums, which could not be used in practical applications. Sure not in actual Power Systems, like IEEE14 bus system.

I suspect there is somewhere a different issue, than just a non-exactly tuned intitial state.

If there is instability it might be more numerical instability than actual instability of the simulated system.

ceraolo, 2024-10-03:
Negative extcitation voltages are used to contrast severe load-sheddings. 
What is possible is that there is some mistake in the implemented AVR, which manifests only when negative excitation voltages are involved. It would be nice to have an opinion from who developped those AVRs (or chose their numerical parameters) on this.

casella, 2024-10-03:
It's a simple proportional controller. I checked, the upper and lower limits are +/- 5 p.u.

I guess this was part of the PSM tests, maybe @adriguir can comment on that.

ceraolo, 2024-10-03:
Ok.
When we starded to work on PowerLib 2.0.0 I manifested the need to avoid to manually copy all the PF output to the transient. I strongly believed I was right. 
At those times I did find a way to automatically transfer PF output to transient, but was told that EPF was better since did not make usage of hard disk. I took for granted that the two models (PF with brown terminals and transient with EPF) would give exactly the same results.

But what we get, e.g. for node 1, is as follows: an enormous difference in reactive power, that in turn might justify a negative Efield, which is normally beyond stability limits of synchronous generators!
[image]
[image]

If EPF is so inaccurate, I would rethink of automatic passage of PF data from PF studies (those with bronw terminals) to transient (blue terminals), through the mechanism I proposed. It worked very well in Dymola; it needed just a couple of tickets to be solved, to work the same in OM.

casella, 2024-10-03:
I don't think the EPF is inaccurate per se. It's just that the way the initialization problem was set up in this example is not correct, so we get an initial value of VPu = 1 at all generator buses, which is not what we want; we should get values that are around 1.05, if you check the solution in 1.0.3.

We'll fix that ASAP.

ceraolo, 2024-10-09:
> I don't think the EPF is inaccurate per se. It's just that the way the initialization problem was set up in this example is not correct, so we get an initial value of VPu = 1 at all generator buses, which is not what we want

I don't understand totally the point.
You used the same powerflow set points for both IEEE14BusPowerFlow and IEEEStaticNetworkEPF, right? So we should get the same output except, at most, small consequences of rounding errors. The picture in https://github.com/PowerGrids/PowerGrids/issues/110#issuecomment-2391997956 shows that this is not the case.

Then, the presence of a steady-state negative EMF, which is totally unrealistic, may well be the culprit for the instability shown in pictures in https://github.com/PowerGrids/PowerGrids/issues/110#issuecomment-2388599392 and https://github.com/PowerGrids/PowerGrids/issues/110#issuecomment-2389042549. That negative EMF, as I said, is probably due to the too large (machine-absorbed) reactive power coming from PowerFlow, even though I did not check this numerically.

casella, 2024-10-16:
> I don't understand totally the point. You used the same powerflow set points for both IEEE14BusPowerFlow and IEEEStaticNetworkEPF, right?

Not really. The initialization is configured to keep the power flow values at the generator ports and adjust the set points backwards accordingly. The problem is, if we switch to embedded power flow, we get a different power flow than the one that the other RTE tool computed, so you get different voltages and different set points. A completely different initial solution. Which may as well be outside the stability region of voltage control.

ceraolo, 2024-10-17:
> > I don't understand totally the point. You used the same powerflow set points for both IEEE14BusPowerFlow and IEEEStaticNetworkEPF, right?
> 
> Not really. The initialization is configured to keep the power flow values at the generator ports and adjust the set points backwards accordingly. 

I thought this. What I said is "You used the same powerflow set points for both IEEE14BusPowerFlow and IEEEStaticNetworkEPF, right?" which means the P-Q and P-V values at the PF ports are the same, and the backwards-calculated regulator setpoints are consistent with them.
It this is true, the two PF solutions , BusPowerflow and StaticNetworkEPF should be the same

casella, 2024-10-17:
> What I said is "You used the same powerflow set points for both IEEE14BusPowerFlow and IEEEStaticNetworkEPF, right?" which means the P-Q and P-V values at the PF ports are the same, and the backwards-calculated regulator setpoints are consistent with them.

We didn't. In the old tests, the voltages in the initial fixed power flow are around 1.05 p.u., in the new one they are 1 p.u.

ceraolo, 2024-10-17:
> We didn't. In the old tests, the voltages in the initial fixed power flow are around 1.05 p.u., in the new one they are 1 p.u.

aha! So, it is just the matter of changing a few "1.0" 's to "1.05" or so? Doing this should require not more than half an hour... I'm looking forward to seeing wheter this works. At this point there are good chances that they will, and this will be a cornerstone for PowerGrids, since would be a good validation test for EPF.

casella, 2024-10-17:
> aha! So, it is just the matter of changing a few "1.0" 's to "1.05" or so?

Not really. The problem is that the original problem has voltage-dependent loads, so if you really wanted to replicate the original case using the EPF, it would be quite complicated.

But I don't think this is really necessary.

If you already have a PF and you just want to reproduce it, you set the PF values directly in the components, and that's it. No need to use the embedded PF. The embedded PF is to be used when you don't have a pre-computed PF, so you start from scratch.

In order to validate the embedded PF (= show that the EPF is identical to the initial steady state) I think we should use a dedicated test (could still be IEEE14Bus), in which we need to set alpha = beta = 0 on the loads. This is already planned.

ceraolo, 2024-10-17:
> aha! So, it is just the matter of changing a few "1.0" 's to "1.05" or so?
> Not really. The problem is that the original problem has voltage-dependent loads

ok. But at least, we'll likely find a solution not ouside the stability limits of the system, which then will not drift. That would be a very important step.

> In order to validate the embedded PF (= show that the EPF is identical to the initial steady state) I think we should use a dedicated test (could still be IEEE14Bus), in which we need to set alpha = beta = 0 on the loads. This is already planned.

This will be welcome as well, obviously. But the outmost priority for me, before recommending PowerGrids to professionals,  is to have a credible IEEE14bus simulation, which is probably obtained just changing some 1.0" 's to "1.05" 's or so.

casella, 2024-10-17:
I'll fix this properly ASAP

AndreaBartolini, 2026-08-21:
The frequency collapse was generated by a bug in the initialization, as explained in #168. Now the frequency collapse has been solved.

AndreaBartolini, 2026-09-17:
after the PR #196 not only the frequency collapse has been solved but also the voltages seem to be stabilized:

[image]

so I close the ticket.

closed 2026-09-17 by AndreaBartolini