PowerGrids/PowerGrids #111: Disable P and Q DynamicSelect display in buses, which is useless and clutters the diagram

casella, 2023-12-27:
Buses by construction have P = 0 and Q = 0, so it doesn't make sense to clutter the diagrams by adding that information on the interactive diagrams.

The display of P and Q should be subject to an extra parameter, which is disabled in buses.

ceraolo, 2024-01-18:
Except for slack bus!

casella, 2024-01-18:
Indeed, good point. @AndreaBartolini, could you add one extra base class with DynamicSelect labels to be used for the Reference bus (dynamic) and slack bus (power flow) components?

AndreaBartolini, 2024-01-19:
> Buses by construction have P = 0 and Q = 0, so it doesn't make sense to clutter the diagrams by adding that information on the interactive diagrams.
> 
> The display of P and Q should be subject to an extra parameter, which is disabled in buses.

Currently the buses show U and Uphase on the diagrams, not P nor Q

AndreaBartolini, 2024-01-19:
> Indeed, good point. @AndreaBartolini, could you add one extra base class with DynamicSelect labels to be used for the Reference bus (dynamic) and slack bus (power flow) components?

That's right, reference bus, slack bus and also the equivalent grid can have P/Q different from 0, I will add these dynamic labels on their icons.

closed 2024-01-19 by AndreaBartolini