PowerGrids/PowerGrids #113: Restructure tutorial

casella, 2023-12-27:
The current tutorial is organized in two sections:
- Library Architecture
- Examples

The first section goes very deep into the description of the library structure, which is interesting if you want to understand more and possibly develop your own models, but is definitely not necessary to start using the library. The tutorial should be restructured in three sections
- Using the Library
- Examples
- Library Architecture

where the first section provides an introduction to the drag-n-drop use of the library without getting into the details on how it is organized internally.

ceraolo, 2024-01-27:
It seems to me (looking at the ENTSOE TestCase1) that the dynamic labels on diagram show values at the beginning of transient (i.e. PF output), which, if confirmed, is very good.
Maybe a few words on this can be added in the tutorial as well?

AndreaBartolini, 2024-03-08:
As a reminder, the following arguments should be added:
- embedded power flow
   - overview
   - slack node repositioning to a generator node
   - auxiliary initial equations in the generator model to be used to compute external offset for PmPu and ufPu
   - some significant examples (see PowerGrids.Electrical.Test.SynchronousMachine4WindingsPF)

casella, 2024-07-25:
The section about using the library is now complete. I just need to add the section on the EPF