PowerGrids/PowerGrids #114: Introduce system-wide validity limits

casella, 2023-12-27:
It often happens that numerical simulation of PowerGrids models reach values of voltage, current, and frequency which are completely unrealistic and way beyond the validity assumptions of some component. 

In some cases, this eventually leads to simulation failure, which is potentially problematic because the feedback from the solver in that case does not give any indication about the root cause of the problem. Even worse, in some cases, these solutions are still numerically sound and thus accepted by the simulation tool, even though they actually don't make much sense. See for example #110: the IEEE14Bus have some AVR stability problems that mostly went unnoticed and only caused failures in some test cases but not in others, even though they reach values of frequency and current which are clearly unrealistic.

Currently, on the [develop](https://github.com/PowerGrids/PowerGrids/tree/develop) branch there are some hard-wired assertions enforcing a maximum current of 1.5 p.u in loads, which was meant to catch bogus initial equations with very low voltage and very high current.

In order to improve usability and diagnostics, particularly for inexperienced novice users, we should probably expand this feature by adding more such assertions, making the limits customizable from the system object. On the other hand, it should probably also be possible to disable these checks in case the generated code is meant to be used inside some other software system, which takes care of enforcing those limits.

Proposal:
- add `parameter Boolean enableAssertions = true` in the system object, to globally activate and de-activate such limits
- add the system-level defaults in the system object:
  - VPuMin = 0.7
  - VPuMax = 1.3
  - IPuMax = 1.5
  - omegaPuMin = 0.9
  - omegaPuMax = 1.1

The voltage limits should be applied to all buses, with default values set by the system object; the current limits should be applied to all loads, with default values set by the system object; the frequency limits should be applied to all generator models, with default values set by the system object.

In this way, the default set up ensures reasonable operational limits with zero set-up effort, allowing to immediately pin-point potentially critical points in the sytem - the limits can then easily be customized, either at the system level, or locally, or possibly completely disabled if deemed necessary.

The idea

AndreaBartolini, 2024-02-20:
In case of assertion error OpenModelica does not show the value of the failed condition nor produces results to be analized, this makes very difficult the investigation of the cause that has triggered the assertion.

To mitigate this problem I propose to add in the System Object also a parameter `assertionLevel`, so that all the assertions mentioned in this issue can be set to warning for debug purposes.

AndreaBartolini, 2024-02-20:
> The voltage limits should be applied to all buses and synchronous generators, with default values set by the system object; the current limits should be applied to all loads

Basically there are two ways in order to implement this limit checking
1) duplicate the assert clauses in all loads an bus models
2) add the assert clauses to the base class OnePortACVI

The second way has the advantage to avoid the code duplication but at this level is not possible to indentify the type of component, so it is no possible to choose the type of check (Vpu, IPu) according to the component type (bus or load).

I don't see any problem in checking both VPu and IPu independently from the type of component, so as first implementation I'll always check both VPu and IPu. In case of problem is easy to add boolean filter in order to exclude some test.

AndreaBartolini, 2024-02-20:
In some scenarios (e.g., fault with low impedance to ground) some limits could be temporarly violated during the transient, if the default of the assertion level is set to "error" this stops the simulation.

The better way is to set the default of the assertion level to "warning".

AndreaBartolini, 2024-02-20:
implemented by PR #130

AndreaBartolini, 2024-02-21:
I forgot to implement the flag `enableAssertions`

AndreaBartolini, 2024-02-21:
Implemented by PR #131

closed 2024-02-21 by AndreaBartolini