PowerGrids/PowerGrids #116: Enhance Power Flow output on diagrams

ceraolo, 2024-01-17:
**This ticket regards only the PF feature of the new PowerGrids.**

Now we have the possibility to see the PF results directly on the system diagrams. This is a very important and useful feature. Indeed, when doing PF studies, I think users will 90% of the time have a look at just the diagram with numerical result on it, and only rarely will read other results through the output variables.

I like to seee these results in PU. However, for this case, I think it is important to make a change to make it even more useful.

Now active and reactive powers are referred to the nominal machines they belong to, which in general are different from different machines in the system. For instance a 100 MVA generator can be connected to a 130 MVA transformer.
So, when we output the PF results on the diagram in PU, PU numbers are not directly comparable. 
Consider for instance bus33 in the picture below. the power coming from pv33 is 100 MW, which goes into the two transformers below it. In fact their nominal powers are 75 MVA, so they receive 0.6667*75= 50 MW each. This is not immediately evident from the numerical labels on the diagram, since they refer to different base units. 

The situation could be improved a lot if all active and reactive powers are shown in a unique base, e.g. 100 MVA for the whole grid.

My proposal is therefore to add to the SystemObject an option to be selected through a checkBox, whose name could be "Use unique power base for power display on diagrams".
The this checkbox is selected an input field should become active, allowing specifying the value of the unique power base (e.g. 100 MVA).

In that case the user would have a clear picture of the power flow just having a look at the diagram:

- they could select the UNom value for buses that to correspond to the adjacent transformer windings
- they could select a unique power base for the system e.g. 100 MVA. This will override local nominal values for the computation of the PQ active and reactive powers displayed on diagrams.

Is this possible with a reasonable effort?
Does anyone agree that this could be helpful?

[image]

casella, 2024-01-22:
To be honest, I'm not sure I see the advantage of that over just looking at the MW outputs. If you set the base power to 100 MVA, the difference would just be that the decimal point is moved by two places... That said, if that is what power engineers are used to, so be it.

The whole iPSL library is based on this concept, and I found it quite confusing. But if we keep it at the level of displaying (not writing models), it's fine for me.

BTW, I guess this only makes sense for power, not for voltage, so it would be SNom, not UNom.

The implementation of this feature should be fairly straightforward. However, I would keep adding new features that are not backwards-incompatible for the 2.1.0 version, I think now we should focus on releasing what we have ASAP.

ceraolo, 2024-03-12:
> To be honest, I'm not sure I see the advantage of that over just looking at the MW outputs. If you set the base power to 100 MVA, the difference would just be that the decimal point is moved by two places... That said, if that is what power engineers are used to, so be it.

That's right. But numbers on diagrams make them cluttered, and writing 1.234  (pu of 1 GW) is more compact than 1234000 (in W, i.e., pu of one watt). I think four significant digits are sufficient for numerical data on diagrams.
I agree it is a minor issue,  though.

ceraolo, 2024-03-12:
> To be honest, I'm not sure I see the advantage of that over just looking at the MW outputs. If you set the base power to 100 MVA, the difference would just be that the decimal point is moved by two places... That said, if that is what power engineers are used to, so be it.

Quite true. 
This ticket could be useful, however, to have powers that sum at nodes (the sum of powers flowing into them being zero) and still bus voltages in pu.
But I agree that it has low priority, if some. We will see what happens when people start (hopefully) using the library regularly.

AndreaBartolini, 2026-08-20:
currently it is possible:
- to visualize all quantities in p.u.
- to visualize all quantities in SI

The user can select the visualization mode by setting the following flags in the `system` object:
- `showDataOnDiagramsPu`
- `showDataOnDiagramsSI`

These flags are mutually exclusive, when one is set the other becomes unavailable (disabled). If both are not set the visualization is turned off.

AndreaBartolini, 2026-08-20:
@ceraolo  we can add, if it is useful, a third possibility: power in SI, voltage and phase in p.u.

AndreaBartolini, 2026-08-20:
@ceraolo we can add also the possibility to define in the `system` object a unique `SBaseDisp` and a unique `UBaseDisp` to be used as common base for the visualization.

Let me know what are the best solutions to be implemented

ceraolo, 2026-08-20:
> currently it is possible:
>
> * to visualize all quantities in p.u.
> * to visualize all quantities in SI
>
> The user can select the visualization mode by setting the following flags in the `system` [...]

Please consider this ticket's description. Saying "PU" is vague: which base?

ceraolo, 2026-08-20:
> [@ceraolo](https://github.com/ceraolo) we can add, if it is useful, a third possibility: power in SI, voltage and phase in p.u.

Personally I would not use this option. It seems to me it adds confusion instead of clarity.

ceraolo, 2026-08-20:
> [@ceraolo](https://github.com/ceraolo) we can add also the possibility to define in the `system` object a unique `SBaseDisp` and a unique `UBaseDisp` to be used as common base for the visualization.
> 
> Let me know what are the best solutions to be implemented
MMM

AndreaBartolini, 2026-08-20:
Let me understand... 
the point (as far as I understood) is that the current visualization in PU may refer to different bases because there is no a unique base in the library (each component uses a local base defined by the local UNom and the SNom), and this may be a bit confusing.

The idea to use a unique base for the visualization on the diagram (to be set in the `system` object) is in general good for me, the only caveat I see is that the values of the same variable in PU displayed on the trend may be different, because the trend uses the local base of the component.

If this difference between the bases used for the diagram and for the trends is not a problem I can easily implement the modification...

AndreaBartolini, 2026-10-07:
Implemented by PR #212

closed 2026-10-07 by AndreaBartolini