PowerGrids/PowerGrids #120: Bad power expression detected by Dymola

ceraolo, 2024-01-20:
Consider model Compare_SI_PU.wipSI in
[Test.mo.txt](https://github.com/PowerGrids/PowerGrids/files/13997111/Test.mo.txt)

It runs in OM, but Dymola gives the following error:

Error: The following error was detected at time: 0
Model error - power: (pv33x2.port.u.re^2+pv33x2.port.u.im^2) ^ (-0.5) = (0) ^ (-0.5)

So, the base of a power with negative exponent is zero. Mathematically it should be correct and equal to 1, but I'm not sure what MS says on this.

closed 2024-01-22 by casella

ceraolo, 2024-01-23:
In this example I used ordinary PowerFlow, not the Embedded power flow feature.

casella, 2024-01-23:
Interesting. Are you sure all UStart, PStart, and QStart parameters are set correctly?

In any case, this looks like a Dymola bug. If things are done correctly, `pv33x2.port.u` should have a start value with non-zero magnitude.