PowerGrids/PowerGrids #121: Bus icon should be shorter

ceraolo, 2024-01-21:
Buses are somewhat accerssory parts of diagrams.

Power system diagrams are often filled of several components, and to have clear view of everything accurate choice of model icon locations is important.

This brings to the need to put buses near to other components, e.g. generators.
An example is in the following diagram:
[image]
Here I selected the bus. As you can see, the area that can be used for dragging buses (the four small red squares) overlaps generator's. This is suboptimal, since makes trickier to move generators: clicking on the lower part of the generator icon and dragging moves the bus instead of the generator.

I propose to make the vertical span of buses smaller. E.g. (-40,40) or (-30,30)  instead of (-100, +100).

ceraolo, 2024-01-21:
This ticket is partly wrong. Maybe I'll reopen it in the future

casella, 2024-01-22:
I understand the issue, and I agree that the extent should be close to the real one. However, if we do this, we affect all the existing diagrams in the library, where each and every bus will have to be resized, which is a really ugly work.

In any case, if we ever wanted to do this, we absolutely need to do it _before_ the release, otherwise we would affect all the models built with the library.

casella, 2024-01-23:
@ceraolo, apparently if all models with diagrams are opened in OMEdit, when the extent of the icon is modified, the extents in all the diagrams that use it are updated automatically. Which means, we can do this in a very short time. @AndreaBartolini will take care of it.

AndreaBartolini, 2024-01-23:
Unfortunately this is just an illusion created by OM... the diagrams are "graphically" updated but nothing is changed in the model annotations, so if you close and reopen OM all the diagrams have the icons enlarged and/or deformed.

We need to study another strategy...

casella, 2024-01-23:
Wait, if the diagrams are (somehow) updated in OMEdit, it should be possible to save this information. We need to talk to @adeas31.

casella, 2024-01-23:
The reason why the open diagrams were not changed was probably some caching. 

See OpenModelica/OpenModelica#11866 for a request of a proper solution. Let's see what @adeas31 thinks.

casella, 2024-01-24:
BTW, I guess we should do something similar for the Synchronous Machine and Load icons. Currently, they have a lot of space in the upper half of the extent, which of course overlaps with the empty extent of the bus icon placed in the neighbourhood 😄 

This makes grabbing and moving those icons really awkward. I guess this should be fixed, if we don't want new users to get angry very soon while using the library...

AndreaBartolini, 2024-01-24:
I've already done that for all icons in PowerGrids :( but due to the fact that OM doesn't update the annotations of the other models this is the final (orrible) effect:

[image]

AndreaBartolini, 2024-01-24:
obviously I've not pushed the branch....

casella, 2024-01-24:
I'm discussing this with @adeas31. There are two steps:
- the frontend should identify which annotations must be updated
- OMEdit should compute the new extent

Not trivial, but not too difficult either.

I'm working on it.

ceraolo, 2024-01-25:
There is another option: do not change the icon size, and let OM do like dymola, which lets grabbing icons only where there  is something (see this ticket's description)

AndreaBartolini, 2024-01-25:
I agree with @ceraolo, the alternative option (grabbing icons only where there is something) applies to any library, not only to PowerGrids.

Thera are a lot of my libraries that suffer the same issue....

casella, 2024-01-25:
Sounds reasonable. I'll discuss that in https://github.com/OpenModelica/OpenModelica/issues/11866

casella, 2024-01-25:
See [comment](https://github.com/OpenModelica/OpenModelica/issues/11866#issuecomment-1910685676) in #11866.

casella, 2024-01-25:
Nice team work 😄

casella, 2024-02-13:
Now that OpenModelica/OpenModelica#11866 is fixed, also OpenModelica allows to grab only the drawn portion of the extent, so there is no need to actually change it.