PowerGrids/PowerGrids #124: Three phase short and fault

ceraolo, 2024-01-26:
When looking for a bus fault the average user wonders what the difference between ThreePhaseFault and ThreePhaseShort is.
Unfortunanately, Faults and ThreePhaseShort have no info, and therefore this question remains unanswered.

AndreaBartolini, 2024-01-29:
The `ThreePhaseFault` component was born after a long discussion around the `Ground` component previously present in PowerGrids (see https://github.com/PowerGrids/PowerGrids/issues/62#issuecomment-1590046471).
Maybe the `Fault` package is not the best place to store it, as far as I know its use is mainly in test cases of components in which a fixed reference point is needed (e.g., PowerGrids.Electrical.Test.TestEquivalentGrid).

For sure properly documentation shal be added to the component (I'll change the name of this ticket in this direction).

ceraolo, 2024-01-29:
I see. Looling at TestEquivalentGrid I consider that maybe what you did is not so bad, but maybe can be improved.
Since we are in a rare situation in which we have not to be warried by backward compatibility, some more reasoning can be done. 

In Power Systems, faults are series fault (openings) and shunt faults (shorts).
So, in principle, folder "Faults", since it does not have any series fault, nor can n contain (by library design) unsymmetrical faults, should only contain one component, i.e. the one currently called ThreePhaseFault, but better called ThreePhaseShort instead.
Then, for the TestEquivalentGrid one could just use that component, using as startTime a negative balue.

AndreaBartolini, 2024-01-29:
Just a remark... this story seems to be very similar to that in https://github.com/PowerGrids/PowerGrids/issues/62#issuecomment-1590046471, that ended with the creation of the two components `ThreePhaseShort` and `ThreePhaseFault`.

Just to be sure before to make similar changes in the library...  the current `ThreePhaseFault` component cannot prescribe v=complex(0) as the current `ThreePhaseShort` do, because its equation is i = Y*v, is this the intended behavior?

ceraolo, 2024-01-30:
> Just a remark... this story seems to be very similar to that in [#62 (comment)](https://github.com/PowerGrids/PowerGrids/issues/62#issuecomment-1590046471), that ended with the creation of the two components `ThreePhaseShort` and `ThreePhaseFault`.
> 
> Just to be sure before to make similar changes in the library... the current `ThreePhaseFault` component cannot prescribe v=complex(0) as the current `ThreePhaseShort` do, because its equation is i = Y*v, is this the intended behavior?

Is there a good technical reason not to write v=Z¡ instead?

AndreaBartolini, 2024-01-30:
The `ThreePhaseFault` works as a "breaker" with assigned impedance (that is transformed in equivalent admittance), that is open/closed according to the assigned time interval. 
The only way to "open" the breaker is to set its admittance to zero (by using an impedance it should be set to infinite), this the reason because the equation is admittance-based.

ceraolo, 2024-01-30:
So there is perfect duality: if we want to open and close the breaker me must accept the parameters (either impedance or admittance) to assume zero or very large values.
No big problems in either case.
I would personally prefer impedance, but is just a question of taste.

AndreaBartolini, 2024-01-30:
The point is another, the `ThreePhaseFault` model cannot be used (as currently written) to prescribe v=0, as the `ThreePhaseShort` does, so the proposal to use `ThreePhaseFault` to substitute the `ThreePhaseShort` in the `TestEquivalentGrid` (as far as I understood) is not applicable.

@casella any suggestion?

casella, 2024-01-30:
At the end of the day, I understand the `ThreePhaseShort` has no use in real test models: there's no point having a component in a system that permanently short-circuits the three phases with zero resistance. It is a purely fictitious component, that in fact is only used in one test model `PowerGrids.Electrical.Test.TestEquivalentGrid`.

Real-life short circuits are always transient, and they always have a non-zero short-circuit resistance. On the other hand, when they are not applied, the resistance is infinite. So, they are best written as V = Y*i, since you can compute the short-circuit admittance from the non-zero impedance, and keep Y = 0 when the fault is not active. As we do in the current `ThreePhaseFault` component.

My proposal:
- remove the `ThreePhaseShort` component, as it is basically useless
- replace it by the ThreePhaseFault component in `TestEquivalentGrid` with a very small resistance (say R = 1e-6) and startTime = -Modelica.Constants.Inf, to provide validation of the equivalent grid component

Do you agree?

AndreaBartolini, 2024-01-30:
> * remove the `ThreePhaseShort` component, as it is basically useless

I agree

> * replace it by the ThreePhaseFault component in `TestEquivalentGrid` with a very small resistance (say R = 1e-6) and startTime = -Modelica.Constants.Inf, to provide validation of the equivalent grid component

I agree with a note: it is not really the same thing. 
Currently we use the equation v=0 at the left side of the line, with the new solution we'll use v=i/Y with a very high Y, that is slightly different... but probably irrelevant i the `TestEquivalentGrid` case (maybe not in more complex cases, but this is another story).

waiting for @ceraolo final ok to proceed with the change in the library.

ceraolo, 2024-01-30:
Since it is already implemented, the component implementing direct connection of the wires, previously called "Ground" may be left in the library. Maybe it could be called "DirectConnection" or "WireToWireConnection". 
It would not harm, but not be very useful either. 

An indication of its low usefulness is that we have difficulty in finding a place (folder) to put such component in.

casella, 2024-01-30:
> I agree with a note: it is not really the same thing. Currently we use the equation v=0 at the left side of the line, with the new solution we'll use v=i/Y with a very high Y, that is slightly [...]

From a numerical point of view, for a small example, it is the same thing. The point is to test that we haven't made _gross_ mistakes in the model of the equivalent grid. A difference on the 6th decimal digit is totally irrelevant, and I don't expect numerical issues on such a small test case.

casella, 2024-01-30:
> An indication of its low usefulness is that we have difficulty in finding a place (folder) to put such component in.

I would leave it out. It can generate lots of confusion and is basically useless

closed 2024-02-14 by AndreaBartolini