PowerGrids/PowerGrids #125: Data on diagram issues

ceraolo, 2024-02-13:
I find some issues on the option of "ShowDataOndiagrams" feature:
1) I suppose not all people need to see all those labels, that clutter large diagrams. If one selects ShowDataOnDiagramsPu=false and ShowDataOnDiagramsSi=false numerical data should be omitted from diagrams, I suppose.
2) no info on this feature is in the documentation (which is lacking for a lot of other aspects). For instance I had to see by trial that the data showen in a transient are those at the beginning of the it (at the end of PF computation) and not at the end of it.

----------------------------------------------------------

A third issue is more involed. Consider a line icon. The instance name can be put above o below the line through a flip operation (here I used PowerGrids 1.0, because of the issue 1 above): 
[image]

The name above the icon is the "classic" position of MSL.

In PowerGrids now it is possible to visualize also P and Q;
[image]

The natural order of P and Q is, first P, then Q, because they are always listed this way, which is also the importance and alphabetic order. To do this, one has to use the line icon in such a way that P and Q are below the line. This is somewhat unnatural, since usually fluxes are shown above lines and components.

This part of this ticket it to propose to exchange positions of P and Q for lines and transformers. If we do this, we obtain P and Q in the right order when they are above the line. This implies that the instance name is below, but this is a minor inconvenience, because those that do not use display of P and Q can always flip the icon to have names above them (if issue 1 above is solved).

casella, 2024-02-13:
Regarding the first issue, there is a fundamental problem: even though the ShowDataOnDiagrams parameters are false when you are building a model, you may later on switch them to true. If we do not show P,Q on the diagrams as placeholders, their space could be inadvertently occupied by other icons, and then actual results will be messed up when you turn on ShowDataOnDiagram. So, we propose to keep P and Q there as placeholders anyway.

Regarding the second issue: the feature is currently documented in section 3.8 of the LibraryArchitecture tutorial. As we already agreed, we need to write the Using Power Grids section for end users (not developers), currently it is just a copy of the LibraryArchitecture, we'll take care of it ASAP. How tools show the results shown by DynamicSelect features is tool dependent: OMC shows the initial values by default, while Dymola shows the final one; then, you can move the slider and the other controls (play, FFW, rewind) and see all other values. Not sure we should discuss these tool-specific aspect in the library documentation, since we don't control them (at least not Dymola, or other commercial tools).

Regarding the third issue, you are right. Andrea originally put the names above as in the MSL, though in fact putting them below seem a lot more natural. We are still in time to change this, since 2.0.0 is not backwards-compatible, so no harm is done to existing system models.

AndreaBartolini, 2024-02-14:
solved by PR #127

closed 2024-02-14 by AndreaBartolini