PowerGrids/PowerGrids #126: Obscure warning

ceraolo, 2024-02-13:
Consider Model Umach1WIP in the following package.
[Test.mo.txt](https://github.com/PowerGrids/PowerGrids/files/14268441/Test.mo.txt)

Running it gives the following warnigs, that, I'm afraid I'm unable to decode.
[image]

Since the model does not run, understanding this warning is therefore critical for me. Is it my fault or something in the diagnostics can be improved?

Tested with OM 1.23.0-310 and Powergrids 2.0.0-develop dated 15/1/2024

casella, 2024-02-13:
The models does not run, but the reason is not due to these warnings (we'll open a separate ticket on them), but rather to the error that shows up later on.
```
[6] 17:39:30 Symbolic Error
An independent subset of the model has imbalanced number of equations (987) and variables (988).
```
The set of imbalanced equations is omitted (it's huge), but it contains almost exclusively `componentPF` variables, i.e., it is the embedded power flow problem.

The problem with the EPF is that you connected a synchronous machine model (which embeds a `PVBus` EPF model) and a reference bus (which embeds a `SlackBus` EPF model). Obviously this generates a singular system, because both try to fix the same voltage and say nothing about the current.

In fact, we already experienced this problem in the test models for the synchronous machine, that was connected to an infinite bus to fix the machine voltage at the terminal. In that case, the cure was to disable the embedded power flow, because we really want to fix that voltage.

For the case at hand, we're afraid the only possible cure on your side is to move the reference bus somewhere else (e.g. close to a load), in the same place you would put the slack bus in the PF model. For us, is to write in the documentation something like this:

> If the embedded power flow is enabled, synchronous machine models, which embed a PVBus, cannot be connected to components whose embedded power flow component also tries to fix the voltage, such as ReferenceBus, InfiniteBus, and EquivalentGrid.

Unfortunately, we don't see any other feasible way to catch the error on the library side. Improvements on the tool side could help figuring out the source of the singularity of the embedded power flow. Dymola is somewhat smarter, and tells you that 

> Part of the problem for Real elements is underdetermined.
> 1 scalar Real equations are missing for the variables or its derivatives.
> We did not find equations for:
> gen42x2.iPF.im
> But it could be other indeterminate variables as well.

This was a strong hint that something was wrong with the EPF around that generator's terminal.

AndreaBartolini, 2024-02-16:
fixed by PR #129

AndreaBartolini, 2024-02-16:
Indeed the PR #129 fixes the "obscure warnings" but not the cause of the simulation error, so I reopen the issue

AndreaBartolini, 2024-02-22:
This discussion is better explained in the Issue #132

closed 2024-02-22 by AndreaBartolini