PowerGrids/PowerGrids #132: Slack bus in EPF equivalent to a pair Reference bus + Generator in Dynamic simulation

AndreaBartolini, 2024-02-22:
Is common practice in the PF calculation to substitute a powerful node with a slack node, in order to avoid the iterative balancing of the network. In this scenario the slack node represents a generator big enough to supply the same power that is supplied by the slack node in the PF. 

From the PowerGrids point of view this means that the EPF should substitute a `generator` (choosen by the user) with a `SlackBus` (instead of a `PVBus`) and the `ReferenceBus` with a `BusPF` (insted of a `SlackBus`).

This behavior can be implemented in different ways, herebelow two possibilities:
1. to make replaceable the EPF component in both the `generator` and the `ReferenceBus` (default remains `PVBus` and `SlackBus` respectively), by this way the user can choose the EPF behaviour by selecting the EPF component in the parameters dialog, 

2. to double the `generator` and the `ReferenceBus` models, by changing the EPF component, by this way the user can choose the EPF behaviour during the model building, by selecting the proper components in the library.

The first way is the most simple to be implemented and avoids the replication of the models in the library, but from the user point of view is a more bit complicated to be learned and probably less intuitive. 

The second way is probably more simple from the user point of view but less efficient from both the coding and the code maintenance point of view.

Personally I prefer the first way, I await comments before starting with the implementation...

@max-privato and @casella, can you provide comments and or suggestions?

casella, 2024-02-22:
I agree I prefer solution 1.

Andrea also mentioned another possible solution at some point, namely the reference bus only setting the phase of the voltage. Was that dismissed?

ceraolo, 2024-02-22:
Using a p-theta node would not have significant usages, IMO.

AndreaBartolini, 2024-02-23:
> I agree I prefer solution 1.
> 
> Andrea also mentioned another possible solution at some point, namely the reference bus only setting the phase of the voltage. Was that dismissed?

Because it did not match the request (I undesrtood the point only after a call with @ceraolo ).
The point is to substitute a generator with slack node in the PF, and then to transfer both the active and the reactive power supplied by the slack node to the generator in the dynamic model, in order to initialize the simulation.

This can be achived only by embed the slack node in the generator as EPF component, if the slack node is outside the generator it is no possible to use its power to automatically initialize the generator (at least with the current implemented EPF).

AndreaBartolini, 2024-02-27:
The proposed solution actually does not work under OpenModelica because OMEdit does not updates the parameters pop-up window when a replaceable model is redeclared.
See issue [#12037 ](https://github.com/OpenModelica/OpenModelica/issues/12037)

AndreaBartolini, 2024-03-07:
Implemented by PR #138 (please see the PR comments)

AndreaBartolini, 2024-03-08:
some additional features implemented by PR #140

AndreaBartolini, 2024-03-09:
some additional features implemented in PR #142

casella, 2024-04-06:
@ceraolo suggests that a fairly standard practice is to connect the slack node to a strong generator, so the configuration made possible by this ticket should be the reference one for models not connected to an infinite grid.

The currently implemented solution is not particularly user-friendly, as it requires to redeclare the EPF model of the generator to slack node and the EPF model of the ReferenceBus to Bus. This is not particularly intuitive and potentially error-prone. From an end-user perspective, the EPF should preferably "just work".

We came up with a different solution, that would just require to set a Boolean parameter `slackEPF` in the slack generator and connected ReferenceBus components. The idea is to modify `OnePortAC` to have two conditionally instantiated EPF components: `componentPF` and `componentPFSlack`. If `slackEPF = false` the first is activated, otherwise the second is.

casella, 2024-10-02:
Also note that all tutorial models using generators are now complaining that componentPF.SNom and componentPF.UNom have no value. Apparently the redeclare is causing the parameter propagation to be lost. We need to figure out a more robust way to implement this feature.

AndreaBartolini, 2026-09-14:
Solved by PRs from #198 to #202 

Now the new EPF approach allows one generator in the island to be the reference (and slack in the EPF) node (flag `isRefNode` in the `Embedded PF` tab).

A commented example is available in `PowerGrids.Electrical.Test.SynchronousMachine4WindingsPF`

closed 2026-09-14 by AndreaBartolini

casella, 2026-09-15:
For the record, we eliminated the ReferenceBus outright. If you have a connection to a strong point of the network, there's no need for it, otherwise there's really no need to place it anywhere else than where a strong generator is. So, let's have the generator do the job. Its EPF counterpart will just be a slack node.