PowerGrids/PowerGrids #151: PF-EPF continuity

ceraolo, 2024-03-18:
I think the average user will need to have some "continuity" between PF and initial solution of transients.
I've prepared a few simple  models, based on Electrical.Test.SynchronousMachine4WindingsPF, available here: [TestEPF.mo.txt](https://github.com/PowerGrids/PowerGrids/files/14634062/TestEPF.mo.txt). I ran them in Dymola 2023x.

Model **manSetPoint**s gives ouput is as follows:
[image]

They are rather good, but something is probably to be improved.
The voltage amplitudes on the two busbars are exactly equal to each other (in all their available digits in Dymola Variable Browser), and the phase on reference/slack bus is zero for both. This should imply be exactly equal power flows in the two examples, only roundoff and convergence errors remaining.
In the numerical results, however, we see non-neglible differences: one percent in active power of _machine1_ is small, but still meaningful. 
**Maybe there is a way to enhance complancy of the two results?**

**Model autoSetPoints** is derived from man setPoints just sutbstituting pm and uf inputs to the machine. AFAIK, this should be correct, and give the same result as manSetPoint. **Do I miss something?**

**Model noEPF** is like manSetPoints, but it does not use EPF. Instead, here I set bus1 voltage manually to the value we have in the PF model (25540 V). I got this:

[image]

The transient power flows are very different from the PF's. **I expected them to be equal**. Am I wrong?

casella, 2024-04-06:
I discussed this with @ceraolo and @AndreaBartolini. There are multiple issues at stake here. The most important is that these tests are set up with option GlobalSteadyStateFixedPowerFlow, which is meant to back-compute generator inputs to get the required power flow. This needs extra initial equations prescribing the power flow and extra `fixed = false` parameters to adjust the initial values of the generator inputs.

Unfortunately, we put the extra initial equations in the controllers, not in the machine, so they are there when using the ControlledGenerator model, see the IEEE14Bus examples, but not here. The reason was that we wanted to achieve steady-state initialization also in the case of controller saturation, and this was theoretically possible only by putting those equations in the AVR. In practice, this [never worked](https://libraries.openmodelica.org/branches/master/PowerGrids/files/PowerGrids_PowerGrids.Examples.Tutorial.GridOperation.Controlled.ControlledGridWithReferenceCalculationSat.sim), so it turns out that was not such a good idea.

We need to move the extra initial equation to the SynchronousGenerator model, so that the power flow is always enforced, regardless of the fact that a control system is present, see #152.

The other issue is that set points can be back-computed by adding the FreeOffset block, or by setting them to parameters with `fixed = false`. This also wasn't done in the above-reported experiments.

casella, 2024-04-06:
In any case, we agreed with @ceraolo that it would be nice to show one example in the tutorial where the PF result and the actual initial steady-state value of the simulation are exactly the same. This requires to use the same model for the loads, i.e., the dynamic model should use PQ loads with zero depency on voltage. We also agreed to add one more load to the other bus, to make such an example less trivial.

ceraolo, 2024-11-24:
> In any case, we agreed with @ceraolo that it would be nice to show one example in the tutorial where the PF result and the actual initial steady-state value of the simulation are exactly the same. [...]

No need for this, @casella. Transient loads can recive from PF P, Q, V. Then, they compute the nominal absorbed power according to these values. Can still have alpha and beta.
Used this years ago (and reported in some thicket in PowerGrids) and worked well.

AndreaBartolini, 2026-09-14:
fixed by PR #202

closed 2026-09-14 by AndreaBartolini

casella, 2026-09-15:
Yesss! 😃