PowerGrids/PowerGrids #152: Move initial equations for GlobalSteadyStateFixedPowerFlow from controllers to SynchronousMachine

casella, 2024-04-06:
The `GlobalSteadyStateFixedPowerFlow` initialization option is meant to back-compute generator inputs to get the required power flow. This needs extra initial equations prescribing the power flow and extra `fixed = false` parameters to adjust the initial values of the generator inputs.

We put the extra initial equations in the controllers, not in the machine, because we wanted to achieve steady-state initialization also in the case of controller saturation, and this was theoretically possible only by putting those equations in the AVR. In practice, this [never worked](https://libraries.openmodelica.org/branches/master/PowerGrids/files/PowerGrids_PowerGrids.Examples.Tutorial.GridOperation.Controlled.ControlledGridWithReferenceCalculationSat.sim), so it turns out that was not such a good idea.

We need to move the extra initial equation to the SynchronousGenerator model, so that the power flow is always enforced, regardless of the fact that a control system is present. In case the power flow cannot be achieved because of excitation saturations, we have two options:
1. fail the initialization
2. remove the saturations with `if initial()` clauses, so a solution is found, but then when the simulation starts, the saturations will be enforced and there will be a transient.

1. is much easier to achieve. @ceraolo what do you think?

AndreaBartolini, 2026-08-18:
I added two new initial equations in the generator, they can be used to set external offset by prescribing P and V calculated by the EPF if the `PV bus` is used as EPF component.
They are activated by the flag `useEPFtoSetExternalOffsetPV=true`. (PR #191)

The previous flag `useEPFtoSetExternalOffset` has been renamed to `useEPFtoSetExternalOffsetPQ`.

AndreaBartolini, 2026-08-19:
New example available: ` PowerGrids.Electrical.Test.SynchronousMachine4WindingsPF3`.

In this example two generators are present, one works as __slack__ node in the EPF, the other as a __PV bus__. The dynamic initialization matches the results of the EPF.

AndreaBartolini, 2026-09-14:
Solved by PRs from https://github.com/PowerGrids/PowerGrids/pull/198 to https://github.com/PowerGrids/PowerGrids/pull/202

Now the new EPF approach allows one generator in the island to be the reference (and slack in the EPF) node (flag isRefNode in the Embedded PF tab).

A commented example is available in `PowerGrids.Electrical.Test.SynchronousMachine4WindingsPF`

closed 2026-09-14 by AndreaBartolini

casella, 2026-09-15:
For the record, we eliminated the ReferenceBus outright. If you have a connection to a strong point of the network, there's no need for it, otherwise there's really no need to place it anywhere else than where a strong generator is. So, let's have the generator do the job of being the reference node during initialization. Its EPF counterpart will just be a slack node.

AndreaBartolini, 2026-09-15:
I'm not sure that the ReferenceBus can be avoided in case of the _local_ steady state initialization, in which the generators are virtually _detached_ from the network during the initialization.

In this case I suppose that to balance the network a ReferenceBus acting as slack is still necessary, we have to discuss about this.

AndreaBartolini, 2026-09-15:
The `ReferenceBus` is also still used by the following models:
- `PowerGrids.Examples.Tutorial.HowToBuild.Examples.TestSimpleGenerator`
- `PowerGrids.Examples.ENTSOE.TestCase1`

I'm not sure that we can use something else in this two cases.

casella, 2026-09-15:
>     * `PowerGrids.Examples.Tutorial.HowToBuild.Examples.TestSimpleGenerator`

Should we add the required initial equations to this simple generator as well?

>     * `PowerGrids.Examples.ENTSOE.TestCase1`

Can't we use the initial equations in the synchronous generator model in this case?

AndreaBartolini, 2026-09-15:
It is not a problem related to the initial conditions (there are no offset to be calculated in that models), the point is related to the boundary condition to be connected to the generator during the simulation.

In bot the cases the `ReferenceNode` is used to simulate an _open breaker_ connected to the generator, by prescribing the voltage and the phase in zero-load condition during the entire simulation.

AFAIK we have no other component in the PowerGrids that can be used for the same purpose.