PowerGrids/PowerGrids #153: Avoid manual iterations to bring the active and reactive slack power of the ReferenceBus to zero

casella, 2024-04-06:
When modelling islanded grids, the ReferenceBus component sets the reference voltage during initialization and provides slack active and reactive power if generation, consumption, and losses are not perfectly balanced. The idea is of course to get this slack power as close as possible to zero, but this requires to perform iterations in which generation or consumption are changed until the slack power is small enough to be acceptable.

One way to avoid this is to connect the reference bus to a generator, so that the excess slack power can be provided by the attached generator, bringing it to zero in just one iteration. However, this still requires a manual iteration. An even better solution in this case would be to automatically adapt the generator inputs (or the generator control system set points) so that the slack power becomes exactly zero.

With the modifications discussed in #132, this is already what happens in the EPF, as the reference bus is turned into a regular bus (thus zero slack power) and the attached generator is converted into a slack bus, so it provides exactly the power required to balance the system. It would be nice to get the same behaviour also for the dynamic model.

This can be achieved by adding a `balanceReferenceBus` Boolean parameter to the initialization parameters of the System object. When that parameter is set to true, two additional equations will be activated in the ReferenceBus, setting `PSlack = 0` and `QSlack = 0`. Of course this requires to free two parameters in the generator that acts as slack node, i.e., the one that has `slackPF = true`, see [#132](https://github.com/PowerGrids/PowerGrids/issues/132#issuecomment-2041202221):
- if the initialization mode is `globalSteadyStateFixedSetPoints`, this will require to put FreeOffset blocks in the generator inputs or generator controller set points, so that they can be adapted to balance the grid without slack power in the ReferenceBus
- if the initialization mode is `globalSteadyStateFixedPowerFlow`, then the given power flow for the generator with `slackPF = true` will be ignored and replaced by the computed values that balance the grid

If the initialization mode is `localSteadyStateFixedPowerFlow` then the ReferenceBus already has `PSlack = 0` and `QSlack = 0`, so it doesn't make sense to turn on `balanceReferenceBus` in this case.

@ceraolo, once we have sorted out #132, this feature is trivial to implement, as it involves a few lines of code, so we can add it to 2.0.0 without problems.

AndreaBartolini, 2026-09-14:
Solved by PRs from https://github.com/PowerGrids/PowerGrids/pull/198 to https://github.com/PowerGrids/PowerGrids/pull/202

Now the new EPF approach allows one generator in the island to be the reference (and slack in the EPF) node (flag isRefNode in the Embedded PF tab).

A commented example is available in `PowerGrids.Electrical.Test.SynchronousMachine4WindingsPF`

closed 2026-09-14 by AndreaBartolini

casella, 2026-09-15:
For the record, we eliminated the ReferenceBus outright. If you have a connection to a strong point of the network, there's no need for it, otherwise there's really no need to place it anywhere else than where a strong generator is. So, let's have the generator do the job. Its EPF counterpart will just be a slack node.