PowerGrids/PowerGrids #154: Connect ReferenceBus to large generator in islanded examples

casella, 2024-04-06:
After #132, we should move the ReferenceBus close to a large generator in the IslandOperation tutorial model and in the IEEE14Bus model. In the former case, I would also keep the current model, so you can show both options, and possibly show different initialization options as discussed in #153,

casella, 2024-07-25:
See also #135

AndreaBartolini, 2026-09-14:
Solved by PRs from https://github.com/PowerGrids/PowerGrids/pull/198 to https://github.com/PowerGrids/PowerGrids/pull/202

Now the new EPF approach allows one generator in the island to be the reference (and slack in the EPF) node (flag isRefNode in the Embedded PF tab).

A commented example is available in `PowerGrids.Electrical.Test.SynchronousMachine4WindingsPF`

closed 2026-09-14 by AndreaBartolini

casella, 2026-09-15:
For the record, we eliminated the ReferenceBus outright. If you have a connection to a strong point of the network, there's no need for it, otherwise there's really no need to place it anywhere else than where a strong generator is. So, let's have the generator do the job. Its EPF counterpart will just be a slack node.