PowerGrids/PowerGrids #155: Generator user masks

ceraolo, 2024-09-18:
This issue relates to branch "Develop" as in 19th of September 2024.

In the synchronous generator user mask (e.g. in ENTSOE.SteadyState.GEN). initialisation compares both in the General and inthe Initialization tabs (screenshots below).
This is a bit confusing. Wouldn't be better to put all the initializations in the initialization tab?

[image]

[image]

casella, 2024-09-18:
@ceraolo, thanks for reporting. 

That initialization group is generated automatically by the GUI, because there are explicit start modifier. This is a mechanism that is used to handle initial equations with fixed attributes, e.g. in the Modelica MultiBody library.

I think this should be easilty solved by making those bindings `final`, so they cannot be changed upon instantiation. The idea is that these start values are managed by means of the bound parameters, e.g. `uPuStart`. 

@AndreaBartolini do you agree?

AndreaBartolini, 2024-09-18:
If the proposed solution (to make the bindings `final`) works fine there are no caveats from my side.

casella, 2024-09-19:
Just bear in mind that we'll get the desired oucome whtn ticket OpenModelica/OpenModelica#12898 is fixed 😅

casella, 2024-10-16:
Everything should be fixed now both on the source code front and on the OMEdit front. I'll check that ASAP.

AndreaBartolini, 2026-09-14:
Fixed by PR #203

commit 8d68fd7d (2024-10-10): Merge pull request #163 from PowerGrids/fix-155

Fix #155
closed 2026-09-14 by AndreaBartolini