PowerGrids/PowerGrids #160: Restructure IEEE14bus test cases

casella, 2024-10-01:
The IEEE14Bus tests in version 1.0.x was set up reproduce the results of a previous set of tests carried out with the PSM tool, which RTE trusted. So, it was a verification test of the whole library. This was obtained by manually setting the power flow results of the PSM test in the UStart, UPhaseStart, PStart and QStart parameters of the system model, and then initializing it so that the initial power flow is fixed as initial states, adapting the AVR and turbine power inputs accordingly.

I think we should keep this set of tests for verification purposes, by still explicitly setting the XXXStart parameters (no embedded power flow), but not use it for the tutorial, where it is just confusing.

As discussed in #135 and #151, for the tutorial we should instead set up two cases that involve using the EPF. The first should be a simple one where the results of the EPF coincide exactly with the initial steady state. The second instead should be set up so that they are slightly different, to demonstrate how that can happen and under what circumstances.

This should also solve issue #110, which I understand is due to poor initialization.

@ceraolo, @AndreaBartolini, do you agree?

AndreaBartolini, 2024-10-02:
I partially agree, in my opinion at least one example in that all the xxxstart parameter are "forced" instead to use EPF should be present, by explaining that this is a possible way to check a solution vs another tool.

ceraolo, 2024-10-02:
> @ceraolo, @AndreaBartolini, do you agree?

Before answering we must really understand #110. I've just wrote something there:
https://github.com/PowerGrids/PowerGrids/issues/110#issuecomment-2388599392

Naturally, I think tutoral examples (all or, as @AndreaBartolini says all but one) should rely on EPF. But if EPF produces something like in the screenshot I reported in https://github.com/PowerGrids/PowerGrids/issues/110#issuecomment-2388599392, I think we've got a big problem.

casella, 2024-10-02:
> I partially agree, in my opinion at least one example in that all the xxxstart parameter are "forced" instead to use EPF should be present, by explaining that this is a possible way to check a solution vs another tool.

Good point, we could have three sub-packages for the tutorial:
1. Test with EPF and matching initial steady-state
2. Test with EPF and non-matching initial steady state (e.g. because of voltage-dependent loads)
3. Test with directly set power flow parameters, for verification against another tool