PowerGrids/PowerGrids #161: Allow re-simulating EPF studies with different PF data

ceraolo, 2024-10-09:
Re-simulating (i.e. simulating without re-building the exe, so see what happens when changing a non-structural parameter), is very useful in general and in particular in Powergrids.

When we make PowerFlow studies, in particular, we often do powerflow computations several times, with different voltage/active power/ reactive power setting points at PV and PQ nodes. 
For these, it is a real pain to have to wait the executable to be built all times before see the outcome of these parameter changes.

PR #158, with a simple action allowed re-simulating in PowerFlow studies (using models from Electrical.PowerFlow sub-library).

Consider now  IEEE14busStaticNetworkEPF. It can be successfully used for PowerFlow studies, whose result can then be used a starting point for transient analysis. So, allowing re-simulating also from here, after changing powerFlow parameters, would be a boon.
Insofar, we can chnge PPF and successfully re-simulate (for instance GEN1.GEN.PPF), but cannot do the same with UPF (in the same example GEN1.GEN.UPF). 

It would be great, if one could change also UPF for re-simulating studies, i.e. make UPF non-structural variable, as PPF already is.
Unfortunately, differently from PR #158 I could not find an as easy way to do so.

Are any ideas on how to attain this very important result?

casella, 2024-10-16:
@ceraolo I need to look into that. To me it's a bit odd that explicitly setting U makes it changeable at runtime, while leaving it at its (equal) default value doesn't.

In any case, the goal is that all non-structural parameters should be changeable at runtime. Maybe we need to change something in the OMC code to achieve that, rather than trying to get this behaviour by modifications to the Modelica code.

ceraolo, 2024-10-17:
> @ceraolo I need to look into that. To me it's a bit odd that explicitly setting U makes it changeable at runtime, while leaving it at its (equal) default value doesn't.

For me it is clear enough: if a value is a formula (i.e. not a literal constant), it cannot be changed, just because it depends on something other.

I also would say that the quality of being"structural" depends on how the tool works internally. AFAIK there is not (and probably cannot) be a universal definition of structural parameters. Therefor e I assume that for a given tool it is "structural" any parameter not allowed to be changed for re-simulations

casella, 2026-08-20:
As far as I understand, this is an OMC bug, the UPF parameter should be changeable at runtime. See OpenModelica/OpenModelica/issues/16360.