PowerGrids/PowerGrids #165: Lack of a load switch component

ceraolo, 2024-11-27:
PowerGrids lacks a switch component that would be used to insert of detach loads.  This is inconvenient.

For instance in Examples.Tutorial.GridOperation.LoadChange, the programmers had to resort to changes by modifier or extends, which is somewhat tricky. LoadChangeByInputSignal is ok, but refers to a totally different model.
In case a switch component is made available, adding new loads would be done by just closing a switch, which would be much nearer to how engineers think and, AFAIK, other tools do.

I think that the addition of a "switch" component with could connect/disconnect elements: lines, loads, etc. would improve the library usability significantly.

I know that there is already available component LineConstantImpedanceWithBreaker, so the usage of this switch for lines is somewhat redundant; for loads, however, it would be actually very useful.

casella, 2024-11-28:
The problem with switch components is, an open switch will have an equation such as 
```modelica
i = 0;
```
If the load behind the switch tries to prescribe P and Q, there won't be any solution to the system equations, and the simulation will crash. The case with lines is different: one can hope that if you trip a line, power will find a way to flow through other lines and reach the load anyway, albeit with greater losses. But in the case of PQ loads, this would happen all the time.

Ditto for synchronous generators, unless the control system has provisions to handle no-load conditions.

@ceraolo how would you handle loads in that case?

ceraolo, 2024-11-28:
The first thing that comes in my mind is that I would take inspiration from MSL.
A switch there has a very small or very large resistance when is closed or open respectively.

Or maybe another way is to use conditional components, and totally remove a load when its switch is open. But I did not check the feasiliblity with Modelica constructs.

casella, 2024-11-28:
You can't remove components during a simulation, unfortunately.

I guess the MSL does not have PQ loads - it has physical models that still work if you cut the current.

ceraolo, 2024-11-28:
> You can't remove components during a simulation, unfortunately.

For n-1 security this is not needed

> 
> I guess the MSL does not have PQ loads - it has physical models that still work if you cut the current.

As I said, MSL switches do not cut current, they just change their internal resistance from a very low value to a very large one and vice-versa. See, for instance Modelica.Electrical.Analog.Ideal.IdealOpeningSwitch

casella, 2024-11-28:
> As I said, MSL switches do not cut current, they just change their internal resistance from a very low value to a very large one and vice-versa. See, for instance Modelica.Electrical.Analog.Ideal.IdealOpeningSwitch

I understand that, but this is not going to save you if you have PQ loads. If you have a large resistance, there's no way you can get the required power through a large resistance, i.e., the network equations won't have a solution and the simulation will crash.

ceraolo, 2024-11-28:
Quite true. I said this  because you mentioned cutting current.

This said, frankly I do not see any difficulty in changing the required P to 0 or something very low.

However, this ticket is not critical. I proposed it because I thought (and still, frankly believe) it was a very easy task. In case it is not, feel free to close it: we already have  
LoadChangeByInputSignal for load changes.

casella, 2024-11-28:
> This said, frankly I do not see any difficulty in changing the required P to 0 or something very low.

But then, why not just having a PQ load with a switch? In fact, we already have that, a PQ load with P and Q inputs - you change the inputs to zero, et voilà.

AndreaBartolini, 2026-08-20:
A second alternative could be to implement a generic breaker that, when opened, connects one side to the ground through a fixed admittance Y (parameter).

This kind of breaker should be used (in theory) for any kind of component, taking care to connect the detached component to the side connected to the ground when the breaker is opened.

AndreaBartolini, 2026-10-12:
Implemented by the PR #213

closed 2026-10-12 by AndreaBartolini