PowerGrids/PowerGrids #168: Is IEEE14busShort4 correct?

ceraolo, 2024-12-14:
I've run IEEE14busShort4 example of PowerGrids (develop branch). Here's the gen8 speed:
[image]

Here I see two strange things:
1. when faults occur, typically nearby generators tend to accelerate, because the reduced voltage causes reduced power absorbed by loads, and (before governors compensate) some input power is transformed into kinetic energy. Here, instead, I see immediate slowing down of the generator.
2. very low damping occurs, which is rather strange expecially in the second part of the plot.

Just to make a comparison, I checked with IEEE_14_buses from OpenIPSL, which allows to see the effects of the same fault at the same bus. 
I know that this model is not identical to PowerGrid's: generator, controller, load models and parameters are different. However, both should give "reasonable" results. From this example, I get:
[image]

This has not any of the two strange things smentioned above.

What could be the cause of those two strange things?

ceraolo, 2024-12-14:
The global behaviour with OpenIPSL shows that the system at the very end of the simulation has a small increase in speed:
[image]

In case of absence of regulators, this is logical the rotating partes should have larger kinetic energy at the simulation end than at the beginning, due to the loss of energy  absorption from loads during short.
With regulators, I'm not sure.

AndreaBartolini, 2026-08-18:
@casella looking in the tutorial I found that the concepts of the initialization through the buses is still present:

__2.1 Drag, drop and connect__
_...it is recommended that you connect different components through Bus components, which provide proper initialization support, as well as clear visualization of node voltages in the grid..._

__2.3 Initialization__
_...To this purpose, all components with nonlinear equations have parameter to set start values (i.e. initial guesses) for the active and reactive power flows into their ports, while bus components provide start values for initial guesses of voltages..._

So I suppose that the mechanism to propagate the initial voltage and phase from the buses to the connected component was removed by mistake during the simplification of the base classes structure.

My opinion is to restore that mechanism, by reintroducing the different base class for the bus (the mechanism requires a pair of input-output connectors to propagate initial voltage and phase from the bus to the components). The side effect is that you must use buses to connect components (due to the input-output pairs), but we had already discussed about this and the conclusion was that this is not a big constrain.

AndreaBartolini, 2026-08-21:
After some discussion with @casella, we have decided to avoid to restore the old mechanism to transfer the start values from the buses to the connected components. The reasons to avoid the restore are the following: 
- if the EPF is used then the start values are calculated by the EPF, `UStart` and `UPhaseStart` are no longer necessary,
- if the start values are taken from an external PF then it is reasonable that all the values from the external PF will be transferred to the Modelica model by using scripts (not by hand), so the transfer mechanism is not necessary at all.

Following this decision, I reverted to the original IEEE14bus network where `UStart` and `UPhaseStart` values ​​were applied directly to the loads and generators, and the frequency collapse is now resolved in all transient cases.

Here below the transient of the IEEE14BusShort4

[image]

[image]

The following issues are still open:
1. why the frequency does not increases when the loads suddenly decrease theirs power request,
2. why there is a very low-damped transient in some synchronous condensers (only the `GEN1` and the `GEN2` supply active power).

We have to further investigate to solve them.

AndreaBartolini, 2026-08-21:
Frequency collapse solved by PR #196

AndreaBartolini, 2026-09-17:
I made some further investigations and I noticed that in the OpenIPSL the fault component has `R=0 pu` and `X=1e-5 pu`, so I put the equivalent parameters (`R=0 ohm` and `X=4.761e-4 ohm`) in the `FAULT4` model of `IEEE14busShort4`.

Then I re-simulated the transient, the generator speed still decreases in PowerGrids because the increasing of the total power dissipated by the lines due to the overcurrent generated by the short is higher than the load reduction generated by the voltage decreasing (see figures below - total power calculation of lines and loads are now available on the develop branch):

[image]

[image]

I also made a check of the parameters used in the PowerGrids and they are according to the PSCAD IEEE 14 Bus report.  There are some differences between PowerGrids and OpenIPSL in the parameters used in the loads, OpenIPSL uses different values for `alpha` and `beta` (they are not specified in the PSCAD report) and (maybe consequently) different values for `PRef` and `QRef`.

To align the two libraries is not so easy because by changing the loads parameters all the initial values should be (probably) recalculated consequently... so my opinion is to avoid to do it if it is not strictly necessary.

The main point, in my opinion, is to understand if the phenomenon highlighted in the figure above is acceptable or not and, if not, to investigate the model in order to find the origin of it.

@ceraolo  and @casella , what do you think about this? any suggestion?

ceraolo, 2026-09-19:
**First** of all let me first ask a favour. Please when you upload plots ensure that the textual labels and legend are near in size the message text. this is very easily obtained: just do copy/paste starting from a smaller sized plot. In this way, also the line widths will become larger, and therefore more visible. This will avoid the need of continuously zoom and pan to reason on your messages. Thanks.

**Second**.  Your explanation of the generator slowing down may be reasonable or not. 
In textbooks, the behaviour I mentioned in this ticket's description is explained considering a very simple machine model: an EMF in series with a transient reactance. In this case the power delivered is 
P=EU/Xt *sin(beta)
where E (the emf) and U (the terminal voltage)  as a first approximation can be held constant, Xt is the transient reactance (may be assumed roughly equal to xpdPu). If U goes down a lot, this power reduces, because, whatever beta, the maximum deliverable Pmax=EU/Xt will reduce, so the actually delivered power.
I cannot be sure about what happens in a so more complicated model as the IEEE14bus. The total power dissipated by lines is not so significant for me.

**Third** acceptability.
I think that in case of large models like this, the library should not rely on the opinion on one (or more) reader about the "acceptability" of a result.
I think that for PowerGrids be credible, at least some results of the IEEE14 bus should be compared with some reference/benchmark. Otherwise, we'll be always unsure of what we get.
This was done with ENTSOE, and the entsoe examples are still the most useful models in PowerGrids.
I remember to have read in the past papers with comparison with benchmark of some OpenIpsl models, but do not remember details.

I do not know if there is on the PowerGrids developers' part enough man-time left to try to make a comparison with a benchmark. Sure it would be extremely useful for the users, and might possibly be rewarded by the publication of a paper.

A starting point could be to check papers on OpenIpsl and find (I expect, and hope, some exist) comparison with of results with a benchmark and try to make a one or more comparison with a benchmark of its results.

Obviously this is not the only possible way. Another might be to check with the authors of the original IEEE14bus example in PowerGrids1.0, if they have some results that can be used as a benchmark.

**Fourth.**
You say:
_To align the two libraries is not so easy because by changing the loads parameters all the initial values should be (probably) recalculated consequently... so my opinion is to avoid to do it if it is not strictly necessary._

I do not understand this. The entire effort to move from PowerGrids 1.0 to 2.0 was originated (at least I see this way) from the need to avoid setting manually all the initial values.
The very big advantage of PowerGrids over competition is PowerFlow. But if having PowerFlow means later having to manually re-rite hundreds of numbers, 99% of the advantage is lost.
I think that in year 2026 it is crazy to ask people to manually copy many numbers from a model to another,.
Do I miss something?

ceraolo, 2026-09-20:
Further thought.
I realise only now that you mention that PowerGrids parameters according to the PSCAD IEEE 14 Bus report.

To it seems to me that the easiest and more effective thing to do is to make a check of the response of Powergrids and PSCAD to the same perturbation (best candidate; our fault on node 4), isn't it?

AndreaBartolini, 2026-09-20:
@ceraolo  thanks for your answer and suggestions.
I agree to the necessity to make a validation against a benchmark, and also I agree to use the PSCAD as benchmark reference.

So far, I haven't found any PSCAD reports to use as a benchmark, particularly for the fault transients, I will continue looking for them.

ceraolo, 2026-09-20:
I must add that the comparison will not be straightforward: Powergrids is a phasor simulator (and only direct sequence), while EMTDC a detailed time domain one (lile the "Electrical.Analog" folder componnts in MSL), so much more detailed.
But the trends of angular speed should be very near to each other.

ceraolo, 2026-09-20:
https://www.pscad.com/knowledge-base/article/26

casella, 2026-09-20:
For the record, if I'm not mistaken, the original IEEE14bus was validated against an IPSL model, which was also validated with RTE former tool. The power flow was identical, the transient was slightly different and we traced the difference to an issue with the IPSL model, that was written in a procedural way with a solution that was apparently incorrect and did not fulfill the original (declarative) residual equations of the system.

Which IMHO proves once more that the a-causal approach to modelling is preferrable.

I'm happy if we can figure out some other reference.

ceraolo, 2026-09-20:
If validation of ieee14bus was made against ipsl, it should be easy to validate also node 4 fault with it.
If, as @AndreaBartolini  says, data for ieee14 was taken from pscad, benchmarking with pscad should be easy. This has the advantage that pscad is supported by RTE.

 If resources are limited, programmers can choose one of  these options; otherwise, both is even better.

AndreaBartolini, 2026-09-21:
> https://www.pscad.com/knowledge-base/article/26

@ceraolo thanks.
Indeed I've already found this link but the benchmark is just a .pscx file, not a report, so I cannot use it.

ceraolo, 2026-09-21:
Maybe it cun be run with a trial version of PSCAD?