PowerGrids/PowerGrids #173: New system Object - Regression in Develop version?

ceraolo, 2025-09-24:
System model in PowerGrids 1.0.3 contains the important variable `referencefrequency`, which allows to choose whether the phasors are to be computed with reference to a Gauss' plane rotating at a set frequency or a frequency made equal to the angular speed of a generator.
This was very useful, and the difference well explained in models Tutorial.IslandOperation.TwoGenerators.Generator and Tutorial.IslandOperation.TwoGenerators.Reference.

Unfortunately, in the latest (November 2024) develop version this very important choice has gone. 
System object has become much more complex, but has lost this. The two examples mentioned above are not there anymore.

@casella @AndreaBartolini It this a feature or a bug? Frankly speaking,  I liked the possibility of having these two options. 

Moreover, since the new system object has no info, the user does not know how the correspondence time -> phasors is done in this develop version. 
I can imagine that the correspondence is intended to be made always with the reference frequency, but this is disadvantageous for both efficiency and convergence performance of the models, in comparison with the solution in which the system object is connected to a generator, these disadvantages can well be seen using the already mentioned Tutorial.IslandOperation.TwoGenerators.Generator and Tutorial.IslandOperation.TwoGenerators.Reference models

ceraolo, 2025-09-25:
Thank you, @AndreaBartolini.
Sorry you lost time in explaining here something already in the docu.

In principle, you could have just mentioned where to read, but I'm glad you did more.
In fact,  in my opinion, the explaiation in _PowerGrids.UsersGuide.Tutorial.LibraryArchitecture - section: 3.3_  is too technical, more oriented to developers than the final users.
I woul expect that a library, expecially in its tutorials (except, possibly for some "more in depth" section), would use a language adaquate to the final users, just as you did in this ticket.

Is it mandatory that _all_ the PowerGrids users should know about overconstrained connectors? Not sure.

Thanks again. I'm going to close this ticket.

AndreaBartolini, 2025-09-25:
@casella I can add something like my comment above to the library documentation.... what do you think?

AndreaBartolini, 2025-09-25:
> when I have a governor which tries to control the speed to 1 p.u. it will control it to be equal to the angular speed of the strongest generator (or the one having priority=0)?

This is true only in the _Island_ scenario, if one _Infinite bus_ or an _Equivalent grid_ is present (_synchronous_ scenario) then the _reference frequency_ is set to 1 per-unit.

If the point is to have the possibility to set a _reference frequency_ independently from the generators speed also in _Island_ scenario (i.e., without _Infinite bus_ or _Equivalent grid_) then we need to define a _dummy_ bus that does not exchange power and just prescribes the _reference frequency_.

Let me know is this is the requirement.

ceraolo, 2025-09-25:
> This is true only in the _Island_ scenario, if one _Infinite bus_ or an _Equivalent grid_ is present (_synchronous_ scenario) then the _reference frequency_ is set to 1 per-unit.

Yes, sure
 
> If the point is to have the possibility to set a _reference frequency_ independently from the generators speed also in _Island_ scenario (i.e., without _Infinite bus_ or _Equivalent grid_) then we [...]

Is it possible with a limited effort?
I'm not posing _requirements_, but trying to envigage usage patterns. If what you mention here is possible, it seems to me it solves the problem to having 1 pu to correspond to 50 Hz (or whatever, mostly 50 Hz in Europe), also in island mode.

Note that _island mode_ does not refer just to when after a strong perturbation the grid is split into parts. It refers also to whole systems such as the IEEE 14 node network implemented in PG.

ceraolo, 2025-09-26:
A further thought. 
In the current implementation of pg 2.0, if a system splits into islands every island gets an omega reference of its own, equal to the speed of the most powerful generator, true?
And this implies that 1 pu for speed /frequency means a different SI value for each of the island?

Afaict, this is very far from what system engineers want! They want a unique frequency reference value for the whole system, even If split into islands.

ceraolo, 2025-09-29:
Researchers never lose focus! (even at nights, sometimes)

But they can also take complete breaks: last weekend in Capraia I went on some wonderful hikes, completely detached for many hours from cellular phones and the Internet